thingsboard/thingsboard-gateway #15: Custom gateway create device problem

anton-pytel, 2017-05-08:
I was trying to implement a python script that would do an interface between Vera Lite (Z-wave controller) and Thingsboard via MQTT, but got across a very specific problem.  
 
I have published correct JSON to topic v1/gateway/connect. Althought device appeared to be created successfully, it was not possible to send telemetry or attributes through v1/gateway/attributes or v1/gateway/telemetry topics.
When I checked log it was always returning this message:
```
Failed to process publish msg [v1/gateway/connect][-1]
java.lang.IllegalArgumentException: messageId: -1 (expected: 1 ~ 65535)_
```  

After several hours of looking for a problem got this:   
https://github.com/thingsboard/thingsboard-gateway/blob/c8d7ce11dfe31c31c51c5266d0788f72db109582/src/main/java/org/thingsboard/gateway/service/MqttGatewayService.java#L128

There is a special Message Id set to header during publish. Problem is that many mqtt client implementations (python, lua, newest nodejs) does not have implemented this feature, so it makes it complicated to create a custom gateway not based on Java. Is it really needed to set message id to publish message?

Thank you for response in advance

ashvayka, 2017-05-09:
@anton-pytel Thanks for explanation. Yes, this message id is required to be able to identify request id. Every telemetry upload requires confirmation (while using QoS 1) and we need to know the msg id to use it as request id.

What client are you using? Are you using QoS 1?

ashvayka, 2017-05-09:
@anton-pytel Don't get me wrong, we can discuss and implement workaround if this is a showstopper, but I just want to make sure that this is showstopper.

anton-pytel, 2017-05-10:
Thank you for your comments, I have come back today to that with trying QoS1 using Python MQTT Paho client. I would swear that I definitely tried QoS1 as I was trying so many combinations (JSON formats, different clients, different QoS, ...) Today I set QoS = 1 for publishing to v1/gateway/connect, v1/gateway/attributes and v1/gateway/telemetry and it all worked. So no showstopper all good, thank you again.

closed 2017-05-10 by anton-pytel