thingsboard/thingsboard-gateway #16: Error: Too many publishes in progress

nmenoni, 2017-05-10:
When the tb-gateway (GW) loses the connection with the server for a period of time such that (I suspect) the number of messages not transmitted are greater than the "maxInFlight" parameter, then the connection is re-established the GW generate this error in the log:
```
2017-05-10 00:11:07,311 [MQTT Call: 452a5e28-9587-47bd-a61f-6d03124f6b21] WARN  o.t.g.service.MqttGatewayService - [Sensor_Verde][1370] Failed to publish device telemetry!
org.eclipse.paho.client.mqttv3.MqttException: Too many publishes in progress
	at org.eclipse.paho.client.mqttv3.internal.ClientState.send(ClientState.java:498)
	at org.eclipse.paho.client.mqttv3.internal.ClientComms.internalSend(ClientComms.java:136)
	at org.eclipse.paho.client.mqttv3.internal.ClientComms.sendNoWait(ClientComms.java:165)
	at org.eclipse.paho.client.mqttv3.MqttAsyncClient.publish(MqttAsyncClient.java:1126)
	at org.thingsboard.gateway.service.MqttGatewayService.publishAsync(MqttGatewayService.java:334)
	at org.thingsboard.gateway.service.MqttGatewayService.onDeviceTelemetry(MqttGatewayService.java:198)
	at org.thingsboard.gateway.extensions.mqtt.client.MqttBrokerMonitor.onDeviceData(MqttBrokerMonitor.java:181)
	at org.thingsboard.gateway.extensions.mqtt.client.listener.MqttTelemetryMessageListener.messageArrived(MqttTelemetryMessageListener.java:42)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.deliverMessage(CommsCallback.java:469)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.handleMessage(CommsCallback.java:380)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.run(CommsCallback.java:184)
	at java.lang.Thread.run(Thread.java:745)
```
 
and in the server-side user interface Device telemetry data does not arrive.

Restart the tb-gateway service does not fix anything.
The only way I found to fix the problem is to delete the GW and create it again in the Thingsboard console user interface, which has the effect of losing all historical data.

pathakhcst, 2017-09-07:
I am also facing the same issue. The Gateway stop sending data to THB server and it never recover automatically. Server restart, Gateway restart, services restart, router restart all never helps. The only way to get the data back is to delete the GW from Thingsboard server and create new. Any lead on this

ssmaurya, 2017-09-07:
Gateway creates storage folder for storing unsent messages. You can try deleting it and restart gateway

pathakhcst, 2017-09-07:
Can you share the path to access the storage or how to wipe out the unsent messages from persistent storage?

pathakhcst, 2017-09-07:
I figured the storage folder under **/usr/share/tb-gateway/bin/storage/** on Raspberry Pi
1) Delete the folder with alpha-numeric name(auto generated) by **sudo rm -r <folder_name>**
2) Restart the thingsboard gateway service by **sudo service tb-gateway restart**.
The attribute and telemetry upload will start.

ssmaurya, 2017-09-07:
Glad that it worked. This is an issue related to file persistence. For some reason device names/id get corrupted, which essentially results into failure.  We will try to address this issue and once fixed will do PR

pathakhcst, 2017-09-07:
In IoT ecosystem the gateway is always remote and mostly connected on wireless technology like 4G,LTE. So the connection drop is  always considered because of coverage variance.
After every disconnect I have to perform delete the storage folder and restart the service which is not a good practice.
Is there any solution which help me to run the gateway without any disruption?

nmenoni, 2017-09-07:
I also consider that this is a very critical problem. It would be nice to
fix it soon.
Thanks!!

pathakhcst, 2017-09-12:
Can we disable the logging of unsent messages? I tried 0 for maxinflight variable but its not working.
Can we use previous versions if this bug is not present in them? Version 1.2 is having the same bug.

nmenoni, 2017-10-13:
Components: TB Gateway (v1.2.1) and a TB Server (v1.3.1).

I was doing some network captures with Tcpdump and I can see that of the 3 different types of MQTT PUBLISH (to topics: _"v1/devices/me/telemetry"_, _"v1/gateway/attributes"_ and _"v1/gateway/telemetry"_) that the gateway makes to the TB Server **only those made to the topic "v1/gateway/attributes" are acknowledged** (Publish Ack). 
This agrees that in the storage folder I can see all the messages ID (s-xx.msg) of the unacknowledged messages and all correspond to the unacknowledged topics I saw in the network capture too, and any correspond to attributes message.

So, in other words, apparently only the messages of attributes ("v1/gateway/attributes") are being acknowledged from the TB Server, but not those of telemetry ("v1/devices/me/telemetry" and "v1/gateway/telemetry") and are those that end up filling the storage.

nmenoni, 2017-10-14:
After some more tests, I realized that the PUBLISH ACK were not coming from the server for telemetry because there was a rule that was misconfigured (did not check that the type was defined and also misspelled the json in the alarma processor).

Then it seems that the TB server will not send a PUBLISH ACK if there is a misconfigured rule, but the telemetry value is stored in the data base anyway.
Is this form of operation ok?

masaryk, 2017-10-17:
we got to this error as well, it is really critical. @ashvayka when do you plan fix this bug?

ashvayka, 2017-10-17:
Hi @masaryk, as mentioned by @nmenoni , the issue is in wrong rule configuration. The rule configuration is critical for a data flow and if it is wrong - this means that the overall system behavior is wrong. According to a MQTT RFC, there is no error messages and the only possible option in case of MQTT is to close the connection with the gateway. Once you fix the rule configuration, the GW issue will not be longer valid.

The more accurate tracking of data processing flows is planned for v1.5 and is discussed here:
https://github.com/thingsboard/thingsboard/issues/322

pathakhcst, 2017-10-31:
tb-gateway: Raspberry Pi having mosquitto broker
Tb-server:Windows 10
@ashvayka In my use-case the data was consistent for 4 months without any issuse but the problem started when there was drops in internet connectivity. The data flow was like following:-
Sensor->MQTT broker(Rasp-Pi)->Tb-gateway(Rasp-Pi)->Tb-server
I configured the mqtt-config.json properly and not any other rules at server side. But I faced this issue. If its because of misconfigured rules then it should not worked for 4 months.

ssmaurya, 2017-10-31:
When we ran into this issue, we found out that there was some synchronization issue after network failure, which was resulting into corruption of data being written to storage folder.  Once we implemented queue for our managing data from incoming http data source, the issue went away.   May not be related, but thought of sharing this just in case, if this helps.

nmenoni, 2017-10-31:
What do you mean by "we implemented queue for our managing data from
incoming http data source"? Could you give us a little more detail?
Thanks!

ssmaurya, 2017-11-02:
We had multiple devices sending data over http to gateway. After getting the data from devices, we would put in queue and remove from queue after received data had been processed. This basically ensured sequential publishes and it resolved the. 'too many publishes issue' for us and also fixed data corruption in storage on network failure. We did not yet root cause where the problem resides. Since it worked, we moved on with this approach.

masaryk, 2017-11-03:
@ssmaurya what do you propose please as a solution? this is really unpleasant bug. :-(

ashvayka, 2017-12-29:
@masaryk @ssmaurya @nmenoni @pathakhcst 

Hi Guys,
we are actively working on new solution that is based on HSQLDB and different MQTT client that is based on netty. First PR is [here](https://github.com/thingsboard/thingsboard-gateway/pull/34). The fork of netty mqtt client is [here](https://github.com/ikulikov/netty-mqtt) We will move this fork to thingsboard organization once it is ready.

Regards!

nmenoni, 2017-12-29:
That is very good news! Thank you very much!

masaryk, 2017-12-29:
@ashvayka great news to hear, thanks! this bug is very annoying and causes serious problems in maintenance running instances in real cases. keep good work! 👍

mp-loki, 2018-04-17:
This was partially fixed in thingsboard-gateway releases 1.2.2 and 1.4.0. Fixed in master branch

closed 2018-04-17 by mp-loki