thingsboard/thingsboard-gateway #24: Modbus Support (Feature suggestion)

Javierete, 2017-09-26:
Hello!

First excuse me if it's not the place to make a suggestion.
I think will be good to support modbus TCP as many industrial devices (PLC's) doesn't have the capabilty to do OPC-UA and modbus is a widely used protocol.
I have no idea how to implement it (no javascript knowledge) so asking here..

Congratulations for the great project!

dyosick, 2018-04-08:
Can you help with testing or maybe have some new ideas how to improve? Please, see [this PR](https://github.com/thingsboard/thingsboard-gateway/pull/54(https://github.com/thingsboard/thingsboard-gateway/pull/54)

Javierete, 2018-04-12:
Hello @dyosick , that's great!!

Javierete, 2018-04-12:
Hello @dyosick, that's great!! 
I can test this feature maybe on next weeks, first I'll need to setup dev environment to compile tb-gateway (I currently have dev env to compile TB).

Many thanks for the feature, I can test many devices with modbus here mainly with industrial PLC's from Schneider, Bekchoff (with modbus NOT ads) and custom ones such as power analyzers.
 
I believe TB team will merge this in the near time, from my point of view (I'm not a programmer) is very well documented.