thingsboard/thingsboard-gateway #26: build fails due to dependency issues

druidtides, 2017-09-29:
I downloaded the lasted versions of thingsboard and gateway per instructions while things board build succeeds, gateway fails:[INFO] Scanning for projects...  I have both projects under common thingsboard directory.

[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building Thingsboard Gateway 1.0.0
[INFO] ------------------------------------------------------------------------
[WARNING] The POM for org.thingsboard.common:data:jar:1.1.0 is missing, no dependency information available
[WARNING] The POM for org.eclipse.paho:org.eclipse.paho.client.mqttv3:jar:1.1.1-SNAPSHOT is missing, no dependency information available
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 1.675 s
[INFO] Finished at: 2017-09-29T16:18:28+01:00
[INFO] Final Memory: 11M/150M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project gateway: Could not resolve dependencies for project org.thingsboard:gateway:jar:1.0.0: The following artifacts could not be resolved: org.thingsboard.common:data:jar:1.1.0, org.eclipse.paho:org.eclipse.paho.client.mqttv3:jar:1.1.1-SNAPSHOT: Failure to find org.thingsboard.common:data:jar:1.1.0 in https://repo.eclipse.org/content/repositories/paho-snapshots/ was cached in the local repository, resolution will not be reattempted until the update interval of Eclipse Paho Repo has elapsed or updates are forced -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException

smilebot, 2018-02-24:
@druidtides Did you ever figure out the solution to this issue? I'm still facing this issue.

NKAVY, 2018-03-06:
Hi,

I also faced this build issue and hence reverted to building "thingsboard 1.2" version and then, building latest tb-gateway. It worked.

mp-loki, 2018-04-17:
you need to check out and build the [ThingsBoard source code](https://github.com/thingsboard/thingsboard) and then update your gateway pom.xml with the corresponding thingsboard.version (current is 1.4.1-SNAPSHOT):

```xml
<thingsboard.version>1.4.1-SNAPSHOT</thingsboard.version>
```

closed 2018-04-17 by mp-loki