thingsboard/thingsboard-gateway #30: Documented OPC-UA example doesn't work

LilaLipetti, 2017-11-07:
I have installed latest thingsboard + gateway + KEPServerEX
and followed the OPC-UA tutorial
https://thingsboard.io/docs/iot-gateway/getting-started/#step-9-connect-to-external-opc-ua-server

The documented tutorial doesn't work.

See more
https://groups.google.com/forum/#!topic/thingsboard/lBeH7zRSLy0

See my git for raw debug logs
https://github.com/LilaLipetti/thingsboard-opc-testing

bodiroga, 2018-02-16:
It's a shame that the OPC UA extension doesn't work well, because the whole platform (IoT gateway and server) is awesome!

Many thanks for your hard work and looking forward to seeing this fixed someday!

Best regards,

Aitor

mp-loki, 2018-03-15:
Fixed in ThingsBoard Gateway versions 1.2.2 and 1.4.0

closed 2018-03-15 by mp-loki