thingsboard/thingsboard-gateway #36: Code base errors

yimingliu0216, 2017-12-20:
I got successful maven build.

However, this is lots of code errors in the code base. The errors are categories to four types.

1. Eclipse complains log object does not define. 
    e.g.  log.info("[{}:{}] MQTT broker connection established!", configuration.getHost(), configuration.getPort());

2.  Eclipse complains final variables not initialized. "The blank final field nodeId may not have been initialized"

3. Object function setter & getter function undefines. I saw properties but there is not explicitly setter and getter for those properties. 

4 Object not constructor. I saw some objects only has properties (some of them define as final), However, this is not constructors for those objects.

Do you I missing some setting here?

mp-loki, 2017-12-20:
Hi @yimingliu0216 !
You have to install [Lombok plugin for Eclipse](https://projectlombok.org/setup/eclipse)
Lombok library does all the stuff that you mentioned (@Data, @Slf4j annotations etc). Please visit [Project Lombok](https://projectlombok.org/) for more info.

closed 2017-12-20 by mp-loki

yimingliu0216, 2017-12-20:
Thanks. It works!