thingsboard/thingsboard-gateway #52: How to explore data from OPC-UA devices?

eugene08, 2018-03-12:
I met a problem in my exploration, The opc-ua server can connection successful.
But cannot read tag value in devices, No changes have been made to any of the files.
Does KEPServer the expiration have an impact?
Please help me check where is a problem.
Thanks!

mp-loki, 2018-03-12:
Hi @eugene08 ! Could you please tell which ThingsBoard and Gateway versions are you using?

eugene08, 2018-03-13:
Hi @mp-loki ，Thanks for your reply. ThingsBoard is currently version 1.4.0 and gateway is currently version 1.2.1.

mp-loki, 2018-03-13:
Please check out the gateway release 1.2.2. OPC-UA extension had issues in 1.2.1.

eugene08, 2018-03-14:
Thanks, I went to try. :-)

mp-loki, 2018-04-17:
Hi @eugene08. Is this issue resolved? If no more concerns, I'll close it

closed 2018-06-27 by mp-loki