thingsboard/thingsboard-gateway #53: Gateway not working on thingsboard - Ubuntu 16.04

BeIllegalBeagle, 2018-04-03:
Problem

I have the EMQ mqtt broker running fine on my ubuntu build, and can run everything it is expected to.
I am trying to connect my EMQ broker to Thingsboard using the gateway, however Thingsboard does not acknowledge the gateways connection, yet my EMQ broker does as shown:

[image]

[image]

[image]

The Broker is running on port 1883. The gateway.yml is correct here:

[image]

And finally my mqtt json config file(which I had to download from [here](https://thingsboard.io/docs/iot-gateway/getting-started/#step-82-mqtt-broker-gui-configuration) because the default file is not accepted by the Thingsboard UI):

[image]

My Gateway always displays 'Not sync':

[image]

What am I doing wrong here?

mp-loki, 2018-04-03:
Hi @BeIllegalBeagle !
Can you please tell us which version of ThingsBoard Gateway are you using?
The GUI configuration is only available from version 1.4, but the gateway.yml you provided seems to be from v.1.2

mp-loki, 2018-06-27:
Any updates on this issue? otherwise I will close it

BeIllegalBeagle, 2018-06-28:
Hello again! Sorry I'm unable to check which version I had but my issue was the fact that I had not referenced the access token in my gateway on thingsboard, after I did that it was ok and I could see my data. Other wise I didn't have any other issue

closed 2018-06-28 by mp-loki

shrinivassridhar, 2019-01-30:
@BeIllegalBeagle hello ! i wanted to know weather the status got synced or not . As you said the default mqtt config.json file is not accepted by the thingsboard UI, but even when i upload the downloaded file the file gets uploaded but the status is still not in sync. Does anyone know how to resolve the issue ?

mp-loki, 2019-01-30:
@shrinivassridhar I suggest you create another issue for your case. Please upload your tb-gateway.yaml, tb-gateway.log, and mqtt-config.json that you are using. Don't forget to put the device access token in tb-gateway.yml

shrinivassridhar, 2019-01-30:
@mp-loki hello sir ! As u told, i verified all my files and even have added my access token to the gateway.yml, and still if i import my mqtt-config.json file the sync status gets unsynced. 

[mqtt-gui-extension-configuration.txt](https://github.com/thingsboard/thingsboard-gateway/files/2811846/mqtt-gui-extension-configuration.txt)

[thingsboardyml.txt](https://github.com/thingsboard/thingsboard-gateway/files/2812002/thingsboardyml.txt)

[gatewayyml.txt](https://github.com/thingsboard/thingsboard-gateway/files/2812006/gatewayyml.txt)