thingsboard/thingsboard-gateway #56: Maven failed to build, unable to find the "nl.jk5.netty-mqtt" package

caoyingde, 2018-04-11:
unable to  build maven
<dependency>
            <groupId>nl.jk5.netty-mqtt</groupId>
            <artifactId>netty-mqtt</artifactId>
            <version>${netty-mqtt.version}</version>
        </dependency>
**I tried to compile the https://github.com/jk-5/netty-mqtt project into a jar package, but there was a lack of class, and I would like to give me a complete netty-mqtt.jar file. Thank you very much!**

mp-loki, 2018-04-11:
Try building from our fork:
https://github.com/thingsboard/netty-mqtt

caoyingde, 2018-04-11:
Unable to find the MqttClientCallback file, has the code been submitted?

[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] F:\gitWorkSpace\thingsboard-gateway\src\main\java\org\thingsboard\gateway\service\gateway\MqttGatewayService.java:[29,18] 错误: 找不到符号
[ERROR]   符号:   类 MqttClientCallback
  位置: 程序包 nl.jk5.mqtt
F:\gitWorkSpace\thingsboard-gateway\src\main\java\org\thingsboard\gateway\service\gateway\MqttGatewayService.java:[107,72] 错误: 找不到符号
[INFO] 2 errors 
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 12.711 s
[INFO] Finished at: 2018-04-11T11:51:42+08:00
[INFO] Final Memory: 31M/422M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:2.5.1:compile (default-compile) on project gateway: Compilation failure: Compilation failure:
[ERROR] F:\gitWorkSpace\thingsboard-gateway\src\main\java\org\thingsboard\gateway\service\gateway\MqttGatewayService.java:[29,18] 错误: 找不到符号
[ERROR] 符号:   类 MqttClientCallback
[ERROR] 位置: 程序包 nl.jk5.mqtt
[ERROR] F:\gitWorkSpace\thingsboard-gateway\src\main\java\org\thingsboard\gateway\service\gateway\MqttGatewayService.java:[107,72] 错误: 找不到符号
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException

mp-loki, 2018-04-18:
Hi @caoyingde ! Is the issue resolved? If so, I plan to close it.

caoyingde, 2018-04-20:
@mp-loki   I've solved this problem.

closed 2018-04-20 by mp-loki