thingsboard/thingsboard-gateway #57: Can the same tb-gatway enable opc-ua and MQTT protocols to connect to external services?

BIT-masterpy, 2018-04-12:
Hi, I want to connect two external services to support the MQTT and opc-ua protocol respectively. Can I enable both of these functions through configuration in a tb-gateway?
[image]

mp-loki, 2018-04-12:
Yep, that's possible. Which version of TB Gateway are you using? The config file is different between 1.2.2 and 1.4

mp-loki, 2018-04-12:
Please use 1.2.2 instead. 1.2.1 is more than a year old an multiple issues have been fixed since.
https://github.com/thingsboard/thingsboard-gateway/releases

BIT-masterpy, 2018-04-13:
Thank you very much. In version 1.4, how to configure MQTT and OPCUA to start simultaneously?
[image]

mp-loki, 2018-04-13:
`remoteConfiguration: false`
extensions:
- 
  id: "mqtt"
  type: "MQTT"
  extensionConfiguration: mqtt-config.json
-
  id: "opc-ua"
  **type: "OPC UA"**
  extensionConfiguration: opc-config.json

BIT-masterpy, 2018-04-16:
thank you @mp-loki