thingsboard/thingsboard-gateway #58: Thingsboard gateway device connected but not created in thingsboard devices.

NikhilBijagarni, 2018-04-13:
Hi,
I in running thingsboard application and thingsboard gate way application on the same machine. I am establishing a gateway between mqtt mosquitto and thingsboard application. when i am publishing the data i am getting device connected in thingsboard gateway log but the device is not getting created.
The log i am getting is,

2018-04-13 12:14:43,481 [MQTT Call: cbb9442b-ec70-4178-a88d-92452c60dcb2] INFO  o.t.g.s.gateway.MqttGatewayService - [Niks] Device Connected!
2018-04-13 12:14:43,786 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-04-13 12:14:43,786 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-04-13 12:14:44,791 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress

I am continuously getting  outgoing queue is not empty. [1] messages are still in progress.

my thingsboard gateway.yml configuration is.

server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"

gateways:
  tenants:
    -
      label: "tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: "${GATEWAY_HOST:localhost}"
        port: 1883
        retryInterval: 3000
        maxInFlight: 100
        security:
          accessToken: "${GATEWAY_ACCESS_TOKEN:yWcNs3hZ9ch4rPpI9t79}"
      remoteConfiguration: true
extensions:
        -
          id: "http"
          type: "HTTP"
          extensionConfiguration: http-config.json
        -
          id: "mqtt"
          type: "MQTT"
          extensionConfiguration: mqtt-config.json

Please help me the solve this issue.

closed 2018-04-14 by NikhilBijagarni

mp-loki, 2018-04-14:
Hi @NikhilBijagarni ! So what was the issue?

rof20004, 2018-07-10:
@mp-loki `...when i am publishing the data i am getting device connected in thingsboard gateway log but the device is not getting created.`

NikhilBijagarni, 2018-07-10:
The solution for me that worked was, I observed that each time you change parameters from GUI i was not getting Sync status. I had to restart thingsboard service every time i update configuration, then i would see the new configuration was sync. After this when i published data to mosquitto the device was getting created and data was also visible on thingsboard. I was using Snapshot version to build application. but when i used the released version this issue was gone and now its working fine.