thingsboard/thingsboard-gateway #60: OPC-UA not connecting/keystore poblem

quirosaur, 2018-04-16:
Hello!

I discovered a potential bug, while following example. 
I have gateway and thingsboard on version 1.4.0, both installed on same computer. 
i try to connect to prosys opcua server on raspberry pi, and i am using default keystore and certification.

however, when starting tb-gateway, log shows error messages
```java
ERROR o.t.gateway.util.ConfigurationTools - Keystore configuration: 
[KeystoreConfiguration(type=PKCS12, location=example.pfx, fileContent=, 
password=secret, alias=gateway, keyPassword=password)] is invalid!
java.security.UnrecoverableKeyException: Get Key failed: Given final block not properly padded. 
Such issues can arise if a bad key is used during decryption.
````
when using jks, error is:
````java
ERROR o.t.gateway.util.ConfigurationTools - Keystore configuration: 
[KeystoreConfiguration(type=JKS, location=example.jks, fileContent=, password=secret, 
alias=gateway, keyPassword=password)] is invalid!
java.security.UnrecoverableKeyException: Cannot recover key
````
and apparently caused by this, opcua tries to connect to null (i've tried ip with and without the opc.tcp:// header)
````java
INFO  o.t.g.e.opc.OpcUaServerMonitor - Initializing OPC-UA server connection to [null:4840]!
````

mp-loki, 2018-04-16:
can you provide your **tb-gayeway.yml**? Is remoteConfiguration **true** or **false** ?

quirosaur, 2018-04-16:
i have tried with both. when true and configured following example , the error is:

[tb-gateway.log](https://github.com/thingsboard/thingsboard-gateway/files/1915399/tb-gateway.log)

and here is the tb-gateway.yml

[tb-gateway.yml.txt](https://github.com/thingsboard/thingsboard-gateway/files/1915406/tb-gateway.yml.txt)

mp-loki, 2018-04-17:
Hi @quirosaur ! Thank you for reporting this issue. I found that the resource files that are provided in the package are not accurate. As a workaround you have to do two things:

1) in opc-onfig.json change "keyPassword": "secret"
2) in opc-onfig.json add "host": <YOUR_HOST>, at the same level as "port" property.
If there are any other issues with the certificate, you can re-generate it as described [here](https://thingsboard.io/docs/iot-gateway/getting-started/#step-91-provision-gateway-credentials-to-kepserverex)
Meanwhile I will update the release archives

BIT-masterpy, 2018-04-17:
Have you solved the problem? @quirosaur

quirosaur, 2018-04-17:
Hi, i tried the solutions, keystore is not apparently problem anymore but opc-ua connection problem persists. here are the logs and config:
[opc-config.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/1918458/opc-config.json.txt)
[without_opc_tcp_header_tb-gateway.log](https://github.com/thingsboard/thingsboard-gateway/files/1918459/without_opc_tcp_header_tb-gateway.log)
[with_opc_tcp_header_tb-gateway.log](https://github.com/thingsboard/thingsboard-gateway/files/1918460/with_opc_tcp_header_tb-gateway.log)

in "host", should i use ip as "opc.tcp//xxx.xxx.xxx.xxx" or just bare ip?

anyways, i tried both, neither worked.

mp-loki, 2018-04-18:
No, you don't need the "opc.tcp" part, the tb-gateway will add it by itself.
So what happens is that the gateway connects to "opc.tcp://<host>:<port>/", gets the list of endpoints and iterates over them to find the one that matches the security policy.

It looks like that the port 4840 that you are trying to connect to is a local discovery port (by default) in Prosys OPC-UA server, and you have to connect to UA TCP port, which is 53530 by default.
I don't have a working Prosys OPC-UA server, so I was unable to check if the assumption is correct or not

quirosaur, 2018-04-18:
I'm sorry, i accidentally gave wrong name of opc-ua server. It's codesys server.

anyhow, i apparently managed to get connection to opcua. in raspberry pi /etc/CODESYSControl.cfg i added 

[CmpOPCUA] 
NetworkAdapter=eth0
NetworkPort=4840

this was done in remoteconfig = false, because apparently opcua extension is broken in web GUI.
[tb-gateway.log](https://github.com/thingsboard/thingsboard-gateway/files/1923371/tb-gateway.log)

@mp-loki thank you for help! 👍

closed 2018-04-18 by quirosaur