thingsboard/thingsboard-gateway #65: The gateway connects opuca server to fail：UnrecognizedPropertyException: Unrecognized field "fileName"

BIT-masterpy, 2018-04-17:
Hello!
I have gateway and thingsboard on version 1.4.0, both installed on same computer.
I tried to use the gateway to connect to OpenOpcUa server with the following error.

`14:23:33.806 [pool-3-thread-1] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration

14:23:33.885 [pool-3-thread-1] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [opc][OPC UA]

14:23:33.891 [pool-3-thread-1] INFO  o.t.g.e.opc.DefaultOpcUaService - Initializing OPC-UA service!
14:23:33.975 [pool-3-thread-1] ERROR o.t.gateway.util.ConfigurationTools - Failed to load class org.thingsboard.gateway.extensions.opc.conf.OpcUaConfiguration configuration from {"servers":[{"applicationName":"Thingsboard OPC-UA client","applicationUri":"","host":"localhost","port":23562,"scanPeriodInSeconds":10,"timeoutInMillis":5000,"security":"Basic128Rsa15","identity":{"type":"anonymous"},"keystore":{"type":"PKCS12","location":"example.pfx","password":"secret","alias":"gateway","keyPassword":"secret","fileName":"example.pfx",
"file":"MIIKIgIBAzCCCdwGCSqGSIb3DQEHAaCCCc0EggnJMIIJxTCCBWoGCSqGSIb3DQEHAaCCBVsEggVXMIIFUzCCBU8GCyqGSIb3DQEMCgECoIIE+jCCBPYwKAYKKoZIhvcNAQwBAzAaBBTBjlJoN3UVdEU9BnUHdWaUob11AwICBAAEggTIZ3WuZbOvZDX8zjfZuTw0fReCq6UNbKwXI6MNfJiI44eer4kK8Rz6Uqre0sN0eIE51J1SJL2+VzrfJkYBLc2uQMYYOJvUycZeXLltX2G2mKlw0jqGO/o3fVOyE9YD0BQ5kWLUEf+/TaWTifrD6vE3v/W4EuQVOvG916B2PidfGT0pRCykWXJWaIhHIw+/kTybqxIvRRqtF6qRZJhqFyoRzcGHLgNOtfH9JYcP3vTZgvC+Jf7kVE7x4xWyNGuJAO3NJDx/sZhpkm6pbrP9UpEXIkBhpdlIZLgN1vKEv3x0KWCNnPJRX6+7fAdUGlJ8qNwS7yNps+2bvS9I+cpRfHvikUvPmZ/6yY7CAmqRSeDOlJ2ZiMdo27L4rcd9y0ymEMJW7d6G0TKsKV2vyIgPM4dKfGKinaWAlPJ6BzL+60fJyOXF1i5lpGu7hcVDgFrCx26SvMwAnDySg7ZTCgWUA7vQ0BcSGiavUJR3tcFn1UKzMIisuN+KESaQSyokVTkIIVwE/X8bCIihfTP+xaD6jRbT3lq5VD8UtUmpp91zwUOfaW48CY91Dot17ohgUDIXlYQAMTu0rdERvWv8UsbEV0OC0OfMp8+BwkqZgVsqn4BRtF4hgPnttPolkK410vRAMmNqtNJNlTCacvBdBsYSjkd0lA+aZWj/CnH+rIVOcBVvGqoQBfbGAcnPxnJZnBgdVGRYvG3F8VfvbBxdueTF80w37B9tkoor3Q1QE6OL0pIXsGx1udZRkwh/L3uGwUhdjtWXzcgwedLMc3krgg+s+HiYv9ijYbtlf4xLCqeDWMYfoePaYYBcZqB0tCjqOl9z8Qg8dOp+nSWm7aU/iZfb/aKWHJ4qnA6OwiCsFb1OlxgOKoKXb/nirsA61zCnTv1T6WMaWcVyenZlWQH9VJRbxGF4mDQwUH1ODwMOEkjzq6tQf1/pBAMLnPRJJVxwugRI3BhOPlh9dKHx6Kz/IOsWD6a74JGxeNwrlKJWdiFxT5PE0RdyAUb7hXTd+JfaTzW/y1vQP9fpiQtR/VkksjWkvhareumlOImMqNlETtMZpnhKJYHvf2KwvZcLlPJiZJRP4poH2RlKs0+PyWlSDLorfWU/Aih2HvP7CHsY4qqv5fyHHwkrdtRpEcSILP0x229wjEzgOBP7Dhqv0/FYRkQ6p4lYG0GfjGYLNtfhLjT4i8Kk2EX4D5TCuI5YEjB2uAO1bY6q7wCcEDJ1J/fuk+/PSF3POPSuT+gawtq7Wzzxd7l0KqcJcr/3DOfvwKepywPD1qXrg5N197/reQ5pWxRN0yu86Fvqdo4Vcz8rZzPsKijf3kO2BJohH8zdPEZeBpPbx9u19XIzIF1ALn74gse6GoVRQaygTF/qKWqmcpW4KnRlHLmgiWYEfPSO9PAG5UOObHDq7h/lLpCD2AbL6oaB90EINCcDT35qJenZJbaRiUdJXqvx79cqNWosUbDwuOIpJ+ZvVsDGIgXlOPv9kDIcZ4r1EbiLtmsEMSNWnjUXE/pPRKElFgbHmG07CbzMWIhXCEebGhnEy4FdTraidI6r8Q6CjpBgAxWntRPTucg2gC0fEBDB6HBSlAb+MekBFgQbwCk7KFCo+fuFd/zoFDy0BqRpPvwWL5XSj5HqMUIwHQYJKoZIhvcNAQkUMRAeDgBnAGEAdABlAHcAYQB5MCEGCSqGSIb3DQEJFTEUBBJUaW1lIDE1MjAyMDU3NzY5OTYwggRTBgkqhkiG9w0BBwagggREMIIEQAIBADCCBDkGCSqGSIb3DQEHATAoBgoqhkiG9w0BDAEGMBoEFMffFY5blz/6TYqX7DQ9ofE/vBZMAgIEAICCBABr4otaoQA6aScKn/oGap/7XxJQYoj5mDIs040v8kujDVtdIrBDVKzaU4A5XP2OjOgCx8e+bYop0F7uvnGRFPsrkCqdWBecT+yVY6z7UxTjqq1Dt1it/6GGbGpx9LieGRAsMCWZB80ly5b8QUfx565Ym8Bfl6eMHA8pVByAFdXpsUDJU0KqJjHmUGzxHe32ly9PguNtj8qh7ulPUfIsJXDOYWtbF7hPdz1qMIEbFfqs5Xf0ehy7gpVCyCfX0xCjrDKeosg0KbcZ3rbnCwxBCWpbDXMEyQ/c1DnUuoie/RuIV7QayZHP+rzxJ6RV549hT1lm2oIIOJtdUR/+vOCCSQgzVESXnS5e7HzGpSfzLFC6FSG2/BESY9PaYY+V5BXitOrY5mmNZbgGNKyi1p+Kp1lwHrjmB0szgXFynAxtGkA/kq05j31kycYMhY7ckkoZ92NUPELV190LAlz3IOoTx9C4yeyHYbC/Y0nzZs5FQvkajuzAU4EsI+Te3iFh1kohC0Ulk4SQKCPq+Ob/dvModO9l61R8yCHDxsZeFACaCCsH7RQJRxeV1LUVHoiX1aJVuE01bj3jEBQN+ZI1A8qNdbCI+/q1o/GMVKQxtz5PY02A8VIa1cMGUwY5ljJJlTdRzPqiy3qP5ClbHKsbWucUECKfhAkqTuS0Uqm+h3Y6IN6yYxYqjbKEIuqgwbGiOJZzu82nVG51QpBtvMf1eDyOSOkQ2JZey+vXlOghavQl8aFluv5FHigw9Zk9yVZf8Z8Jk1ruVvFzlB1NXnn2YqKtKhYb+J/dqVhKiToDMQccYY7Hw8HSBXRzl345EVLxyjCngyAU6VjztqjCW6HYNjNggyUVYmDQ49XOsq6U/7QhzBEbhYo48f7qvPVetRxShSjXt5Jnc+5mJG7UXHAhBSUayFv5T5CUmKlhbcuKyzX0Euib6WgajPdKcgib4bUpAD7L1Y2cEYorWE/yASnY06ZNabRAcyRSQnkZecZ0X95GTUmX08oOblskmQlBzQunjQSRvdpLBCu1356NwYZ8CJ/bqMp1g25FBKXCStKoDuluD1PT9Trd7Dv+f65EPvKTt6S90h/EscPV3D1sx+foPf6kdPl3LfX7/o4H2b01yErbTDCYcew0qkoRLpqHJr8GAyTjW6478sgqxCDbW1R0dpGiJkCHyqEQd8g/b9zu+4Pab4Y1O5lIKhhnnVWfajdWmY1NJk7hvpJwpkydRTuic/JhMnmUafuoGW5ZLxfjyejZnH4z8SIxVXJeljHuOgp0wMxGiXbq+dcj2t5o9F3AMJ5lwfS5mzhvNIieGdOjGiyORx/HuAXwLOxg/6IhhP1usbZ1NXYyrPEMiN3icg0i8juDnvsiMD0wITAJBgUrDgMCGgUABBTQs2HzR+FLxtn2YexYPyg+nzXREAQUj8/6NHfuC4w2C2cblAttX3rFlzcCAgQA"},

"mapping":[{"deviceNodePattern":"Channel1\\.Device\\d+$","deviceNamePattern":"Device ${_System._DeviceId}","attributes":[{"key":"Tag1","type":"string","value":"${Tag1}"}],"timeseries":[{"key":"Tag2","type":"long","value":"${Tag2}"}]}]}]}

14:23:33.980 [pool-3-thread-1] ERROR o.t.g.e.opc.DefaultOpcUaService - OPC-UA service configuration failed!
com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException: Unrecognized field "fileName" (class org.thingsboard.gateway.util.KeystoreConfiguration), not marked as ignorable (6 known properties: "alias", "location", "type", "password", "fileContent", "keyPassword"])
 at [Source: N/A; line: -1, column: -1] (through reference chain: org.thingsboard.gateway.extensions.opc.conf.OpcUaConfiguration["servers"]->java.util.ArrayList[0]->org.thingsboard.gateway.extensions.opc.conf.OpcUaServerConfiguration["keystore"]->org.thingsboard.gateway.util.KeystoreConfiguration["fileName"])
	at com.fasterxml.jackson.databind.exc.UnrecognizedPropertyException.from(UnrecognizedPropertyException.java:62)
	at com.fasterxml.jackson.databind.DeserializationContext.handleUnknownProperty(DeserializationContext.java:834)
	at com.fasterxml.jackson.databind.deser.std.StdDeserializer.handleUnknownProperty(StdDeserializer.java:1093)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownProperty(BeanDeserializerBase.java:1477)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.handleUnknownVanilla(BeanDeserializerBase.java:1455)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:282)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:140)
	at com.fasterxml.jackson.databind.deser.SettableBeanProperty.deserialize(SettableBeanProperty.java:499)
	at com.fasterxml.jackson.databind.deser.impl.MethodProperty.deserializeAndSet(MethodProperty.java:101)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:276)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:140)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:287)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:259)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:26)
	at com.fasterxml.jackson.databind.deser.SettableBeanProperty.deserialize(SettableBeanProperty.java:499)
	at com.fasterxml.jackson.databind.deser.impl.MethodProperty.deserializeAndSet(MethodProperty.java:101)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.vanillaDeserialize(BeanDeserializer.java:276)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:140)
	at com.fasterxml.jackson.databind.ObjectMapper._readValue(ObjectMapper.java:3770)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2099)
	at com.fasterxml.jackson.databind.ObjectMapper.treeToValue(ObjectMapper.java:2596)
	at org.thingsboard.gateway.util.ConfigurationTools.readConfiguration(ConfigurationTools.java:39)
	at org.thingsboard.gateway.extensions.opc.DefaultOpcUaService.init(DefaultOpcUaService.java:54)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:81)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttHandlerImpl.updateConfiguration(MqttHandlerImpl.java:164)
	at org.thingsboard.gateway.service.gateway.MqttHandlerImpl.onGatewayAttributesGet(MqttHandlerImpl.java:145)
	at org.thingsboard.gateway.service.gateway.MqttHandlerImpl.lambda$onMessage$0(MqttHandlerImpl.java:88)
	at java.util.concurrent.Executors$RunnableAdapter.call$$$capture(Executors.java:511)
[... cut ...]
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2099)
	at com.fasterxml.jackson.databind.ObjectMapper.treeToValue(ObjectMapper.java:2596)
	at org.thingsboard.gateway.util.ConfigurationTools.readConfiguration(ConfigurationTools.java:39)
	at org.thingsboard.gateway.extensions.opc.DefaultOpcUaService.init(DefaultOpcUaService.java:54)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:81)
	... 11 common frames omitted`

BIT-masterpy, 2018-04-17:
**tb-gateway.yml**

[image]

**opc-config.json**
[image]

mp-loki, 2018-04-17:
You are using **remoteConfiguration: true** which means that **opc-config.json** is ignored and the configuration that is provided in GUI (On the Devices -> Gateway -> Extensions tab) will be used instead.
The problem is that ThingsBoard v1.4.0 Has a software issue in OPC-UA Extension configuration GUI, which was fixed in master branch and will be available in v1.4.1.
So as a workaround I recommend you to either stick to **remoteConfiguration: false** or build ThingsBoard 1.4.1-SNAPSHOT from source

BIT-masterpy, 2018-04-18:
@mp-loki    Thank you.I reconfigured the file, but it was still wrong, and I couldn't connect the third-party opc-ua server.Is IOT-gateway only developed for KEPServerEX?Not compatible with other opc-ua servers?

`22:56:34.805 [nioEventLoopGroup-2-1] WARN  i.n.channel.DefaultChannelPipeline - An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
**java.lang.NullPointerException: null**
	at nl.jk5.mqtt.MqttPingHandler.channelRead(MqttPingHandler.java:34)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.handler.timeout.IdleStateHandler.channelRead(IdleStateHandler.java:286)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:310)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:297)
	at io.netty.handler.codec.ReplayingDecoder.callDecode(ReplayingDecoder.java:350)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:265)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1414)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:945)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:146)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:645)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:580)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:497)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:459)
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:886)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.lang.Thread.run(Thread.java:748)
22:56:34.941 [main] INFO  o.t.g.e.opc.OpcUaServerMonitor - Initializing OPC-UA server connection to [D-ZHANGCHENG:12686]!
**22:56:35.566 [main] ERROR o.t.gateway.util.ConfigurationTools - Keystore configuration: [KeystoreConfiguration(type=PKCS12, location=example.pfx, fileContent=, password=secret, alias=gateway, keyPassword=password)] is invalid!**
**java.security.UnrecoverableKeyException: Get Key failed: Given final block not properly padded**

Caused by: javax.crypto.BadPaddingException: Given final block not properly padded`

BIT-masterpy, 2018-04-19:
@mp-loki 
opc-config.json:
[image]

tb-gateway.yml:
[image]
**opc-ua server log:**
[image]
Address space:
[image]
[image]
[image]

**Opc-ua server milo has been successfully connected with gateway,
But there's no telemetry on the Thingsboard!**

mp-loki, 2018-04-19:
That's because you still have the mapping from KepServer example.
Set deviceNodePattern to match your path:
```json
deviceNodePattern: "Objects\\.HelloWorld\\.Dynamic"
`

and then change attributes or telemetry to retrieve the tags that you want, e.g. ${Boolean}, ${Double}, ${Int32} in your case

BIT-masterpy, 2018-04-19:
@mp-loki 

[image]

10:56:57.259 [main] ERROR o.t.gateway.util.ConfigurationTools - **Failed to load class org.thingsboard.gateway.extensions.opc.conf.OpcUaConfiguration configuration from opc-config.json**
10:56:57.262 [main] ERROR o.t.g.e.opc.DefaultOpcUaService - **OPC-UA service configuration failed!**
com.fasterxml.jackson.databind.JsonMappingException: **Unexpected character ('/' (code 47)): maybe a (non-standard) comment? (not recognized as one since Feature 'ALLOW_COMMENTS' not enabled for parser)**
 at [Source: java.io.BufferedInputStream@da28d03; line: 33, column: 12]
 at [Source: java.io.BufferedInputStream@da28d03; line: 32, column: 11] (through reference chain: org.thingsboard.gateway.extensions.opc.conf.OpcUaConfiguration["servers"]->java.util.ArrayList[0]->org.thingsboard.gateway.extensions.opc.conf.OpcUaServerConfiguration["mapping"]->java.util.ArrayList[0])

BIT-masterpy, 2018-04-19:
[image]
There is no error reporting.
But there's no telemetry on the Thingsboard

BIT-masterpy, 2018-04-19:
**opc-config.json
What's wrong with my configuration?The panel has no telemetry data.**
`{
  "servers": [
    {
      "applicationName": "Thingsboard OPC-UA client",
      "applicationUri": "",
      "host": "192.168.56.1",
      "port": 12686,
      "scanPeriodInSeconds": 10,
      "timeoutInMillis": 5000,
      "security": "None",
      "identity": {
        "type": "anonymous"
      },
      "keystore": {
        "type": "PKCS12",
        "location": "example.pfx",
        "fileContent": "",
        "password": "secret",
        "alias": "gateway",
        "keyPassword": "secret"
      },
      "mapping": [
        {
          "deviceNodePattern":  "Objects\\.HelloWorld\\.Dynamic",
          "deviceNamePattern": "Device1",
           "attributes":[
             {
               "key": "Boolean",
               "type": "boolean",
               "value":"${Boolean}"

             }
           ],

          "timeseries": [
            {
              "key": "Double Val",
              "type":"double",
              "value":"${Double}"
            }

          ]
        }
      ]
    }
  ]
}`

BIT-masterpy, 2018-04-19:
**@mp-loki 
What's wrong with my configuration?The panel has no telemetry data.
opc-config.json**

[image]

[image]

BIT-masterpy, 2018-04-19:
Thingsboard  log：
[image]

BIT-masterpy, 2018-04-23:
@mp-loki 
What might be the problem?

closed 2018-05-04 by BIT-masterpy

xkdguolei, 2018-11-19:
@ mp-loki
I tried to use the gateway to connect to OpenOpcUa server with the following error
tb-gateway v2.1.1  thingsboard-windows-2.1.3

server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"

gateways:
  tenants:
    -
      label: "Tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: "${GATEWAY_HOST:192.168.1.150}"
        port: 1883
        retryInterval: 3000
        maxInFlight: 1000
        security:
          accessToken: "${GATEWAY_ACCESS_TOKEN:dHRdHsT58CBDJy9OXBip}"
      remoteConfiguration: true
      extensions:
        -
          id: "opc-ua"
          type: "OPC UA"
          extensionConfiguration: opc-config.json

[{"id":"opc","type":"OPC UA","configuration":{"servers":[{"applicationName":"Thingsboard OPC-UA client","applicationUri":"","host":"localhost","port":49320,"scanPeriodInSeconds":10,"timeoutInMillis":5000,"security":"Basic128Rsa15","identity":{"type":"anonymous"},"keystore":{"type":"PKCS12","location":"example.pfx","password":"secret","alias":"gateway","keyPassword":"secret"},"mapping":[{"deviceNodePattern":"Channel1\\.Device\\d+$","deviceNamePattern":"Device ${_System._DeviceId}","attributes":[{"key":"Tag1","type":"string","value":"${Tag1}"}],"timeseries":[{"key":"Tag2","type":"long","value":"${Tag2}"}]}]}]}}]

OPC SERVER
[image]

[image]

[image]
[image]

mp-loki, 2018-11-19:
@xkdguolei  What was the error though?
Try changing remoteConfiguration to 'false'

xkdguolei, 2018-11-19:
@ mp-loki
[image]
[image]
[image]
[image]
[image]

There is no error reporting.
But there's no telemetry on the Thingsboard

mp-loki, 2018-11-19:
Ok, please disregard my previous comment. 
Please provide your tb-gateway.log file

xkdguolei, 2018-11-19:
@ mp-loki
2018-11-19 18:08:40,759 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Registering beans for JMX exposure on startup
2018-11-19 18:08:40,774 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Bean with name 'integrationMbeanExporter' has been autodetected for JMX exposure
2018-11-19 18:08:40,781 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Located managed bean 'integrationMbeanExporter': registering with JMX server as MBean [org.springframework.integration.monitor:name=integrationMbeanExporter,type=IntegrationMBeanExporter]
2018-11-19 18:08:40,862 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering beans for JMX exposure on startup
2018-11-19 18:08:40,862 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageChannel nullChannel
2018-11-19 18:08:40,872 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Located managed bean 'org.springframework.integration:type=MessageChannel,name=nullChannel': registering with JMX server as MBean [org.springframework.integration:type=MessageChannel,name=nullChannel]
2018-11-19 18:08:40,892 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageChannel errorChannel
2018-11-19 18:08:40,902 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Located managed bean 'org.springframework.integration:type=MessageChannel,name=errorChannel': registering with JMX server as MBean [org.springframework.integration:type=MessageChannel,name=errorChannel]
2018-11-19 18:08:41,046 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageHandler errorLogger
2018-11-19 18:08:41,048 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Located managed bean 'org.springframework.integration:type=MessageHandler,name=errorLogger,bean=internal': registering with JMX server as MBean [org.springframework.integration:type=MessageHandler,name=errorLogger,bean=internal]
2018-11-19 18:08:41,105 [main] INFO  o.s.c.s.DefaultLifecycleProcessor - Starting beans in phase 0
2018-11-19 18:08:41,106 [main] INFO  o.s.i.endpoint.EventDrivenConsumer - Adding {logging-channel-adapter:_org.springframework.integration.errorLogger} as a subscriber to the 'errorChannel' channel
2018-11-19 18:08:41,106 [main] INFO  o.s.i.c.PublishSubscribeChannel - Channel 'tb-gateway:9090.errorChannel' has 1 subscriber(s).
2018-11-19 18:08:41,107 [main] INFO  o.s.i.endpoint.EventDrivenConsumer - started _org.springframework.integration.errorLogger
2018-11-19 18:08:41,129 [main] INFO  o.a.coyote.http11.Http11NioProtocol - Initializing ProtocolHandler ["http-nio-0.0.0.0-9090"]
2018-11-19 18:08:41,154 [main] INFO  o.a.coyote.http11.Http11NioProtocol - Starting ProtocolHandler [http-nio-0.0.0.0-9090]
2018-11-19 18:08:41,155 [main] INFO  o.a.tomcat.util.net.NioSelectorPool - Using a shared selector for servlet write/read
2018-11-19 18:08:41,190 [main] INFO  o.s.b.c.e.t.TomcatEmbeddedServletContainer - Tomcat started on port(s): 9090 (http)
2018-11-19 18:08:41,198 [main] INFO  o.t.gateway.GatewayApplication - Started GatewayApplication in 11.663 seconds (JVM running for 13.033)
2018-11-19 18:09:37,902 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1542622177712,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2018-11-19 18:10:37,906 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1542622237714,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2018-11-19 18:10:39,703 [pool-1-thread-2] INFO  o.t.g.s.gateway.MqttGatewayService - Configuration updates arrived! {"configuration":"[{\"id\":\"opc\",\"type\":\"OPC UA\",\"configuration\":{\"servers\":[{\"applicationName\":\"Thingsboard OPC-UA client\",\"applicationUri\":\"opc.tcp://192.168.1.98:49320\",\"host\":\"192.168.1.98\",\"port\":49320,\"scanPeriodInSeconds\":10,\"timeoutInMillis\":5000,\"security\":\"None\",\"identity\":{\"type\":\"anonymous\"},\"keystore\":{\"type\":\"PKCS12\",\"location\":null,\"password\":\"secret\",\"alias\":\"gateway\",\"keyPassword\":\"secret\",\"fileContent\":null},\"mapping\":[{\"deviceNodePattern\":\"Channel1\\\\.Device\\\\d+$\",\"deviceNamePattern\":\"Device ${_System._DeviceId}\",\"attributes\":[{\"key\":\"Tag1\",\"type\":\"string\",\"value\":\"${Tag1}\"}],\"timeseries\":[{\"key\":\"Tag2\",\"type\":\"long\",\"value\":\"${Tag2}\"}]}]}]}}]"}
2018-11-19 18:10:39,704 [pool-1-thread-2] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration
2018-11-19 18:10:39,706 [pool-1-thread-2] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [opc][OPC UA]
2018-11-19 18:10:39,707 [pool-1-thread-2] INFO  o.t.g.e.opc.DefaultOpcUaService - Initializing OPC-UA service!
2018-11-19 18:10:39,707 [pool-1-thread-2] INFO  o.t.g.e.opc.OpcUaServerMonitor - Initializing OPC-UA server connection to [192.168.1.98:49320]!
2018-11-19 18:10:40,769 [pool-1-thread-2] INFO  o.t.g.e.opc.OpcUaServerMonitor - Device scan cycle completed in 443 ms
2018-11-19 18:10:40,776 [pool-1-thread-2] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
2018-11-19 18:10:40,954 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Reported status [Initialized] of extension [opc]
2018-11-19 18:10:51,138 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Device scan cycle completed in 359 ms
2018-11-19 18:10:51,138 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
[... cut ...]
2018-11-19 18:12:03,090 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
2018-11-19 18:12:13,396 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Device scan cycle completed in 305 ms
2018-11-19 18:12:13,397 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
2018-11-19 18:12:23,666 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Device scan cycle completed in 267 ms
2018-11-19 18:12:23,667 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!

mp-loki, 2018-11-20:
can you please set log level to DEBUG and show your log again?

xkdguolei, 2018-11-21:
2018-11-21 11:08:33,978 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Tenant] Initializing gateway
2018-11-21 11:08:35,464 [pool-1-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Configuration arrived! {"shared":{"configuration":"[{\"id\":\"opc\",\"type\":\"OPC UA\",\"configuration\":{\"servers\":[{\"applicationName\":\"Thingsboard OPC-UA client\",\"applicationUri\":\"opc.tcp://192.168.1.98:49320\",\"host\":\"192.168.1.98\",\"port\":49320,\"scanPeriodInSeconds\":10,\"timeoutInMillis\":5000,\"security\":\"None\",\"identity\":{\"type\":\"anonymous\"},\"keystore\":{\"type\":\"PKCS12\",\"location\":\"example.pfx\",\"password\":\"secret\",\"alias\":\"gateway\",\"keyPassword\":\"secret\"},\"mapping\":[{\"deviceNodePattern\":\"Channel1\\\\.Device\\\\d+$\",\"deviceNamePattern\":\"Device ${_System._DeviceId}\",\"attributes\":[{\"key\":\"Tag1\",\"type\":\"string\",\"value\":\"${Tag1}\"}],\"timeseries\":[{\"key\":\"Tag2\",\"type\":\"long\",\"value\":\"${Tag2}\"}]}]}]}}]"}}
2018-11-21 11:08:35,537 [pool-1-thread-1] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration
2018-11-21 11:08:35,655 [pool-1-thread-1] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [opc][OPC UA]
2018-11-21 11:08:35,733 [pool-1-thread-1] INFO  o.t.g.e.opc.DefaultOpcUaService - Initializing OPC-UA service!
2018-11-21 11:08:35,874 [pool-1-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Initializing OPC-UA server connection to [192.168.1.98:49320]!
2018-11-21 11:08:36,577 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1542769715376,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2018-11-21 11:08:36,628 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerAdapter - Looking for @ControllerAdvice: org.springframework.boot.context.embedded.AnnotationConfigEmbeddedWebApplicationContext@28864e92: startup date [Wed Nov 21 11:08:28 CST 2018]; root of context hierarchy
2018-11-21 11:08:36,928 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/sigfox/{deviceTypeId}],methods=[POST]}" onto public void org.thingsboard.gateway.extensions.http.HttpController.handleSigfoxRequest(java.lang.String,java.lang.String,java.lang.String) throws java.lang.Exception
2018-11-21 11:08:36,936 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/uplink/{converterId}],methods=[POST]}" onto public void org.thingsboard.gateway.extensions.http.HttpController.handleRequest(java.lang.String,java.lang.String) throws java.lang.Exception
2018-11-21 11:08:36,959 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/error]}" onto public org.springframework.http.ResponseEntity<java.util.Map<java.lang.String, java.lang.Object>> org.springframework.boot.autoconfigure.web.BasicErrorController.error(javax.servlet.http.HttpServletRequest)
2018-11-21 11:08:36,960 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/error],produces=[text/html]}" onto public org.springframework.web.servlet.ModelAndView org.springframework.boot.autoconfigure.web.BasicErrorController.errorHtml(javax.servlet.http.HttpServletRequest,javax.servlet.http.HttpServletResponse)
2018-11-21 11:08:37,082 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/webjars/**] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-11-21 11:08:37,082 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/**] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-11-21 11:08:37,211 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/**/favicon.ico] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-11-21 11:08:37,900 [main] INFO  o.s.s.c.ThreadPoolTaskScheduler - Initializing ExecutorService  'taskScheduler'
2018-11-21 11:08:38,571 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Registering beans for JMX exposure on startup
2018-11-21 11:08:38,600 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Bean with name 'integrationMbeanExporter' has been autodetected for JMX exposure
2018-11-21 11:08:38,610 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Located managed bean 'integrationMbeanExporter': registering with JMX server as MBean [org.springframework.integration.monitor:name=integrationMbeanExporter,type=IntegrationMBeanExporter]
2018-11-21 11:08:38,816 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering beans for JMX exposure on startup
2018-11-21 11:08:38,819 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageChannel nullChannel
2018-11-21 11:08:38,826 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Located managed bean 'org.springframework.integration:type=MessageChannel,name=nullChannel': registering with JMX server as MBean [org.springframework.integration:type=MessageChannel,name=nullChannel]
2018-11-21 11:08:39,187 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageChannel errorChannel
2018-11-21 11:08:39,188 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Located managed bean 'org.springframework.integration:type=MessageChannel,name=errorChannel': registering with JMX server as MBean [org.springframework.integration:type=MessageChannel,name=errorChannel]
2018-11-21 11:08:39,301 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageHandler errorLogger
[... cut ...]
2018-11-21 11:09:22,802 [pool-6-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
2018-11-21 11:09:33,268 [pool-6-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Device scan cycle completed in 465 ms
2018-11-21 11:09:33,268 [pool-6-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
2018-11-21 11:09:35,417 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1542769775377,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!

xkdguolei, 2018-11-21:
@ mp-loki
2018-11-21 11:08:33,978 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Tenant] Initializing gateway
2018-11-21 11:08:35,464 [pool-1-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Configuration arrived! {"shared":{"configuration":"[{\"id\":\"opc\",\"type\":\"OPC UA\",\"configuration\":{\"servers\":[{\"applicationName\":\"Thingsboard OPC-UA client\",\"applicationUri\":\"opc.tcp://192.168.1.98:49320\",\"host\":\"192.168.1.98\",\"port\":49320,\"scanPeriodInSeconds\":10,\"timeoutInMillis\":5000,\"security\":\"None\",\"identity\":{\"type\":\"anonymous\"},\"keystore\":{\"type\":\"PKCS12\",\"location\":\"example.pfx\",\"password\":\"secret\",\"alias\":\"gateway\",\"keyPassword\":\"secret\"},\"mapping\":[{\"deviceNodePattern\":\"Channel1\\\\.Device\\\\d+$\",\"deviceNamePattern\":\"Device ${_System._DeviceId}\",\"attributes\":[{\"key\":\"Tag1\",\"type\":\"string\",\"value\":\"${Tag1}\"}],\"timeseries\":[{\"key\":\"Tag2\",\"type\":\"long\",\"value\":\"${Tag2}\"}]}]}]}}]"}}
2018-11-21 11:08:35,537 [pool-1-thread-1] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration
2018-11-21 11:08:35,655 [pool-1-thread-1] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [opc][OPC UA]
2018-11-21 11:08:35,733 [pool-1-thread-1] INFO  o.t.g.e.opc.DefaultOpcUaService - Initializing OPC-UA service!
2018-11-21 11:08:35,874 [pool-1-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Initializing OPC-UA server connection to [192.168.1.98:49320]!
2018-11-21 11:08:36,577 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1542769715376,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2018-11-21 11:08:36,628 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerAdapter - Looking for @ControllerAdvice: org.springframework.boot.context.embedded.AnnotationConfigEmbeddedWebApplicationContext@28864e92: startup date [Wed Nov 21 11:08:28 CST 2018]; root of context hierarchy
2018-11-21 11:08:36,928 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/sigfox/{deviceTypeId}],methods=[POST]}" onto public void org.thingsboard.gateway.extensions.http.HttpController.handleSigfoxRequest(java.lang.String,java.lang.String,java.lang.String) throws java.lang.Exception
2018-11-21 11:08:36,936 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/uplink/{converterId}],methods=[POST]}" onto public void org.thingsboard.gateway.extensions.http.HttpController.handleRequest(java.lang.String,java.lang.String) throws java.lang.Exception
2018-11-21 11:08:36,959 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/error]}" onto public org.springframework.http.ResponseEntity<java.util.Map<java.lang.String, java.lang.Object>> org.springframework.boot.autoconfigure.web.BasicErrorController.error(javax.servlet.http.HttpServletRequest)
2018-11-21 11:08:36,960 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/error],produces=[text/html]}" onto public org.springframework.web.servlet.ModelAndView org.springframework.boot.autoconfigure.web.BasicErrorController.errorHtml(javax.servlet.http.HttpServletRequest,javax.servlet.http.HttpServletResponse)
2018-11-21 11:08:37,082 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/webjars/**] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-11-21 11:08:37,082 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/**] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-11-21 11:08:37,211 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/**/favicon.ico] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-11-21 11:08:37,900 [main] INFO  o.s.s.c.ThreadPoolTaskScheduler - Initializing ExecutorService  'taskScheduler'
2018-11-21 11:08:38,571 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Registering beans for JMX exposure on startup
2018-11-21 11:08:38,600 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Bean with name 'integrationMbeanExporter' has been autodetected for JMX exposure
2018-11-21 11:08:38,610 [main] INFO  o.s.j.e.a.AnnotationMBeanExporter - Located managed bean 'integrationMbeanExporter': registering with JMX server as MBean [org.springframework.integration.monitor:name=integrationMbeanExporter,type=IntegrationMBeanExporter]
2018-11-21 11:08:38,816 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering beans for JMX exposure on startup
2018-11-21 11:08:38,819 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageChannel nullChannel
2018-11-21 11:08:38,826 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Located managed bean 'org.springframework.integration:type=MessageChannel,name=nullChannel': registering with JMX server as MBean [org.springframework.integration:type=MessageChannel,name=nullChannel]
2018-11-21 11:08:39,187 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageChannel errorChannel
2018-11-21 11:08:39,188 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Located managed bean 'org.springframework.integration:type=MessageChannel,name=errorChannel': registering with JMX server as MBean [org.springframework.integration:type=MessageChannel,name=errorChannel]
2018-11-21 11:08:39,301 [main] INFO  o.s.i.m.IntegrationMBeanExporter - Registering MessageHandler errorLogger
[... cut ...]
2018-11-21 11:09:22,802 [pool-6-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
2018-11-21 11:09:33,268 [pool-6-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Device scan cycle completed in 465 ms
2018-11-21 11:09:33,268 [pool-6-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Scheduling next scan in 10 seconds!
2018-11-21 11:09:35,417 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1542769775377,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2