thingsboard/thingsboard-gateway #68: Can't build from sources.

pavellzubkov, 2018-04-18:
How fix this?

> $ mvn clean install
[INFO] Scanning for projects...
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building Thingsboard Gateway 1.4.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[WARNING] The POM for org.thingsboard.common:data:jar:1.4.1-SNAPSHOT is missing, no dependency information available
[WARNING] The POM for nl.jk5.netty-mqtt:netty-mqtt:jar:2.0.0TB is missing, no dependency information available
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 3.190 s
[INFO] Finished at: 2018-04-18T21:24:48+04:00
[INFO] Final Memory: 13M/208M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project gateway: Could not resolve dependencies for project org.thingsboard:gateway:jar:1.4.1-SNAPSHOT: The following artifacts could not be resolved: org.thingsboard.common:data:jar:1.4.1-SNAPSHOT, nl.jk5.netty-mqtt:netty-mqtt:jar:2.0.0TB: Could not find artifact org.thingsboard.common:data:jar:1.4.1-SNAPSHOT -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException

mp-loki, 2018-04-18:
Check out and build [Thingsboard](https://github.com/thingsboard/thingsboard) 
and [netty-mqtt](https://github.com/thingsboard/netty-mqtt) source code first

pavellzubkov, 2018-04-19:
Ok.Thank yoy! It build succesfull. But file tb-gateway.exe is missing in windows folder. Can i use file tb-gateway.exe from release archive? ... i try it, but no. 

> Unable to locate tb-gateway.xml file within exe cutable directory or any parents..

mp-loki, 2018-06-27:
@pavellzubkov I assume this can be closed now

closed 2018-06-27 by mp-loki