thingsboard/thingsboard-gateway #69: opcua telemetry data on thingsboard gui

quirosaur, 2018-04-19:
After configuring opc-config.json, i ran into this problem:

if i use 
"deviceNodePattern": "Objects\\.Raspberry\ Pi\\.Application\\d+$",
"deviceNamePattern": "Device ${_System._DeviceId}",

(this should be the correct path)

log gives these messages:

2018-04-19 11:31:02,010 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-04-19 11:31:02,557 [http-nio-0.0.0.0-9090-ClientPoller-1] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524126662557; nextExpiration=1524126662556; keyCount=0; hasEvents=false; eval=false
2018-04-19 11:31:02,557 [http-nio-0.0.0.0-9090-ClientPoller-0] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524126662557; nextExpiration=1524126662556; keyCount=0; hasEvents=false; eval=false
2018-04-19 11:31:03,557 [http-nio-0.0.0.0-9090-ClientPoller-1] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524126663557; nextExpiration=1524126663557; keyCount=0; hasEvents=false; eval=false
2018-04-19 11:31:03,557 [http-nio-0.0.0.0-9090-ClientPoller-0] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524126663557; nextExpiration=1524126663557; keyCount=0; hasEvents=false; eval=false
2018-04-19 11:31:04,558 [http-nio-0.0.0.0-9090-ClientPoller-0] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524126664558; nextExpiration=1524126664557; keyCount=0; hasEvents=false; eval=false
2018-04-19 11:31:04,558 [http-nio-0.0.0.0-9090-ClientPoller-1] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524126664558; nextExpiration=1524126664557; keyCount=0; hasEvents=false; eval=false

if using 
"deviceNodePattern": "Objects",
"deviceNamePattern": "Objects",

log shows this:

2018-04-19 11:34:56,142 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=91}, name=ReferenceTypes, fqn=Types.ReferenceTypes)
2018-04-19 11:34:56,142 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-19 11:34:56,142 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected
2018-04-19 11:34:56,142 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=3345
2018-04-19 11:34:56,143 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=90}, name=DataTypes, fqn=Types.DataTypes)
2018-04-19 11:34:56,143 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-19 11:34:56,143 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected
2018-04-19 11:34:56,143 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=3346
2018-04-19 11:34:56,146 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=87}, name=Views, fqn=Views)
2018-04-19 11:34:56,146 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-19 11:34:56,146 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected
2018-04-19 11:34:56,146 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=3347
2018-04-19 11:34:56,147 [pool-7-thread-1] INFO  o.t.g.e.opc.OpcUaServerMonitor - Device scan cycle completed in 189 ms

Have i understood correctly use of the device node/name patterns, and why can't i see them in telemetry data or devices on web gui?

quirosaur, 2018-04-19:
and can i add "telemetry" to these?

           "attributes": [
            {
              "key": "Tag1",
              "type": "string",
              "value": "${Tag1}"
            }
          ],
          "timeseries": [
            {
              "key": "Tag2",
              "type": "long",
              "value": "${Tag2}"
            }

like this:

            "attributes": [
            {
              "key": "Tag1",
              "type": "string",
              "value": "${Tag1}"
            }
          ],
          "timeseries": [
            {
              "key": "Tag2",
              "type": "long",
              "value": "${Tag2}"
            }
            "telemetry": [
            {
              "key": "Tag3",
              "type": "long",
              "value": "${Tag3}"
            }

BIT-masterpy, 2018-04-19:
I have a similar problem with you, I don't know how to solve it.In addition, I have found that opc and MQTT protocols are not running at the same time.

closed 2018-05-08 by quirosaur