thingsboard/thingsboard-gateway #71: The data in opc-ua server is not collected. The thingsboard panel does not display telemetry data.

BIT-masterpy, 2018-04-24:
**Hello, I use the third-party opc-ua server (milo) and tb-gateway and thingsboard for the test of data collection.Milo and TB -gateway were successfully connected, and tb-gateway and thingsboard were successful, but there was no telemetry data.The log files of the three are as follows:**
@mp-loki  @ashvayka

[opc-server（milo）.txt](https://github.com/thingsboard/thingsboard-gateway/files/1941411/opc-server.milo.txt)
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/1941412/tb-gateway.txt)
[thingsboard.txt](https://github.com/thingsboard/thingsboard-gateway/files/1941415/thingsboard.txt)

[image]

**tb-gateway.yml:**

gateways:
  tenants:
    -
      label: "Tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: localhost
        port: 1882
        retryInterval: 3000
        maxInFlight: 1000
        security:
          accessToken: "4cq1eFC25o6ZpSLkyXec"
      remoteConfiguration: false
      extensions:
              -
                id: "opc-ua"
                type: "OPC UA"
                extensionConfiguration: opc-config.json

**opc-config.json：**
{
  "servers": [
    {
      "applicationName": "Thingsboard OPC-UA client",
      "applicationUri": "",
      "host": "192.168.56.1",
      "port": 12686,
      "scanPeriodInSeconds": 10,
      "timeoutInMillis": 5000,
      "security": "None",
      "identity": {
        "type": "anonymous"
      },
      "keystore": {
        "type": "PKCS12",
        "location": "example.pfx",
        "fileContent": "",
        "password": "secret",
        "alias": "gateway",
        "keyPassword": "secret"
      },
      "mapping": [
        {
          "deviceNodePattern": "Objects\\.HelloWorld\\.Dynamic",
          "deviceNamePattern": "Device1",
           "attributes":[
             {
               "key": "Boolean val",
               "type": "boolean",
               "value":"${Boolean}"
             }
           ],
          "timeseries":[
            {
              "key": "Double Val",
              "type":"double",
              "value":"${Double}"
            }
          ]
        }
      ]
    }
  ]
}

**Milo address space：**
[image]

mp-loki, 2018-04-25:
1. **deviceNodePattern** should have double backslashes:
`"deviceNodePattern": "Objects\\.HelloWorld\\.Dynamic",`
2. delete any gateway logs.
3. Go to logback.xml, set **org.thingsboard* log level to **TRACE**, restart the gateway and post your gateway log here. 
This way we will be able to see if it got connected to OPC-UA server and found the devices or not.

BIT-masterpy, 2018-04-25:
@mp-loki 
Thank you. I have made some changes.The results are as follows:

[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/1945182/tb-gateway.txt)

[image]
[image]
[image]

BIT-masterpy, 2018-04-25:
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/1945196/tb-gateway.txt)

mp-loki, 2018-04-25:
Can you please try to change deviceNodePattern to "HelloWorld/Dynamic" and let me know if it helps?
I think I figured out how the behaviour is different from KEPServer.

mp-loki, 2018-04-25:
So may I assume this issue resolved?

BIT-masterpy, 2018-04-26:
yes

closed 2018-04-26 by mp-loki

liangguorui666, 2019-04-02:
@BIT-masterpy   你好 ，我在解析的时候和你遇到相同的问题  能否告知一下您最后的配置文件的格式。
谢谢了

liangguorui666, 2019-04-02:
opc-config.json 这个文件