thingsboard/thingsboard-gateway #72: opcua data problem

quirosaur, 2018-04-25:
Hi @mp-loki @ashvayka

i contionuously get this in my log

2018-04-25 13:52:37,995 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=2422
2018-04-25 13:52:37,996 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=2264}, name=SoftwareVersion, fqn=Objects.Server.ServerStatus.BuildInfo.SoftwareVersion)
2018-04-25 13:52:37,996 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-25 13:52:37,996 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected
2018-04-25 13:52:37,996 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=2423
2018-04-25 13:52:38,000 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=2265}, name=BuildNumber, fqn=Objects.Server.ServerStatus.BuildInfo.BuildNumber)
2018-04-25 13:52:38,000 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-25 13:52:38,000 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected
2018-04-25 13:52:38,000 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=2424
2018-04-25 13:52:38,001 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=2266}, name=BuildDate, fqn=Objects.Server.ServerStatus.BuildInfo.BuildDate)
2018-04-25 13:52:38,001 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-25 13:52:38,001 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected
2018-04-25 13:52:38,001 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=2425
2018-04-25 13:52:38,003 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=2992}, name=SecondsTillShutdown, fqn=Objects.Server.ServerStatus.SecondsTillShutdown)
2018-04-25 13:52:38,003 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-25 13:52:38,003 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected
2018-04-25 13:52:38,003 [ua-netty-event-loop-1] TRACE o.e.m.o.s.client.UaTcpStackClient - writeAndFlush succeeded for request=BrowseRequest, requestHandle=2426
2018-04-25 13:52:38,005 [pool-7-thread-1] TRACE o.t.g.e.opc.OpcUaServerMonitor - Scanning node: OpcUaNode(nodeId=NodeId{ns=0, id=2993}, name=ShutdownReason, fqn=Objects.Server.ServerStatus.ShutdownReason)
2018-04-25 13:52:38,005 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientSessionManager - getSession(), currentState=Active
2018-04-25 13:52:38,005 [pool-7-thread-1] TRACE o.e.m.o.s.c.ClientChannelManager - getChannel(), currentState=Connected

doesn't matter if i use:

Objects\\.Server
Objects\/Server
Objects/Server

all of those return same log messages:
getSession(), currentState=Active
getChannel(), currentState=Connected
writeAndFlush succeeded for request=BrowseRequest, requestHandle=2425
Scanning node: OpcUaNode.....

but i never get that data to thingsboard. What am i doing wrong?

-Quirosaur

mp-loki, 2018-04-25:
Currently gateway maps on ID instead of fqn (which it should do), so as soon as you have your IDs as 4-digit integer, you can create a regex to map on that number, like:

**"2.\\d{3}"** this will match all devices with 4-digit IDs starting with 2, like: 2XXX. You may adjust the regex per your needs.

quirosaur, 2018-04-26:
part of my node id's are only text. does this mean that i cannot read them?

mp-loki, 2018-04-26:
The Gateway has to map on Path of the object on server (using fully qualified name). e.g. **Objects.Server.Device**. Currently there is a software issue in Gateway which maps device on ID, instead of path (fqn). This will be fixed soon. As a workaround, you can come up with regular expressions with which will match your ID pattern. If you have more than one ID pattern, you can add multiple mappings to match them. If you need help coming up with the right regex, please provide the common examples of your Device IDs and I'll help you out.

closed 2018-04-27 by quirosaur