thingsboard/thingsboard-gateway #73: ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds

quirosaur, 2018-04-25:
```
2018-04-25 15:55:32,601 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-04-25 15:55:32,750 [http-nio-0.0.0.0-9090-ClientPoller-0] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524660932750; nextExpiration=1524660932748; keyCount=0; hasEvents=false; eval=false
2018-04-25 15:55:32,750 [http-nio-0.0.0.0-9090-ClientPoller-1] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1524660932750; nextExpiration=1524660932749; keyCount=0; hasEvents=false; eval=false
```

[thingsboard.yml.txt](https://github.com/thingsboard/thingsboard-gateway/files/1947091/thingsboard.yml.txt)
[tb-gateway.log.txt](https://github.com/thingsboard/thingsboard-gateway/files/1947092/tb-gateway.log.txt)
[tb-gateway.yml.txt](https://github.com/thingsboard/thingsboard-gateway/files/1947093/tb-gateway.yml.txt)
[mqtt-config.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/1947094/mqtt-config.json.txt)

mp-loki, 2018-04-25:
1) Are you able to connect to ThingsBoard using some other MQTT client, like **mosquitto_pub**? Just to check that the host, port & token are correct.
2) in your tb-gateway/conf/logback.xml, can you please set **org.thingsboard** log level to **DEBUG**, and leave the rest as **INFO** because your log file is cluttered with a lots of unrelated messages which are hard to read
3) Are there any logs on ThingsBoard side when the gateway tries to connect?

quirosaur, 2018-04-26:
hi! thanks for reply :) 

i tested mosquitto_pub, and it worked. (sent telemetry data to thingsboard device which was not gateway)

however, tb-gateway to gateway device connection doesn't work. i'll link the logs.

in thingsboard log, lowest entry is from mosquitto_pub and above about in 
2018-04-26 11:32:43,214 timestamp is the connection try of gateway
[thingsboard.log.txt](https://github.com/thingsboard/thingsboard-gateway/files/1950563/thingsboard.log.txt)
[tb-gateway.log.txt](https://github.com/thingsboard/thingsboard-gateway/files/1950564/tb-gateway.log.txt)

quirosaur, 2018-04-26:
i also managed to send telemetry data to my device in thingsboard from my java application (kind of opcua to mqtt adapter). So now i quess the question is: can i dynamically create new devices to thingsboard through mqtt?

mp-loki, 2018-04-26:
There is a syntax error in your mqtt-config.json on lines 50-51
`],
[`
should be just `,`

quirosaur, 2018-04-27:
thanks! also, i noticed that my filte expression missed this   --->  )

closed 2018-04-27 by quirosaur