thingsboard/thingsboard-gateway #74: maven compile erro

jacklicn, 2018-04-25:
[ERROR] thingsboard-gateway-1.4.0.1\src\main\java\org\
thingsboard\gateway\service\MqttMessageSender.java:[131,21] Error: Cannot find symbol
[ERROR]  Symbol: Method isConnected()
[ERROR]   Location: Type is MqttClient Variables tbClient
----------------------------------------------------------------------------
Is this project used https://github.com/jk-5/netty-mqtt in？
but the interface MqttClient does not have the method isConnected

jacklicn, 2018-04-25:
use https://github.com/thingsboard/netty-mqtt

closed 2018-04-25 by jacklicn

Tanushree1989, 2021-04-29:
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.8.1:compile (default-compile) on project data: Compilation failure: Compilation failure: 
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/device/profile/AlarmCondition.java:[29,18] cannot find symbol
[ERROR]   symbol:   class AlarmConditionFilter
[ERROR]   location: class org.thingsboard.server.common.data.device.profile.AlarmCondition
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/device/profile/AlarmCondition.java:[29,40] cannot find symbol
[ERROR]   symbol:   class AlarmConditionFilter
[ERROR]   location: class org.thingsboard.server.common.data.device.profile.AlarmCondition
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/device/profile/AlarmCondition.java:[25,1] cannot find symbol
[ERROR]   symbol:   class AlarmConditionFilter
[ERROR]   location: class org.thingsboard.server.common.data.device.profile.AlarmCondition
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/query/RelationsQueryFilter.java:[21,51] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: package org.thingsboard.server.common.data.relation
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/query/RelationsQueryFilter.java:[35,18] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: class org.thingsboard.server.common.data.query.RelationsQueryFilter
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/query/RelationsQueryFilter.java:[35,44] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: class org.thingsboard.server.common.data.query.RelationsQueryFilter
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/query/RelationsQueryFilter.java:[25,1] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: class org.thingsboard.server.common.data.query.RelationsQueryFilter
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/entityview/EntityViewSearchQuery.java:[22,51] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: package org.thingsboard.server.common.data.relation
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/relation/EntityRelationsQuery.java:[29,18] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: class org.thingsboard.server.common.data.relation.EntityRelationsQuery
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/relation/EntityRelationsQuery.java:[29,44] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: class org.thingsboard.server.common.data.relation.EntityRelationsQuery
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/relation/EntityRelationsQuery.java:[25,1] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: class org.thingsboard.server.common.data.relation.EntityRelationsQuery
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/asset/AssetSearchQuery.java:[22,51] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: package org.thingsboard.server.common.data.relation
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/device/DeviceSearchQuery.java:[22,51] cannot find symbol
[ERROR]   symbol:   class RelationEntityTypeFilter
[ERROR]   location: package org.thingsboard.server.common.data.relation
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/widget/WidgetType.java:[23,33] cannot find symbol
[ERROR]   symbol: class BaseWidgetType
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/widget/WidgetType.java:[35,23] cannot find symbol
[ERROR]   symbol:   class BaseWidgetType
[ERROR]   location: class org.thingsboard.server.common.data.widget.WidgetType
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/device/profile/DeviceProfileTransportConfiguration.java:[34,37] cannot find symbol
[ERROR]   symbol: class CoapDeviceProfileTransportConfiguration
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/device/data/DeviceTransportConfiguration.java:[33,36] cannot find symbol
[ERROR]   symbol: class CoapDeviceTransportConfiguration
[ERROR] /var/jenkins_home/workspace/thingsboard/common/data/src/main/java/org/thingsboard/server/common/data/query/EntityFilter.java:[32,36] cannot find symbol
[ERROR]   symbol: class EntityTypeFilter
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[... cut ...]
I have Jenkins installed on google kubernetes cluster. I selected autometic installation option from global tool configuration to installed java, node and maven that requires to build the project. But for git i provided the path /usr/bin/git which I am not sure is the correct one

java -version
openjdk version "11.0.10" 2021-01-19
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.10+9)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 11.0.10+9, mixed mode)
+ git --version
git version 2.20.1
+ mvn -version
Apache Maven 3.8.1

OS name: "linux", version: "5.4.89+", arch: "amd64", family: "unix"

Kindly assist.