thingsboard/thingsboard-gateway #75: rpc subscription to /v1/gatway/rpc not logged

marcozanghi, 2018-04-26:
The subscription to /v1/gateway/rpc seems not to be performed ! How to check it?

marcozanghi, 2018-04-27:
I have changed the thingsboard version in the pom file from 1.4.1-SNAPSHOT into 1.4.0 for mvn packaging. Are there any conflicts between these releases?

marcozanghi, 2018-04-27:
When I start for the first time the tb-gateway I receive rpc command from server only one time. After that it starts to log this message continuously :
`2018-04-27 14:47:44,837 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-04-27 14:50:06,912 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
`
No other incoming rpc message from the server.

mp-loki, 2018-04-27:
I did not mean you to change version in log file. Let's emphasize:
1) Are you installing ThingsBoard and tb-gateway as a service or running from IDE?
2) If you install as a service, are you building from source or using release packages? Which version?

P.S. The `Waiting until all messages are sent before going to the next bucket` message is ok, there are no issues with that.

marcozanghi, 2018-04-30:
Using tb-gateway from release-1.4 the all telemetry messages have been sent correctly. Can I use tb-gateway from master branch and thingsboard from 1.4 ? Are these versions incompatible?

mp-loki, 2018-04-30:
There should be no issue running tb-gateway v.1.4.1 with TB v.1.4.0 and vice versa

closed 2018-06-27 by mp-loki