thingsboard/thingsboard-gateway #77: Gateway not publish data via Modbus extension or TB can't display it.

pavellzubkov, 2018-04-29:
I'm trying to test chain modbus slave- tbgateway-tb, but something going wrong. Steps:

- connect tbgateway to tb -OK
- configure modbus extension via tb GUI -OK
- tbgateway successfuly pooling modbus slave -OK
- in TB appeared device configured on gateway -OK
**but timeseries and attributes don't update.**
in gateway telemetry i see attributes uploaded -0; devices online -0 ; telemetry uploaded -0;
In gateway log looks like all ok.
[tb-gateway.log](https://github.com/thingsboard/thingsboard-gateway/files/1958577/tb-gateway.log)
In thingsboard log i see WARN logs like:
` o.t.s.t.mqtt.MqttTransportHandler - [mqtt19] Failed to subscribe to [v1/gateway/rp               c][AT_MOST_ONCE]
`
and 

` o.t.s.t.mqtt.MqttTransportHandler - [mqtt19] Failed to process publish msg [v1/gateway/connect][7]
`

dyosick, 2018-04-29:
Are there any error or warning messages in gateway logs regarding to Modbus read failures or something similar?

pavellzubkov, 2018-04-29:
No. In gateway logs all ok no errors and no warnings . I don't know where the problem and what to do.)  May be this issue not modbus extension issue.

dyosick, 2018-04-29:
Can you see from gateway logs what Modbus extension gets new values while periodically polling slaves?

pavellzubkov, 2018-04-29:
About modbus in logs i can see only next:

> 2018-04-29 12:21:07,791 [pool-1-thread-1] INFO  o.t.g.extensions.modbus.ModbusClient - MBS[COM36] connected
2018-04-29 12:21:07,792 [pool-1-thread-1] INFO  o.t.g.extensions.modbus.ModbusClient - MBS[COM36] going to first read
2018-04-29 12:21:07,849 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerAdapter - Looking for @ControllerAdvice: org.springframework.boot.context.embedded.AnnotationConfigEmbeddedWebApplicationContext@a6d14e: startup date [Sun Apr 29 12:20:53 GET 2018]; root of context hierarchy
2018-04-29 12:21:08,089 [pool-1-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [dev1] Device Connected!
2018-04-29 12:21:08,171 [pool-1-thread-1] INFO  o.t.g.extensions.modbus.ModbusClient - MBS[COM36] connected new MBD[dev1]
2018-04-29 12:21:08,185 [pool-1-thread-1] INFO  o.t.g.extensions.modbus.ModbusClient - Start polling MBS[COM36]
2018-04-29 12:21:08,189 [pool-1-thread-1] INFO  o.t.g.extensions.modbus.ModbusClient - MBS[COM36] Schedule polling [dev1] slave for tag(s) [atr2,value,max,min,almax], period 2000 ms
2018-04-29 12:21:08,191 [pool-1-thread-1] INFO  o.t.g.extensions.modbus.ModbusClient - MBS[COM36] Schedule polling [dev1] slave for tag(s) [atr1], period 5000 ms

dyosick, 2018-04-29:
What max log level is set up? To better understanding I recommend TRACE

dyosick, 2018-04-29:
If some value changes, there will be log like
- for an attribute: MBD[{}] attribute update: key '{}', val '{}'
- for timeseries: MBD[{}] timeseries update: key '{}', val '{}'

closed 2018-06-27 by mp-loki