thingsboard/thingsboard-gateway #78: ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds

quirosaur, 2018-05-02:
````
2018-05-02 14:22:08,331 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:11,339 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:14,350 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:17,358 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:20,367 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:23,376 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:26,386 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:29,396 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:32,406 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:35,417 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:38,425 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:41,436 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:44,444 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:47,455 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:50,464 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:53,473 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:56,483 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:22:59,496 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:23:02,504 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:23:05,514 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-05-02 14:23:08,524 [pool-4-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds

´´´´

i'm having this problem with gateway.
[tb-gateway.out.log](https://github.com/thingsboard/thingsboard-gateway/files/1967014/tb-gateway.out.log)
[thingsboard.log](https://github.com/thingsboard/thingsboard-gateway/files/1967015/thingsboard.log)
[tb-gateway.yml.txt](https://github.com/thingsboard/thingsboard-gateway/files/1967030/tb-gateway.yml.txt)
[mqtt-config.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/1967031/mqtt-config.json.txt)
[thingsboard.yml.txt](https://github.com/thingsboard/thingsboard-gateway/files/1967032/thingsboard.yml.txt)

mp-loki, 2018-05-02:
1. Your tb-gateway.out.log contains several days of logs, but it does not have the lines that you mentioned. Please clean up your logs before capturing and sending them here.
2. the gateway log file is supposed to be tb-gateway.log. 
3. In ThingsBoard I see `constraint [TS_KV_UNQ_KEY]; nested exception is org.hibernate.exception.ConstraintViolationException: could not execute statement`. We had this issue before and I think there was a fix before 1.4.0. So the question is - which ThingsBoard version are you using?

quirosaur, 2018-05-02:
[tb-gateway.log](https://github.com/thingsboard/thingsboard-gateway/files/1967131/tb-gateway.log)

oops, accidentally uploaded wrong log file.

my gateway and thingsboard are 1.4.0

quirosaur, 2018-05-03:
solved. i did my mqtt-config.json again. apparently i had too much or too little '{' or'}'

closed 2018-05-03 by quirosaur