thingsboard/thingsboard-gateway #79: Failed to connect to external modbus TCP server. TB has no display data.

BIT-masterpy, 2018-05-07:
Hello, I tried to connect to the external modbus TCP server with tb-gateway, and tb-gateway and TB did not report wrong, but the data was never collected.

**tb-gateway.log:**
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/1979740/tb-gateway.txt)

**TB.log:**
[Thingsboard.txt](https://github.com/thingsboard/thingsboard-gateway/files/1979744/Thingsboard.txt)

**modbus-config.json:**
[image]

**Modbus Slave:**
[image]

dyosick, 2018-05-07:
@BIT-masterpy As I see, you use a local configuration but did you disable a remote configuration in tb-gateway.yml? Because I cannot find any logs from Modbus extension and it seems that the gateway log file is not full.

Also to map a boolean type for Modbus functions 3 and 4 you must specify [the bit field](https://thingsboard.io/docs/iot-gateway/modbus/#data-conversion-rules). See the second example [there](https://thingsboard.io/docs/iot-gateway/modbus/#examples).

BIT-masterpy, 2018-05-08:
@dyosick 

**modbus-config.json:**
`{
  "servers": [
    {
      "transport":
      {
        "type": "tcp",
        "host": "localhost",
        "port": 502
      },
      "devices": [
        {
          "unitId": 1,
          "deviceName": "Temp Sensor",
          "timeseries": [
            {
              "tag": "ee",
              "type": "long",
              "functionCode": 3,
              "address": 1
            },
            {
              "tag": "qq",
              "type": "long",
              "functionCode": 3,
              "address": 2
            },
            {
              "tag": "rr",
              "type": "long",
              "functionCode": 3,
              "address": 3,
              "bit": 14,
              "byteOrder": "little"
            }
          ]
        },
        {
          "unitId": 2,
          "deviceName": "Humidity Sensor",
          "attributesPollPeriod": 5000,
          "timeseriesPollPeriod": 5000,
          "attributes": [
            {
              "tag": "tt",
              "type": "string",
              "functionCode": 3,
              "address": 4,
              "registerCount": 6,
              "pollPeriod": 3600000
            }
          ],
          "timeseries": [
            {
              "tag": "uu",
              "type": "long",
              "functionCode": 3,
              "address": 5,
              "byteOrder": "LITTLE"
            },
            {
              "tag": "ii",
              "type": "long",
              "functionCode": 3,
              "address": 6,
              "registerCount": 2,
              "byteOrder": "BADC"
            },
            {
              "tag": "zz",
              "type": "long",
              "functionCode": 3,
              "address": 7,
              "registerCount": 4,
              "byteOrder": "AB CD EF GH"
            },
            {
              "tag": "vv",
              "type": "double",
              "byteOrder": "2301",
              "functionCode": 3,
              "address": 8,
              "registerCount": 2
            },
            {
              "tag": "oo",
              "type": "double",
              "byteOrder": "23 01 67 45",
              "functionCode": 3,
              "address": 9,
              "registerCount": 4
            }
          ]
        }
      ]
    }
  ]
}`

**tb-gateway.yml:**
`remoteConfiguration: false
      extensions:
        -
          id: "modbus"
          type: "modbus"
          extensionConfiguration: modbus-config.json`

**tb-gateway.log:**
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/1981984/tb-gateway.txt)

dyosick, 2018-05-08:
The extension type (in tb-gaateway. yml) must be in upper case, i. e. MODBUS.

Also the bit field is only for the boolean type   and Modbus functions 3 and 4 (see the 'RR tag).

closed 2018-06-27 by mp-loki