thingsboard/thingsboard-gateway #80: I want to use individual KVs on extend MQTT

kaizen8501, 2018-05-14:
Hi~ 
This question is related with #40

I also operate to send individual KVs like as below.

- Topic : v1/usr/openhouse/down
- Payload : {"cmdId":11809,"humidity":["2018-05-14 13:05:53",35.2]}

- Topic : v1/usr/openhouse/down
- Payload : {"cmdId":11809,"temperature":["2018-05-14 13:05:44",33.2]
 

So I writed mqtt-config.json as below.

```
  "brokers": [
    {   
      "host": "test.sktiot.com",
      "port": 1883,
      "ssl": false,
      "retryInterval": 3000,
      "credentials": {
        "type": "basic",
        "username": "xxxx",
        "password": "xxxx"
      },  
      "mapping": [
        {
          "topicFilter": "v1/usr/openhouse/down",
          "converter": {
            "type": "json",
            "filterExpression": "$[?(@.humidity)]",
            "deviceNameJsonExpression": "${$.cmdId}",
            "deviceTypeJsonExpression": "SF_TempHumi",
            "attributes": [
            ],
            "timeseries": [
              {
                "type": "double",
                "key": "humidity",
                "value": "${$.humidity[1]}"
              }
            ]
          }
        },
        {
          "topicFilter": "v1/usr/openhouse/down",
          "converter": {
            "type": "json",
            "filterExpression": "$[?(@.temperature)]",
            "deviceNameJsonExpression": "${$.cmdId}",
            "deviceTypeJsonExpression": "SF_TempHumi",
            "timeseries": [
              {
                "type": "double",
                "key": "temperature",
                "value": "${$.temperature[1]}"
              }
            ]
          }
        }
      ],
      ...
      ...
```
```
[MQTT Call: d5681eba-9f33-4148-8939-354ece59e6e2] INFO  o.t.g.e.m.c.c.m.MqttJsonConverter - kaizen json message: {"cmdId":11809,"temperature":["2018-05-14 13:25:27",33.2]}
[MQTT Call: d5681eba-9f33-4148-8939-354ece59e6e2] INFO  o.t.g.e.m.c.c.m.MqttJsonConverter - kaizen Data before filtering {"cmdId":11809,"temperature":["2018-05-14 13:25:27",33.2]}
[MQTT Call: d5681eba-9f33-4148-8939-354ece59e6e2] INFO  o.t.g.e.m.c.c.m.MqttJsonConverter - kaizen Data after filtering [{"cmdId":11809,"temperature":["2018-05-14 13:25:27",33.2]}], $[?(@.temperature)

[MQTT Call: d5681eba-9f33-4148-8939-354ece59e6e2] INFO  o.t.g.e.m.c.l.MqttTelemetryMessageListener - v1/usr/openhouse/down,{"cmdId":11809,"humidity":["2018-05-14 13:05:53",35.2]}
[MQTT Call: d5681eba-9f33-4148-8939-354ece59e6e2] INFO  o.t.g.e.m.c.c.m.MqttJsonConverter - json message: {"cmdId":11809,"humidity":["2018-05-14 13:05:53",35.2]}
[MQTT Call: d5681eba-9f33-4148-8939-354ece59e6e2] INFO  o.t.g.e.m.c.c.m.MqttJsonConverter - Data before filtering {"cmdId":11809,"humidity":["2018-05-14 13:05:53",35.2]}
[MQTT Call: d5681eba-9f33-4148-8939-354ece59e6e2] INFO  o.t.g.e.m.c.c.m.MqttJsonConverter - Data after filtering [], **$[?(@.temperature)]**
```

In that case, I success to send temperature data but humidity doesn't send to ThingsBoard because it was filtered. So I think It has some problem such as TB-Gateway only use last "filterExpression".
Is it right? or Is there anything I'm doing wrong?

I used TB-Gateway 1.4.1 latest version and I just added log file.
Anyone please help me and give a guide.

Thank you.

quirosaur, 2018-05-17:
i have same kind of problem in tb-gateway 1.4.1, where my second filter expression seems to be ignored, and only first is applied to incoming data.

i try to test this on 1.4.0, but that branch is impossible to build. gateway needs thingsboard 1.4 to be built, and thingsboard 1.4 build gets stuck with these:
````
2018-05-17 15:39:53,253 [Akka-akka.actor.default-dispatcher-5] INFO  akka.event.slf4j.Slf4jLogger - Slf4jLogger started
2018-05-17 15:39:53,258 [main] ERROR o.t.s.s.c.rpc.ClusterGrpcService - Failed to start RPC server!
````

i'll let you know if i get this working

rof20004, 2018-07-07:
I am with same problem.

rof20004, 2018-07-14:
@quirosaur I resolved like this:

I created only one entrie for the two values: humidity and temperature, like this:

```
{
          "topicFilter": "v1/usr/openhouse/down",
          "converter": {
            "type": "json",
            "filterExpression": "",
            "deviceNameJsonExpression": "${$.cmdId}",
            "deviceTypeJsonExpression": "SF_TempHumi",
            "timeseries": [
              {
                "type": "double",
                "key": "humidity",
                "value": "${$.humidity[1]}"
              },
              {
                "type": "double",
                "key": "temperature",
                "value": "${$.temperature[1]}"
              }
            ]
          }
        }
```

Then I modified the source code in this class BasicJsonConverter in method getTsKvEntries:
```
String key = eval(document, mapping.getKey());
String strVal = eval(document, mapping.getValue());

if (strVal.contains("${")) {
    continue;
}

```

With this, when humidity has value but temperature not, temperature will be ignored and will not send the literal value to thingsboard(${$.temperature[1]}), only humidity will be send.

For me works, I know this is not the best solution, but for now helped me :)

closed 2019-12-31 by zbeacon