thingsboard/thingsboard-gateway #82: tb-config extensions

quirosaur, 2018-05-16:
when configuring tb-gateway.yml, what are possible extensions? i went through source code and didn't find parser of any kind regarding this. 

also, there is no documentation on web page about changing extension, and what extensions are available to put here.

if i copy mqtt folder and name it mqtt-foo, can i choose it like this:

````
extensions:        
        -
          id: "mqtt-foo"
          type: "MQTT_FOO"
          extensionConfiguration: mqtt-foo-config.json
````

mp-loki, 2018-05-18:
```yaml
extensions:        
        -
          id: "does not matter"
          type: "should be a supported type (see below)"
          extensionConfiguration: "your configuration file name"
```

The supported extension types are stored in **TenantServiceRegistry**. If you add a new extension type, make sure to update **TenantServiceRegistry** class.

quirosaur, 2018-05-18:
thank you! this helped enormously :)

closed 2018-05-18 by quirosaur