thingsboard/thingsboard-gateway #86: Timeout Error Connecting to Thingsboard (MQTT)

Bucknalla, 2018-06-06:
Hi,

I'm having an issue connecting to my Thingsboard Server (my own instance). I am getting a time out error appearing in my /var/log/tb-gateway/tb-gateway.log

`2018-06-06 13:46:51,051 [main] ERROR o.t.g.s.gateway.MqttGatewayService - Unable to connect to ThingsBoard. Connection timed out after [10000] milliseconds
`

My tb-gateway.yml looks as follows:

```
server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"

gateways:
  tenants:
    -
      label: "Tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: "${GATEWAY_HOST:My_Own_Host_IP}"
        port: 1883
        retryInterval: 3000
        maxInFlight: 1000
        security:
          accessToken: "${GATEWAY_ACCESS_TOKEN:My_Own_Token}"
      remoteConfiguration: true
      extensions:
        -
          id: "http"
          type: "HTTP"
extensionConfiguration: http-config.json
```

Do I need to set up my own MQTT broker? The instructions in the documentation seem to suggest that all I need to do is change the host IP and add my access token and the gateway should appear in my thingboard devices page.

Any help would be greatly appreciated!

Thanks,

Alex

mp-loki, 2018-06-06:
Are you able to connect to `My_Own_Host_IP:1883` using a third-party mqtt client, e.g. mosquitto_pub? The error says that it is unable to connect to ThingsBoard. It can be either a) incorrect configuration or b) network issues.

Regarding your other question about mqtt broker - the gateway connects to TB server via MQTT on one side and to third-party systems via extensions on the other. So if you want to configure a MQTT extension, you eventually would want to have an external broker to connect to. But this is not related to the error you have now. Also, given that you use `remoteConfiguration: true` you will configure your extension(s) in TB GUI once you resolve your connection problem.

Bucknalla, 2018-06-12:
So I've tried to connect to the broker manually - I've even got some devices connecting up to my thingsboard instance - and it seems to be functioning correctly. I wonder if my config file is incorrect? 

Am I correct in saying that the ${GATEWAY_HOST:} command is used for running the thingsboard IoT gateway under docker? I'm currently trying to run thingsboard gateway under [Resin.io](https://resin.io) so I wonder if this might be an issue?

kevz93, 2019-12-24:
I am facing the same issue. @Bucknalla , did you manage to clear this out ?
Ive tried v1.4.0.1(this connects to thingsboard but anonymous logins seems to be an issue) 
and 2.2.1rc(fails to connect )  ;/

zbeacon, 2019-12-31:
Hi, @Bucknalla , @kevz93 , @Bucknalla 

 am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
 - Less footprint and memory impact
 - Easier to extend and customize
 - Improved functionality.

You can use [this guide](https://thingsboard.io/docs/iot-gateway/getting-started/) to install and configure the gateway. Please try the new gateway and give your feedback into issues or email us info@thingsboard.io .

closed 2019-12-31 by zbeacon

santobedi, 2021-09-17:
Hello @zbeacon , I've installed Thingboard Gateway with the Python module, however, I'm unable to review the gateway statistics (in the Latest Telemetry tab) in the web UI.  After changing the host, port, and accessToken in tb_gateway.yaml file, I used  sudo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("./tb_gateway.yaml")' command for the new configuration. However, I get the TimeoutError as shown in the attached screenshot. What is going wrong in my situation? 
[image]