thingsboard/thingsboard-gateway #92: Thingsboard gateway not reconnecting

JozsefGuyas, 2018-06-14:
Hi,
I'm using thingsboard gateway 1.2.2 and thingsboard 1.4.0, if the thingsboard restarts, the gateway doesn't reconnect automatically. I have to restart it manually. 
How can I resolve the problem?

mp-loki, 2018-06-14:
I suggest you using tb-gateway version 1.4.0, because it has a number of bug fixes comparing to 1.2.2.
Beware that you will need to update your tb-gateway.yml file.
Regarding your scenario - it was tested both on 1.2.2 and 1.4.0, so I would ask you to set log level to DEBUG in your logback.xml and send us the logs.

cristianhumelnicu, 2018-06-18:
Hi
Same problem here ... if thingsboard is restarted , tb-gateway do not re-connect. 
If i restart tb-gateway after thingsboars is restarted all seems to be ok , but after a while tb-gateway stop and i need to restarted again .. BTW i use tb 1.4.0 and thingsboard 2.0.3

I thing cold be a tb-gateway bug...

mp-loki, 2018-06-20:
As soon as I was unable to reproduce this issue on my environment, I would ask you to:

1. in logback.xml set "org.thingsboard" logger level to DEBUG
2. cleanup your tb-gateway logs
3. run your scenario with ThingsBoard restart
4. send the complete tb-gateway.log
If there is anything specific about your setup, please share the details as well.

JozsefGuyas, 2018-06-20:
Hi

I have managed to reproduce the issue.
You can download the log file here:
https://drive.google.com/open?id=1cgTO6JGyCbneH08-9COYynK4o_HY5cJW

mp-loki, 2018-06-21:
So I see the issue with
```log
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)
```
Which gives almost no information about the cause and how to reproduce it. However, I would assume that it is thrown somewhere outside the try/catch block, so the MQTT sender thread fails and no reconnect is actually performed.

Can I please ask you to check the same scenario with the master branch version and let me know if the issue is still there? If you need help with building from source, I can supply you files.

Also, I would suggest you to use the following logback.xml config, because eclipse paho logs add a lot of clutter, but they are irrelevant to this issue:
```xml
    <logger name="org.thingsboard" level="TRACE" />
    <logger name="org.eclipse.milo" level="INFO" />
    <logger name="org.eclipse.paho" level="INFO" />
```

JozsefGuyas, 2018-06-21:
I have done it.
The log file can be downloaded from this link:
https://drive.google.com/open?id=10okWP7wQwRcmRX72PclX7CqZX-nqP5Ja

mp-loki, 2018-06-27:
Although I was unable to reproduce the issue, I think I was able to figure out where does it come from and merged [this pull request](https://github.com/thingsboard/thingsboard-gateway/pull/104) which I assume has to fix the issue.
I would ask you to check out the latest version, build from source and let me know if the issue was fixed or not.

cristianhumelnicu, 2018-06-27:
@mp-loki 
Hi ,
Could you compile please tb-gateway, with patch for reconnecting when TB is reset and make tb-gateway 1.4.1 release please ?

I asking you that because i really can not compile on RPI3 .. try it 3 times and no success 

Thank you !

JozsefGuyas, 2018-06-27:
Hi!

Thank you for help. I will try it, but I have recently been occupied with
my job. As soon as I try it I will send response for you.

mp-loki, 2018-06-28:
I have created a release draft v2.0.4-draft1  here:
https://github.com/thingsboard/thingsboard-gateway/releases/tag/v2.0.4-draft1
It will be 2.0.4 when ThingsBoard 2.0.4 comes out

cristianhumelnicu, 2018-06-28:
@mp-loki
Hi Valerii , is working !

Thank you very much !

closed 2019-12-31 by zbeacon