thingsboard/thingsboard-gateway #95: Tb-gateway and SSL configuration

pablodbc, 2018-06-15:
Hello, I have thingsboard with SSL configuration set and I am trying to connect the thingsboard gateway 1.4 to that instance. I was using the tb-gateway 1.2.1 before with this configuration:

`gateway:
  reporting:
    interval: 86400000
  persistence:
    type: file
    path: storage
    bufferSize: 1000
  connection:
    host: mytbhost
    port: 1883
    retryInterval: 3000
    maxInFlight: 1000
    security:
      accessToken: gatewayAccessToken
      truststore: MyTruststore
      truststorePassword: myTruststorePassword`

Now, I am using tb-gateway 1.4 with the following configuration:
`gateways:
  tenants:
    -
      label: "Main Tenants"
      reporting:
        interval: 86400000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: galen
        port: 1883
        retryInterval: 3000
        maxInFlight: 1000
        security:
          accessToken: gateway
          truststore: MyTruststore
          truststorePassword: myTruststorePassword
      remoteConfiguration: false`

I don't know if I am placing the truststore configuration in the correct place, but the documentation doesn't say something about it. The thing is that when I start the gateway now, it throws the next:

`2018-06-15 10:56:52,346 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Main Tenants] Initializing gateway
2018-06-15 10:57:02,689 [main] ERROR o.t.g.s.gateway.MqttGatewayService - Unable to connect to ThingsBoard. Connection timed out after [10000] milliseconds
java.util.concurrent.TimeoutException: null
	at io.netty.util.concurrent.AbstractFuture.get(AbstractFuture.java:56)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.initMqttClient(MqttGatewayService.java:611)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.init(MqttGatewayService.java:131)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleElement.invoke(InitDestroyAnnotationBeanPostProcessor.java:366)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor$LifecycleMetadata.invokeInitMethods(InitDestroyAnnotationBeanPostProcessor.java:311)
	at org.springframework.beans.factory.annotation.InitDestroyAnnotationBeanPostProcessor.postProcessBeforeInitialization(InitDestroyAnnotationBeanPostProcessor.java:134)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.applyBeanPostProcessorsBeforeInitialization(AbstractAutowireCapableBeanFactory.java:409)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.initializeBean(AbstractAutowireCapableBeanFactory.java:1581)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.doCreateBean(AbstractAutowireCapableBeanFactory.java:554)
	at org.springframework.beans.factory.support.AbstractAutowireCapableBeanFactory.createBean(AbstractAutowireCapableBeanFactory.java:483)
	at org.springframework.beans.factory.support.AbstractBeanFactory.doGetBean(AbstractBeanFactory.java:325)
	at org.springframework.beans.factory.support.AbstractBeanFactory.getBean(AbstractBeanFactory.java:207)
	at org.springframework.context.annotation.ConfigurationClassEnhancer$BeanMethodInterceptor.intercept(ConfigurationClassEnhancer.java:380)
	at org.thingsboard.gateway.GatewayConfiguration$$EnhancerBySpringCGLIB$$333a4dd0.getGatewayServiceBean(<generated>)
	at org.thingsboard.gateway.GatewayConfiguration$1.getGatewayService(GatewayConfiguration.java:47)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.init(DefaultTenantManagerService.java:83)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
[... 120 lines cut ...]
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1414)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:945)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:146)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:645)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:580)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:497)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:459)
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:886)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.lang.Thread.run(Thread.java:748)
Caused by: io.netty.handler.ssl.NotSslRecordException: not an SSL/TLS record: 103700064d51497364700382003c00136e657474792d6d7174742f6e4f733839546674001449466a4a49466145774b71553146395864485174
	at io.netty.handler.ssl.SslHandler.decodeJdkCompatible(SslHandler.java:1138)
	at io.netty.handler.ssl.SslHandler.decode(SslHandler.java:1203)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:489)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:428)
	... 16 common frames omitted`

Which means that the tb-gateway is not using the SSL configuration. 

Am I placing wrong the truststore location? I saw in the source code that each tenant has a MqttSecurityConfiguration assigned

mp-loki, 2018-06-15:
It would really help if you post your configuration files formatted.

If you are using SSL, you need to remove the **accessToken** property, because it is used as a flag to run in non-SSL mode. You have to specify the following fields instead:

```yaml
...
security:
    keystore: myKeyStore.jks # file that contains private and public key for Gateway certificate. Only JKS is supported
    keystorePassword: yourKeystorePassword
    keystoreKeyAlias: gateway # the alias of the gateway certificate in the keystore
    truststore: myKeyStore.jks # the TrustStore file. May be the same as keystore, but it is still required to populate
    truststorePassword: yourKeystorePassword # TrustStore password

pablodbc, 2018-06-18:
I am really sorry, I though the configuration file was formatted. Thank you so much for the quick answer

pablodbc, 2018-06-18:
Hello, sorry for reopening this issue. I did as @mp-loki said and now I am getting these errors:

On thingsboard, it makes a loop writing this:

`2018-06-18 15:39:17,227 [nioEventLoopGroup-5-7] INFO  o.t.s.t.mqtt.MqttTransportHandler - [mqtt55] Processing connect msg for client: netty-mqtt/25RG1MdX!
2018-06-18 15:39:17,227 [nioEventLoopGroup-5-7] WARN  o.t.s.t.mqtt.MqttTransportHandler - peer not authenticated`

On tb-gateway, at the same time, it throws this:

`2018-06-18 15:39:13,781 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Main Tenants] Initializing gateway
2018-06-18 15:39:44,941 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds`

After a while, it throws this:

`2018-06-18 15:39:47,391 [main] INFO  o.s.b.c.e.t.TomcatEmbeddedServletContainer - Tomcat started on port(s): 9090 (http)
2018-06-18 15:39:47,400 [main] INFO  o.t.gateway.GatewayApplication - Started GatewayApplication in 38.698 seconds (JVM running for 39.767)
2018-06-18 15:39:48,152 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Sending message [MqttPersistentMessage(id=3fdb9e36-1a05-4940-b8bd-a6f7b32d5d98, timestamp=0, deviceId=null, messageId=1, topic=v1/devices/me/attributes/request/1, payload=[123, 34, 115, 104, 97, 114, 101, 100, 34, 58, 34, 99, 111, 110, 102, 105, 103, 117, 114, 97, 116, 105, 111, 110, 34, 125])]
2018-06-18 15:39:48,163 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
2018-06-18 15:39:48,166 [pool-4-thread-1] WARN  o.t.g.s.gateway.MqttGatewayService - Error getiing attributes
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)
2018-06-18 15:39:48,166 [pool-4-thread-1] WARN  o.t.g.service.MqttMessageReceiver - Failed to send message [MqttPersistentMessage(id=3fdb9e36-1a05-4940-b8bd-a6f7b32d5d98, timestamp=0, deviceId=null, messageId=1, topic=v1/devices/me/attributes/request/1, payload=[123, 34, 115, 104, 97, 114, 101, 100, 34, 58, 34, 99, 111, 110, 102, 105, 103, 117, 114, 97, 116, 105, 111, 110, 34, 125])] due to [{}]
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)
`
[image]

I have my configuration exactly like this right now.

Thank you so much in advance

mp-loki, 2018-06-18:
You need to:
1. Export gateway public key in .pem format using keytool
2. In ThingsBoard GUI change Device authentication option to "X.509 Certificates"
3. Paste the contents of .pem public key file & save

pablodbc, 2018-06-18:
Done, I did all of that but it keeps throwing the same error. I saw in the source code that if you set the access token and a truststore, it should be enough to set the ssl communication, this worked in the version 1.2.1, do you have any idea of why isn't it working now?

mp-loki, 2018-06-20:
I am sorry, it is a bug. One-Way-SSL MQTT configuration was broken in 1.2.2/1.4.0.
[Fixed](https://github.com/thingsboard/thingsboard-gateway/pull/99) now in main branch.
You can build from source if you want. Your initial configuration should work now.

Btw, your configuration for two-way SSL should work also, if all the keys are generated and imported correctly.

pablodbc, 2018-06-26:
Hello, I followed your instructions and I could establish both One-Way-SSL and Two-Way-SSL connections. Thank you so much, now I have another issue but I will open another thread and close this one to avoid confusion

closed 2018-06-26 by pablodbc

scpf19, 2019-10-30:
Hi, is the configuration for the SSL communication between the IoT Gateway and the Thingsboard server still valid? 

I did the configuration as described in the manual and used the scripts to generate the keys for the server and client from the official website. However, the communication between the IoT Gateway and the Thingsboard server is not working.