thingsboard/thingsboard-gateway #103: RPC subscription failure

pablodbc, 2018-06-26:
Hello, 

I was trying to connect tb-gateway (version from master branch) to thingsboard 2.0.3. But when the gateway tries to initialize, thingsboard throws this:

> 2018-06-26 14:38:13,649 [nioEventLoopGroup-12-7] INFO  o.t.s.t.mqtt.MqttTransportHandler - [mqtt7] Processing connect msg for client: netty-mqtt/uG2KQSDz!
2018-06-26 14:38:13,717 [nioEventLoopGroup-12-7] WARN  o.t.s.t.mqtt.MqttTransportHandler - [mqtt7] Failed to subscribe to [v1/gateway/rpc][AT_MOST_ONCE]
>
 
At the same time, tb-gateway throws this:

> 2018-06-26 14:38:10,802 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Main Tenants] Initializing gateway
2018-06-26 14:38:12,087 [nioEventLoopGroup-2-1] WARN  i.n.channel.DefaultChannelPipeline - An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
java.lang.NullPointerException: null
	at nl.jk5.mqtt.MqttChannelHandler.handlePuback(MqttChannelHandler.java:215)
	at nl.jk5.mqtt.MqttChannelHandler.channelRead0(MqttChannelHandler.java:37)
	at nl.jk5.mqtt.MqttChannelHandler.channelRead0(MqttChannelHandler.java:11)
	at io.netty.channel.SimpleChannelInboundHandler.channelRead(SimpleChannelInboundHandler.java:105)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at nl.jk5.mqtt.MqttPingHandler.channelRead(MqttPingHandler.java:39)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.handler.timeout.IdleStateHandler.channelRead(IdleStateHandler.java:286)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:310)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:284)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.handler.ssl.SslHandler.unwrap(SslHandler.java:1389)
	at io.netty.handler.ssl.SslHandler.decodeJdkCompatible(SslHandler.java:1159)
	at io.netty.handler.ssl.SslHandler.decode(SslHandler.java:1203)
	at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:489)
	at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:428)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:265)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1414)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:945)
	at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:146)
	at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:645)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:580)
	at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:497)
	at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:459)
	at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:886)
	at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	at java.lang.Thread.run(Thread.java:748)
2018-06-26 14:38:12,118 [pool-2-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-06-26 14:38:12,155 [pool-1-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Configuration arrived! {}
2018-06-26 14:38:12,163 [pool-1-thread-1] WARN  o.t.g.s.gateway.MqttGatewayService - Failed to process arrived message!
>

Do you know what is happening?

Best regards and thanks in advance!

rof20004, 2018-07-19:
I am with same problem, someone can help?

:)

rof20004, 2018-07-19:
I see many Outgoing queue is not empty. [1] messages are still in progress messages, I need to restart gateway to messages send again.

mp-loki, 2018-07-20:
Outgoing queue is not empty. [1] messages is ok, although I see many people find it misleading. I will improve the log message

rof20004, 2018-08-08:
I solved by this way:

MqttMessageSender.java
```
private boolean checkOutgoingQueueIsEmpty() {
        if (!outgoingQueue.isEmpty()) {
            int pendingCount = 0;
            boolean allFinished = true;
            for (Future<Void> future : outgoingQueue) {
                if (!future.isDone()) {
                    pendingCount++;
                    continue;  // Continue to next item of iteration, only this solved the problem.  
                }
                allFinished &= future.isDone();
            }
            if (allFinished) {
                outgoingQueue = new ConcurrentLinkedQueue();
                return true;
            }
            log.info("Outgoing queue is not empty. [{}] messages are still in progress", pendingCount);
            return false;
        }
        return true;
    }
```

rof20004, 2018-08-08:
@mp-loki I will send tonight, but I do not know if the changes is for this specifc issue:
`
 But when the gateway tries to initialize, thingsboard throws this:
`

I think this issue is for warning when gateway starts, my correction is about the "Outgoing queue is not empty", that is really wrong, because that iteration set allFinished false to all item of iteration then lock all subsequent messages.

Sorry, my english is basic.

pablodbc, 2018-09-10:
I have an update on this:

I did the change that @rof20004 proposed and it works fine. However it didn't fixed the issue I had. I realized that this error only appears if you try to connect the gateway to thingsboard when the gateway device already has an extension configuration. I solved my issue by removing the extension, restarting the gateway and importing the extension again. I don't know where this happens in the code, but I hope this is helpful to find this issue cause.

rof20004, 2018-09-10:
@pablodbc Thanks for update and yes, my fix is not for your problem about initialization, is only for "Outgoing queue is not empty" and the RPC data was not send, because it was locked into threads, and now works great, all my rpc data was send normal now :)

closed 2019-12-31 by zbeacon