thingsboard/thingsboard-gateway #105: MQTT & SIGFOX Simultaneously...

AngeLinuX99, 2018-07-01:
Hi

I have an 1.2.2 instance, but i've noticed that is not possible have up both services MQTT and SigFox simultaneouisly. They works like a charm as single service (mqtt OR sigfox) but not both of them.

Any clue to fix it ?

Thanks.

mp-loki, 2018-07-02:
This should work. Can you please share your tb-gateway.yml ?

AngeLinuX99, 2018-07-02:
#
# Copyright © 2017 The Thingsboard Authors
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#

gateway:
  reporting:
    interval: 60000
  persistence:
    path: storage
    bufferSize: 1000
    pollingInterval: 1000
  connection:
    host: "demo.thingsboard.io"
    port: 1883
    retryInterval: 3000
    maxInFlight: 1000
    security:
      accessToken: my_access_token
#      keystore: mqttclient.jks
#      keystorePassword: password
#      keystoreKeyAlias: clientalias
#      truststore: mqttclient.jks
#      truststorePassword: password

opc:
  enabled: false
  configuration: opc-config.json

mqtt:
  enabled: true
  configuration: mqtt-config.json

sigfox:
  enabled: true
  configuration: sigfox-config.json

server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"

AngeLinuX99, 2018-07-02:
``#
# Copyright © 2017 The Thingsboard Authors
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
#

gateway:
  reporting:
    interval: 60000
  persistence:
    path: storage
    bufferSize: 1000
    pollingInterval: 1000
  connection:
    host: "demo.thingsboard.io"
    port: 1883
    retryInterval: 3000
    maxInFlight: 1000
    security:
      accessToken: My_Access_Token
#      keystore: mqttclient.jks
#      keystorePassword: password
#      keystoreKeyAlias: clientalias
#      truststore: mqttclient.jks
#      truststorePassword: password

opc:
  enabled: false
  configuration: opc-config.json

mqtt:
  enabled: true
  configuration: mqtt-config.json

sigfox:
  enabled: false
  configuration: sigfox-config.json

server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: `"${UPDATES_ENABLED:true}"`

adamfadil, 2018-07-03:
Hi, According to thingsboard documentation (https://thingsboard.io/docs/iot-gateway/sigfox/). you should  change the line ( configuration: sigfox-config.json) of your tb-gateway.yml.txt  with (  configuration: http-config.json)

AngeLinuX99, 2018-07-03:
I've changed it as you suggest,  but I have the same problem. MQTT only works. If I select only SIGFOX on tb-gateway.yml, I can find this "Mapped URL Path" on log files but not when I select MQTT and SIGFOX...

2018-07-03 22:59:12,773 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerAdapter - Looking for @ControllerAdvice: org.springframework.boot.context.embedded.AnnotationConfigEmbeddedWebApplicationContext@16c0663d: startup date [Tue Jul 03 22:58:51 CEST 2018]; root of context hierarchy
2018-07-03 22:59:13,525 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/sigfox/{deviceTypeId}],methods=[POST]}" onto public void org.thingsboard.gateway.extensions.sigfox.SigfoxController.handleRequest(java.lang.String,java.lang.String,java.lang.String) throws java.lang.Exception
2018-07-03 22:59:13,552 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/error],produces=[text/html]}" onto public org.springframework.web.servlet.ModelAndView org.springframework.boot.autoconfigure.web.BasicErrorController.errorHtml(javax.servlet.http.HttpServletRequest,javax.servlet.http.HttpServletResponse)
2018-07-03 22:59:13,555 [main] INFO  o.s.w.s.m.m.a.RequestMappingHandlerMapping - Mapped "{[/error]}" onto public org.springframework.http.ResponseEntity<java.util.Map<java.lang.String, java.lang.Object>> org.springframework.boot.autoconfigure.web.BasicErrorController.error(javax.servlet.http.HttpServletRequest)
2018-07-03 22:59:13,871 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/webjars/**] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-07-03 22:59:13,872 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/**] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-07-03 22:59:14,179 [main] INFO  o.s.w.s.h.SimpleUrlHandlerMapping - Mapped URL path [/**/favicon.ico] onto handler of type [class org.springframework.web.servlet.resource.ResourceHttpRequestHandler]
2018-07-03 22:59:14,209 [pool-5-thread-1] INFO  o.t.g.s.update.DefaultUpdateService - New version 1.4.0 is available!