thingsboard/thingsboard-gateway #106: gateway 2.0.3 issue

JozsefGuyas, 2018-07-02:
Hi,

I have installed the tb-gateway 2.0.4 . The gateway always sends the first publish, so after the first publish the gateway does not send telemetry to thingsboard (2.0.3).
I have uploaded the config files here:

https://drive.google.com/open?id=1bj4HHy6kTe8HWU6BskaCZP7zx5ED_5Hw

Could you tell me whats worng?

mp-loki, 2018-07-02:
Please attach the gateway log file as well.
Just to confirm - `localhost` in **mqtt-config.json** and `192.168.100.203` in **tb-gateway.yml** are different hosts or the same host?

mp-loki, 2018-07-02:
```
java.io.IOException: No route to host
	at sun.nio.ch.FileDispatcherImpl.read0(Native Method)
	at sun.nio.ch.SocketDispatcher.read(SocketDispatcher.java:39)
```
and then:
```
2018-07-02 16:15:12,605 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Attempting to reconnect to ThingsBoard.
2018-07-02 16:15:22,607 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
```
I suspect this is a network issue

JozsefGuyas, 2018-07-02:
When I restart the gateway, I can publish again, but thingsboard does not received the second and other publish.

AlexanderJP91, 2018-07-06:
I've also noticed the same issue with the upgrade to v2 of TB and TB-GW.
I've analyzed more deeply the issue, this arises using both the configuration file and the UI configuration.
I've tried sending a single packet, the gateway received it and also generated another packet, from the following message in the LOG:
`[pool-2-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [2] messages are still in progress`
I've decoded the messages, resulting that message in position 0 is auto-generated and is the following:
`{"device":"MyDevice","type":null}`
While the second packet contains the actual telemetry I'm uploading
At this point the packet containing the telemetry message is forwarded to TB and the device is correctly generated and contains the uploaded telemetry.
On the other side the the GW did not consume the message in position 0 (the auto-generated one). So it blocks waiting on the pipeline to be consumed and all other telemetries cannot be uploaded until the GW is not restarted.
The LOG has the following loop:
`[pool-2-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress`
I've tried sending bigger packets, but the GW always gets stuck on this auto-generated one.
The problem arises in the method `checkOutgoingQueueIsEmpty()` inside the file [thingsboard-gateway/src/main/java/org/thingsboard/gateway/service/MqttMessageSender.java](https://github.com/thingsboard/thingsboard-gateway/blob/master/src/main/java/org/thingsboard/gateway/service/MqttMessageSender.java).
It looks like the future job is never finished, so the check
`if (!future.isDone()) {
                    pendingCount++;
}`
Will always set the pendingCount to 1, so the method always returns false, thus meaning that the queue never results empty.
@mp-loki please give check to this issue, I think it is very critical, since it affects the communication between TB and the GW.
I've been working on this issue for a couple of days, but still couldn't find a solution, so that I could ask for a merge, but I gained various info regarding this so please let me know if more info is required!

AlexanderJP91, 2018-07-06:
Ok I got a way around the problem, but it is a very rough solution since I still haven't got a clear view on how the messages are managed in the MQTT Service.
By changing the `run()` method you can drop the packet and not include it in the list that will after be queued towards TB.
You can check the modification on my [repo](https://github.com/AlexanderJP91/thingsboard-gateway).
Precisely the modification is in the file [MqttMessageSender.java](https://github.com/AlexanderJP91/thingsboard-gateway/blob/master/src/main/java/org/thingsboard/gateway/service/MqttMessageSender.java). In the `run()` method you can see the counter.
I repeat that this is just a fast workaround and that there is probably a better way. I think the actual problem can be solved in `getMqttPersistentMessages` method in the [PersistentFileServiceImpl.java](https://github.com/AlexanderJP91/thingsboard-gateway/blob/master/src/main/java/org/thingsboard/gateway/service/PersistentFileServiceImpl.java) since it seems to me that here the first message is dropped (oldestFile).
@mp-loki please give a check and let us know, thank you.

AlexanderJP91, 2018-07-10:
For more info I've posted an error stack trace of the [Thingsboard log](https://pastebin.com/KSKCr63r).
The error arises when the GW gets blocked on the queued message and the Thingsboard log reports a JsonNull exception.

mp-loki, 2018-07-10:
@AlexanderJP91 thank you for you research! The issue you mentioned with `UnsupportedOperationException: JsonNull` was fixed [here](https://github.com/thingsboard/thingsboard/pull/875/commits/489a1a6b6281deb1160840268e0fae0d0db0f2fc) a while ago and is available in master branch.

AlexanderJP91, 2018-07-10:
@mp-loki thank you for the reply. Actually I didn't sync my fork with the upstream, so I'm probably missing this commit.
Just off topic: is it possible to update to the same/newer version of thingsboard v2 without uninstalling it?

mp-loki, 2018-07-11:
go to your install lib and replace thingsboard.jar

closed 2019-12-31 by zbeacon

DanielRichart, 2020-08-13:
Hello,

I'm facing this same issue in my tb-gateway, where it suddenly stop receiving messages and only receives again after I remove the mqtt extension config file, restart the tb-gateway service and put the mqtt config file back in.

In the logs, i receive lot of lines with:
[pool-3-thread-1] INFO o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
[pool-3-thread-1] INFO o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress

I wish to try the modifications in the MqttMessageSender.java file described above, but **i can't find this file in my system.** 

**Please, where can i find it?**

Thanks a lot,
Daniel Richart