thingsboard/thingsboard-gateway #108: tb gateway 2.04 loses connection to thingsboard service 2.1.0

abrasat, 2018-07-26:
tb gateway, tb service and mqtt broker all running locally on same windows 10 pc.
the tb gateway is not able to maintain a stable connection to the tb service.
this problem did not occur before updating to last 2.04 release, respectively last tb service 2.1.0.
the tb gateway log:

```
2018-07-26 10:36:11,955 [main] INFO  o.t.gateway.GatewayApplication - Starting GatewayApplication v2.0.4-SNAPSHOT on NB158 with PID 9236 (D:\tb\tb-gateway\lib\tb-gateway.jar started by NB158$ in D:\tb\tb-gateway\conf)
2018-07-26 10:36:11,971 [main] DEBUG o.t.gateway.GatewayApplication - Running with Spring Boot v1.4.3.RELEASE, Spring v4.3.5.RELEASE
2018-07-26 10:36:11,971 [main] INFO  o.t.gateway.GatewayApplication - No active profile set, falling back to default profiles: default
2018-07-26 10:36:12,065 [main] INFO  o.s.b.c.e.AnnotationConfigEmbeddedWebApplicationContext - Refreshing org.springframework.boot.context.embedded.AnnotationConfigEmbeddedWebApplicationContext@d5c04e: startup date [Thu Jul 26 10:36:12 CEST 2018]; root of context hierarchy
2018-07-26 10:36:12,112 [background-preinit] INFO  o.h.validator.internal.util.Version - HV000001: Hibernate Validator 5.2.4.Final
2018-07-26 10:36:13,924 [main] INFO  o.s.b.f.config.PropertiesFactoryBean - Loading properties file from URL [jar:file:/D:/tb/tb-gateway/lib/tb-gateway.jar!/BOOT-INF/lib/spring-integration-core-4.3.6.RELEASE.jar!/META-INF/spring.integration.default.properties]
2018-07-26 10:36:13,940 [main] INFO  o.s.i.config.IntegrationRegistrar - No bean named 'integrationHeaderChannelRegistry' has been explicitly defined. Therefore, a default DefaultHeaderChannelRegistry will be created.
2018-07-26 10:36:14,346 [main] INFO  o.s.i.c.DefaultConfiguringBeanFactoryPostProcessor - No bean named 'errorChannel' has been explicitly defined. Therefore, a default PublishSubscribeChannel will be created.
2018-07-26 10:36:14,377 [main] INFO  o.s.i.c.DefaultConfiguringBeanFactoryPostProcessor - No bean named 'taskScheduler' has been explicitly defined. Therefore, a default ThreadPoolTaskScheduler will be created.
2018-07-26 10:36:14,643 [main] INFO  o.s.c.s.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration' of type [class org.springframework.transaction.annotation.ProxyTransactionManagementConfiguration$$EnhancerBySpringCGLIB$$9f981343] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2018-07-26 10:36:14,846 [main] INFO  o.s.b.f.config.PropertiesFactoryBean - Loading properties file from URL [jar:file:/D:/tb/tb-gateway/lib/tb-gateway.jar!/BOOT-INF/lib/spring-integration-core-4.3.6.RELEASE.jar!/META-INF/spring.integration.default.properties]
2018-07-26 10:36:14,846 [main] INFO  o.s.c.s.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'integrationGlobalProperties' of type [class org.springframework.beans.factory.config.PropertiesFactoryBean] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2018-07-26 10:36:14,846 [main] INFO  o.s.c.s.PostProcessorRegistrationDelegate$BeanPostProcessorChecker - Bean 'integrationGlobalProperties' of type [class java.util.Properties] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2018-07-26 10:36:16,174 [main] INFO  o.s.b.c.e.t.TomcatEmbeddedServletContainer - Tomcat initialized with port(s): 9090 (http)
2018-07-26 10:36:16,190 [main] INFO  o.a.catalina.core.StandardService - Starting service Tomcat
2018-07-26 10:36:16,206 [main] INFO  o.a.catalina.core.StandardEngine - Starting Servlet Engine: Apache Tomcat/8.5.6
2018-07-26 10:36:16,362 [localhost-startStop-1] INFO  o.a.c.c.C.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
2018-07-26 10:36:16,362 [localhost-startStop-1] INFO  o.s.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 4297 ms
2018-07-26 10:36:16,612 [localhost-startStop-1] INFO  o.s.b.w.s.ServletRegistrationBean - Mapping servlet: 'dispatcherServlet' to [/]
2018-07-26 10:36:16,612 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'characterEncodingFilter' to: [/*]
2018-07-26 10:36:16,612 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'hiddenHttpMethodFilter' to: [/*]
2018-07-26 10:36:16,612 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'httpPutFormContentFilter' to: [/*]
2018-07-26 10:36:16,612 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'requestContextFilter' to: [/*]
2018-07-26 10:36:16,752 [pool-1-thread-1] TRACE o.t.g.s.update.DefaultUpdateService - Executing check update method for instanceId [d348df3f-cdd0-42d0-9915-ec4cb79c66eb], platform [windows] and version [2.0.4-SNAPSHOT]
2018-07-26 10:36:16,831 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Tenant] Initializing gateway
2018-07-26 10:36:18,034 [nioEventLoopGroup-2-2] DEBUG o.t.g.s.gateway.MqttGatewayService - Gateway connect result code: [CONNECTION_ACCEPTED]
2018-07-26 10:36:18,112 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='null', payload={"shared":"configuration"}, timestamp=0, topic='v1/devices/me/attributes/request/1'id=f7d31efc-9440-4425-966b-8c61e10b550c, messageId=1}]
2018-07-26 10:36:18,112 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-07-26 10:36:18,112 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Tenant] Initializing extension: [MQTT]
2018-07-26 10:36:18,112 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-07-26 10:36:18,112 [pool-4-thread-1] DEBUG o.t.g.service.MqttMessageReceiver - Successfully sent message: [null]
2018-07-26 10:36:18,112 [main] INFO  o.t.g.e.m.c.DefaultMqttClientService - [Tenant] Initializing MQTT client service!
2018-07-26 10:36:18,143 [nioEventLoopGroup-2-1] TRACE o.t.g.s.gateway.MqttGatewayService - Message arrived [v1/devices/me/attributes/response/1] {}
2018-07-26 10:36:18,143 [pool-2-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Configuration arrived! {}
2018-07-26 10:36:18,159 [pool-2-thread-1] WARN  o.t.g.s.gateway.MqttGatewayService - Failed to process arrived message!
2018-07-26 10:36:18,393 [main] DEBUG o.t.g.e.m.client.MqttBrokerMonitor - [localhost:22000] MQTT broker connection attempt!
2018-07-26 10:36:18,409 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] INFO  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:22000] MQTT broker connection established!
2018-07-26 10:36:18,424 [main] INFO  o.t.g.s.gateway.MqttGatewayService - Subscription added: RpcCommandSubscription(deviceNameFilter=.*, listener=org.thingsboard.gateway.extensions.mqtt.client.MqttBrokerMonitor@1bd4fdd)
2018-07-26 10:36:18,424 [main] WARN  o.t.g.s.gateway.MqttGatewayService - Subscription was already added: RpcCommandSubscription(deviceNameFilter=.*, listener=org.thingsboard.gateway.extensions.mqtt.client.MqttBrokerMonitor@1bd4fdd)
2018-07-26 10:36:18,487 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] TRACE o.t.g.e.m.c.c.m.MqttJsonConverter - Parsing json message: {"ts":1532594354788,"values":{"Fr1":    0.00,"Fr2":    0.00,"Fr3":    0.00,"Srg1":   20.50,"Srg2":   27.00,"Srg3":   17.90,"Spd":    0.00,"HeadLength":   13.21,"TailLength":    6.20}}
2018-07-26 10:36:18,674 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] INFO  o.t.g.s.gateway.MqttGatewayService - [HGC] Device Connected!
2018-07-26 10:36:18,674 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] TRACE o.t.g.s.gateway.MqttGatewayService - [HGC][4] Updating device telemetry: [BasicTsKvEntry{ts=1532594354788, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr1'}}, BasicTsKvEntry{ts=1532594354788, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr2'}}, BasicTsKvEntry{ts=1532594354788, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr3'}}, BasicTsKvEntry{ts=1532594354788, kv=DoubleDataEntry{value=20.5} BasicKvEntry{key='Gap1'}}, BasicTsKvEntry{ts=1532594354788, kv=DoubleDataEntry{value=27.0} BasicKvEntry{key='Gap2'}}, BasicTsKvEntry{ts=1532594354788, kv=DoubleDataEntry{value=17.9} BasicKvEntry{key='Gap3'}}, BasicTsKvEntry{ts=1532594354788, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Speed'}}, BasicTsKvEntry{ts=15325943547 [... cut ...]
2018-07-26 10:36:36,191 [pool-4-thread-1] WARN  o.t.g.service.MqttMessageReceiver - Failed to send message [{deviceId='HGC', payload={"HGC":[{"ts":1532594354788,"values":{"Fr1":0.0,"Fr2":0.0,"Fr3":0.0,"Gap1":20.5,"Gap2":27.0,"Gap3":17.9,"Speed":0.0,"HeadLength":13.21,"TailLength":6.2}}]}, timestamp=0, topic='v1/gateway/telemetry'id=c2d199fa-837f-456f-959a-64c592305c0e, messageId=4}] due to [{}]
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)
```

and so on

abrasat, 2018-07-26:
Forgot to tell that in the case above the telemetry data of the device are refreshed only once.

If making the device public, the telemetry data are refreshed correctly, but also cyclically set to 0 with the tb gateway start-time as timestamp (?!).
The tb gateway log in this case:

```
2018-07-26 12:05:39,644 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='HGC', payload={"HGC":[{"ts":1532599715223,"values":{"Fr1":0.0,"Fr2":0.0,"Fr3":0.0,"Gap1":20.5,"Gap2":27.0,"Gap3":17.9,"Speed":0.0,"HeadLength":13.21,"TailLength":6.2}}]}, timestamp=0, topic='v1/gateway/telemetry'id=95b62e69-74a6-4c0a-8e43-2079e31c28ec, messageId=6373}]
2018-07-26 12:05:39,644 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-07-26 12:05:39,644 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-07-26 12:05:39,644 [pool-4-thread-1] DEBUG o.t.g.s.gateway.MqttGatewayService - [6373][HGC] Device telemetry published to ThingsBoard!
2018-07-26 12:05:39,893 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] TRACE o.t.g.e.m.c.c.m.MqttJsonConverter - Parsing json message: {"ts":1532599716225,"values":{"Fr1":    0.00,"Fr2":    0.00,"Fr3":    0.00,"Srg1":   20.50,"Srg2":   27.00,"Srg3":   17.90,"Spd":    0.00,"HeadLength":   13.21,"TailLength":    6.20}}
2018-07-26 12:05:39,893 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] TRACE o.t.g.s.gateway.MqttGatewayService - [HGC][6374] Updating device telemetry: [BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr1'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr2'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr3'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=20.5} BasicKvEntry{key='Gap1'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=27.0} BasicKvEntry{key='Gap2'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=17.9} BasicKvEntry{key='Gap3'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Speed'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=13.21} BasicKvEntry{key='HeadLength'}}, BasicTsKvEntry{ts=1532599716225, kv=DoubleDataEntry{value=6.2} BasicKvEntry{key='TailLength'}}]
2018-07-26 12:05:39,893 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] DEBUG o.t.g.e.m.client.MqttBrokerMonitor - Re-scheduling keep alive timer for device HGC with timeout = 60000
2018-07-26 12:05:40,659 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='HGC', payload={"HGC":[{"ts":1532599716225,"values":{"Fr1":0.0,"Fr2":0.0,"Fr3":0.0,"Gap1":20.5,"Gap2":27.0,"Gap3":17.9,"Speed":0.0,"HeadLength":13.21,"TailLength":6.2}}]}, timestamp=0, topic='v1/gateway/telemetry'id=a0087686-f0dd-4dc6-9898-b90596d637c5, messageId=6374}]
2018-07-26 12:05:40,659 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-07-26 12:05:40,659 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-07-26 12:05:40,659 [pool-4-thread-1] DEBUG o.t.g.s.gateway.MqttGatewayService - [6374][HGC] Device telemetry published to ThingsBoard!
2018-07-26 12:05:40,909 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] TRACE o.t.g.e.m.c.c.m.MqttJsonConverter - Parsing json message: {"ts":1532599717226,"values":{"Fr1":    0.00,"Fr2":    0.00,"Fr3":    0.00,"Srg1":   20.50,"Srg2":   27.00,"Srg3":   17.90,"Spd":    0.00,"HeadLength":   13.21,"TailLength":    6.20}}
2018-07-26 12:05:40,909 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] TRACE o.t.g.s.gateway.MqttGatewayService - [HGC][6375] Updating device telemetry: [BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr1'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr2'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Fr3'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=20.5} BasicKvEntry{key='Gap1'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=27.0} BasicKvEntry{key='Gap2'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=17.9} BasicKvEntry{key='Gap3'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=0.0} BasicKvEntry{key='Speed'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=13.21} BasicKvEntry{key='HeadLength'}}, BasicTsKvEntry{ts=1532599717226, kv=DoubleDataEntry{value=6.2} BasicKvEntry{key='TailLength'}}]
2018-07-26 12:05:40,909 [MQTT Call: 4def0eef-be0f-46f2-8a97-c9bfae83610d] DEBUG o.t.g.e.m.client.MqttBrokerMonitor - Re-scheduling keep alive timer for device HGC with timeout = 60000
2018-07-26 12:05:40,956 [nioEventLoopGroup-2-19] WARN  i.n.channel.DefaultChannelPipeline - An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
java.lang.IllegalStateException: complete already: DefaultPromise@13a561b(failure: java.util.concurrent.CancellationException)
                at io.netty.util.concurrent.DefaultPromise.setSuccess(DefaultPromise.java:97)
                at nl.jk5.mqtt.MqttChannelHandler.handlePuback(MqttChannelHandler.java:215)
[... cut ...]
                at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
                at java.lang.Thread.run(Unknown Source)
2018-07-26 12:05:42,175 [nioEventLoopGroup-2-19] WARN  i.n.channel.DefaultChannelPipeline - An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
java.lang.IllegalStateException: complete already: DefaultPromise@1377c4e(failure: java.util.concurrent.CancellationException)
```

JozsefGuyas, 2018-08-27:
Hello,
I use the tb gateway 1.2.2, and I have a same problem.When I have tried to open the storage files in the /use/share/tb-gateway/bin/stroge/, I have experienced that every files are corrupted, so I have to remove these files.

closed 2020-01-02 by zbeacon