thingsboard/thingsboard-gateway #109: Failed connecting gateway to Kepware Server.

fwaznad, 2018-08-06:
hi @mp-loki @ashvayka ,

I installed tb-gateway on rasp pi. and try to implement opc ua connection with Kepware Server which is installed in my window PC. However the connection is failed
tb-gateway.yml file
```
server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"

gateways:
  tenants:
    -
      label: "Tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: "${GATEWAY_HOST:demo.thingsboard.io}"
        port: 1883
        retryInterval: 3000
        maxInFlight: 1000
        security:
          accessToken: "${GATEWAY_ACCESS_TOKEN:As6VolVwWB0lCe65Pp1U}"
      remoteConfiguration: false
      extensions:
        -
          id: "opc-ua"
          type: "OPC UA"
          extensionConfiguration: opc-config.json
```
opc-conf.json file as below
```
{  "servers": [
    {
      "applicationName": "Thingsboard OPC-UA client",
      "applicationUri": "",
      "host": "192.168.43.233",
      "port": 49320,
      "scanPeriodInSeconds": 10,
      "timeoutInMillis": 5000,
      "security": "Basic128Rsa15",
      "identity": {
        "type": "anonymous"
      },
      "keystore": {
        "type": "PKCS12",
        "location": "example.pfx",
        "fileContent": "",
        "password": "secret",
        "alias": "gateway",
        "keyPassword": "secret"
      },
      "mapping": [
        {
          "deviceNodePattern": "Kepware\\.KEPServerEX\\.V6\\.Painting\\.Oven",
          "deviceNamePattern": "Device1",
          "attributes": [
            {
              "key": "Fan",
              "type": "float",
              "value": "${Fan}"
            }
          ],
          "timeseries": [
            {
              "key": "Pressure",
              "type": "float",
              "value": "${Pressure}"
            }
          ]
        }
      ]
    }
  ]
}
```
Opc server config
[image]
[image]

Kepware server Demo

[image]

In demo.thingsboard.org, it showed opcExstentionError

[image]
 
What configurations do I missed?

Regard
waz

leehawk2001, 2019-03-09:
Hi fwaznad, I've got the same error msg as yours. Have you got any solutions yet? Thanks!

jakem68, 2019-04-14:
Hi, I also have this error and would be interested to know if anybody found a solution. Thanks!

aniketfidelit, 2019-07-31:
Hi, I am also facing same issue. Let me know if anyone found solution. Thanks.

kevz93, 2019-12-23:
Did you guys set anonymous login to 'true' on KepEx ?

zbeacon, 2020-01-02:
Hi, @fwaznad, @jakem68, @aniketfidelit 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
 - Less footprint and memory impact
 - Easier to extend and customize
 - Improved functionality.

You can try to use the new gateway for connection to the OPC-UA server.

closed 2020-01-02 by zbeacon

Houdaarma, 2024-04-03:
Subject: Connection Issue from PLC to KEPServerEX with Azure IoT Hub

Hello,

I'm reaching out to you because I'm currently experiencing a connection issue between my PLC, KEPServerEX, and Azure IoT Hub. Here's a summary of the situation:

Environment Configuration:

I have a PLC that I want to connect to KEPServerEX to retrieve data.
I have configured KEPServerEX to use the MQTT agent to transmit data to Azure IoT Hub.
Issue Encountered:

When attempting to establish the connection between KEPServerEX and Azure IoT Hub, I encounter an error stating "MQTT agent 'AZUREIOTHUB' failed to connect - reason: 'connect timed out'".I am seeking your assistance in resolving this issue. If you have any additional advice, specific checks to perform, or recommendations for resolving this connection error, I would greatly appreciate your guidance.

Thank you very much for your help.

Best regards,
ARMAILI HOUDA

   Error           IoT Gateway                 Failed to connect to server: '255.255.255.255:65535'. Please verify this connection information is correct and that the host can be reached.

bartosztomczakIoT, 2025-06-17:
> Subject: Connection Issue from PLC to KEPServerEX with Azure IoT Hub
>
> Hello,
>
> I'm reaching out to you because I'm currently experiencing a connection issue between my PLC, KEPServerEX, and Azure IoT [...]

Hello Houdaarma,
Did you managed to resolve this problem? 
I'm having the same error: 'Error IoT Gateway Failed to connect to server: '255.255.255.255:65535'. Please verify this connection information is correct and that the host can be reached.'