thingsboard/thingsboard-gateway #112: thingsboard-gateway compilation errors

gigi130358, 2018-08-25:
I  build  thingsboard "2.1.1-SNAPSHOT" and netty-mqtt "2.0.1TB" from source master on  ubuntu 18.04
Then I put both the reference version of these packages in thingsboard-gateway/pom.xml, so there aren't any missing dependency when I build thingsboard-gateway with mvn clean install.
However I get the attached compilation error.
[image].
Any idea ?
Thanks

dyosick, 2018-08-26:
Need to do two things to compile:
1. Remove import of GraphicsPrimitive from MqttGatewayService.
2. Build netty-mqtt of version 2.0.0TB instead of 2.0.1TB.

gigi130358, 2018-08-27:
Thanks Denis,
removing GraphicsPrimitive reduce the compilation errors 
[image]
I try to build different netty-mqtt version (Merge pull request #3 from vkukhtyn/master, update version)
[image]
[image]

but these thingsboard-gateway compilation errors still remains.
Which netty-mqtt version should I use ?

dyosick, 2018-08-27:
Try to checkout and build the 'Fix NPE' commit - this is the latest commit in netty-mqtt before the latest TB gateway (2.O.4) was released.
But this is workaround and the right way is to ping TB developers to fix the issue. Maybe @mp-loki can help?