thingsboard/thingsboard-gateway #114: tb gateway  sync error

JozsefGuyas, 2018-08-27:
Hi,

I use the tb gateway 1.2.2. There was no error so far, but there was no data uploading today.
I have checked the log files, and I have found some error, and I have checked the storage folder too. Every storage file were corrupted. Can I fix these files?
Could you suggest solution?

Log files:
https://drive.google.com/open?id=12nNnjM3GUuhsm_T9lVPXeM0_hRRz3Uvf

doranagy, 2018-08-27:
Hi there,

I bumped into the same issue.

vargaviktor193, 2018-08-27:
Hey
I have run into this situation also. Any idea?

nmenoni, 2018-10-17:
Hi, very similar error here but with the tb gateway 2.1.1.
Everything worked perfect until the internet link was down for approximately 3 hours, after that it was not reconnected and neither uploaded the data cached.

Log:
`2018-10-17 09:30:04,167 [nioEventLoopGroup-2-53] WARN  i.n.channel.DefaultChannelPipeline - An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
java.lang.IllegalStateException: complete already: DefaultPromise@48eb95(failure: java.util.concurrent.CancellationException)
	at io.netty.util.concurrent.DefaultPromise.setSuccess(DefaultPromise.java:97)
	at nl.jk5.mqtt.MqttChannelHandler.handlePuback(MqttChannelHandler.java:215)
	at nl.jk5.mqtt.MqttChannelHandler.channelRead0(MqttChannelHandler.java:37)
	at nl.jk5.mqtt.MqttChannelHandler.channelRead0(MqttChannelHandler.java:11)
...`

I had understood that this had been fixed in these new versions.

closed 2019-12-31 by zbeacon