thingsboard/thingsboard-gateway #115: Sigfox configuration

gigi130358, 2018-08-29:
I'm trying to use sigfox extension with thingsboard "2.1.1-SNAPSHOT" and Thingsboard Gateway 2.0.4-SNAPSHOT (with netty-mqtt "2.0.0TB) build from source master on ubuntu 18.04.
I'm using [tb-gateway.yml.txt](https://github.com/thingsboard/thingsboard-gateway/files/2331475/tb-gateway.yml.txt) and I uploaded in gw extension this modified sigfox-gui-extension-configuration.json file 
[sigfox-gui-extension-configuration-1.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/2331491/sigfox-gui-extension-configuration-1.json.txt)
In order to check the functionality without wait for the sigfox backend (a msh every 5 hours ...) I sent a curl messsagge like this
$ curl  -H "Content-Type: application/json" --header "Authorization:Basic U0lHRk9YX1RFU1RfVE9LRU4=" -d "{"device": "{device}","lat": "{lat}","lng": "{lng}"}" -X POST "http://localhost:9090/sigfox/5a0ecfa95005747c772e768e/"
and I receive this answer
{"message":"Exception occurred while parsing json request body [{device: {device},lat: {lat},lng: {lng}}]"}
Where I'm wrong ?
Thanks

gigi130358, 2018-08-30:
The configuration files below work for the implementation of two gateway extensions (MQTT for TTN and HTTP for Sigfox (both simulated by curl and received form sigfox backend)).
[Gateway configuration-mqtt-httpAWS-5.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/2335500/Gateway.configuration-mqtt-httpAWS-5.json.txt)
[tb-gateway.yml-5.txt](https://github.com/thingsboard/thingsboard-gateway/files/2335476/tb-gateway.yml-5.txt)
[curl-5.txt](https://github.com/thingsboard/thingsboard-gateway/files/2335480/curl-5.txt)

Now I have to decode the related "data" field.

mp-loki, 2018-08-30:
Can you provide your tb-gateway.log (or the portion where the exception occurs)?

mp-loki, 2018-08-30:
You mean that you want to decode 'data': 'c90402010509f0e18901' in gateway? Is it Base64 encoded?
I don't think that we have such functionality right now

gigi130358, 2018-08-30:
Yes, we currently store this raw "c90402010509f0e18901" payload in AWS DynamoDB and decoded it with a quite complex js code in a AWS web site.
Could be possible to define a custom transformer in the extension  that work with js code instead of json ?

zbeacon, 2020-01-02:
Hi, @gigi130358 

Thank you for your feedback. I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:

    Less footprint and memory impact
    Easier to extend and customize
    Improved functionality.

We don't have Sigfox protocol support in the new gateway. Perhaps we will make Sigfox protocol support in one of the following releases.

closed 2020-01-02 by zbeacon