thingsboard/thingsboard-gateway #120: OPC UA extension - NPE when None security

mike0sv, 2018-09-10:
The problem is at org.thingsboard.gateway.extensions.opc.OpcUaServerMonitor:85 - you try to load certificates even if security policy is "None" and since there is no certs, it yields NPE.

I configured extension from GUI, and I noticed that even when you choose "None" security policy (and GUI hides cert configuration), gateway still receives leftover configuration in the keystore field. In my case there was null value in the fileContent field, which got me this NPE.

mp-loki, 2018-09-11:
@mike0sv Thank you for pointing this. This is a known issue and it will be fixed in next minor release.

mp-loki, 2018-09-13:
Fixed
https://github.com/thingsboard/thingsboard-gateway/releases/tag/2.1.1

closed 2018-09-13 by mp-loki