thingsboard/thingsboard-gateway #121: Some messages not forewared to thingsboard. Mqtt connection dropping ?

sebsoft, 2018-09-13:
I'm running Gateway (v2.0.4-SNAPSHOT)  and TB (v2.1.1-SNAPSHOT) on the same windows 10 machine.  Using the http extension i post identical messages to http://localhost:9090/uplink/ttn/  .  About 10 out of 50 result in an error.  See below for gateway logs in case of messages which get through and messages which do not arrive at TB.   (Also note that the HTTP response is also rather slow.)

In case of NO error this is found in the gateway log file:

`2018-09-13 11:16:39,043 [http-nio-0.0.0.0-9090-exec-2] INFO  o.t.g.s.gateway.MqttGatewayService - [mydevid2] Device Connected!
2018-09-13 11:16:39,084 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"device":"mydevid2","type":"mydevid2"}, timestamp=0, topic='v1/gateway/connect'id=cfa0176a-04d4-406e-b79a-0402853197b2, messageId=7}]
2018-09-13 11:16:39,084 [pool-2-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-09-13 11:16:39,086 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-09-13 11:16:39,112 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [mydevid2][mydevid2][7] Device connect event is reported to Thingsboard!
2018-09-13 11:16:39,113 [http-nio-0.0.0.0-9090-exec-2] TRACE o.t.g.s.gateway.MqttGatewayService - [mydevid2][8] Updating device attributes: [StringDataEntry{value='0102030405060708'} BasicKvEntry{key='hardware_serial'}]
2018-09-13 11:16:39,121 [http-nio-0.0.0.0-9090-exec-2] TRACE o.t.g.s.gateway.MqttGatewayService - [mydevid2][9] Updating device telemetry: [BasicTsKvEntry{ts=1536830199015, kv=LongDataEntry{value=30} BasicKvEntry{key='counter'}}]
2018-09-13 11:16:40,086 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"mydevid2":{"hardware_serial":"0102030405060708"}}, timestamp=0, topic='v1/gateway/attributes'id=ee77d9e2-5f37-4b9c-b7bc-ce2cff8fddf5, messageId=8}]
2018-09-13 11:16:40,087 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"mydevid2":[{"ts":1536830199015,"values":{"counter":30}}]}, timestamp=0, topic='v1/gateway/telemetry'id=8f48a09c-899e-42a3-ab9c-d32b197d7305, messageId=9}]
2018-09-13 11:16:40,090 [pool-2-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [2] messages are still in progress
2018-09-13 11:16:40,090 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-09-13 11:16:40,150 [pool-3-thread-1] DEBUG o.t.g.s.gateway.MqttGatewayService - [mydevid2][8] Device attributes were delivered!
2018-09-13 11:16:40,150 [pool-3-thread-1] DEBUG o.t.g.s.gateway.MqttGatewayService - [9][mydevid2] Device telemetry published to ThingsBoard!
2018-09-13 11:16:40,151 [http-nio-0.0.0.0-9090-exec-2] INFO  o.t.g.s.gateway.MqttGatewayService - [mydevid2][10] Device Disconnected!
2018-09-13 11:16:41,091 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"device":"mydevid2"}, timestamp=0, topic='v1/gateway/disconnect'id=763f7510-42d0-462a-862a-b8b71fcd24ae, messageId=10}]
2018-09-13 11:16:41,093 [pool-2-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-09-13 11:16:41,093 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-09-13 11:16:41,118 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [mydevid2][10] Device disconnect event is delivered!
`

In case of error this is found in the log file:

`2018-09-13 11:17:31,491 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"device":"mydevid2","type":"mydevid2"}, timestamp=0, topic='v1/gateway/connect'id=0f91a0ae-6623-44c7-b129-4f88d0ae1952, messageId=59}]
2018-09-13 11:17:31,491 [pool-2-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress
2018-09-13 11:17:31,492 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-09-13 11:17:31,493 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [mydevid2][mydevid2][59] Device connect event is reported to Thingsboard!
2018-09-13 11:17:31,493 [http-nio-0.0.0.0-9090-exec-5] TRACE o.t.g.s.gateway.MqttGatewayService - [mydevid2][60] Updating device attributes: [StringDataEntry{value='0102030405060708'} BasicKvEntry{key='hardware_serial'}]
2018-09-13 11:17:31,494 [http-nio-0.0.0.0-9090-exec-5] TRACE o.t.g.s.gateway.MqttGatewayService - [mydevid2][61] Updating device telemetry: [BasicTsKvEntry{ts=1536830250563, kv=LongDataEntry{value=30} BasicKvEntry{key='counter'}}]
2018-09-13 11:17:32,493 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"mydevid2":{"hardware_serial":"0102030405060708"}}, timestamp=0, topic='v1/gateway/attributes'id=9a994392-8521-4320-87bc-c53c5e893dd7, messageId=60}]
2018-09-13 11:17:32,496 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"mydevid2":[{"ts":1536830250563,"values":{"counter":30}}]}, timestamp=0, topic='v1/gateway/telemetry'id=2a80e077-e55c-421c-9fff-393f696de92d, messageId=61}]
2018-09-13 11:17:32,513 [pool-3-thread-1] DEBUG o.t.g.s.gateway.MqttGatewayService - [mydevid2][60] Device attributes were delivered!
2018-09-13 11:17:32,541 [pool-3-thread-1] DEBUG o.t.g.s.gateway.MqttGatewayService - [61][mydevid2] Device telemetry published to ThingsBoard!
2018-09-13 11:17:32,542 [http-nio-0.0.0.0-9090-exec-5] INFO  o.t.g.s.gateway.MqttGatewayService - [mydevid2][62] Device Disconnected!
2018-09-13 11:17:32,543 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='mydevid2', payload={"device":"mydevid2"}, timestamp=0, topic='v1/gateway/disconnect'id=bd36346f-0191-44da-a498-e9c343caef85, messageId=62}]
2018-09-13 11:17:32,545 [nioEventLoopGroup-2-1] WARN  i.n.channel.DefaultChannelPipeline - An exceptionCaught() event was fired, and it reached at the tail of the pipeline. It usually means the last handler in the pipeline did not handle the exception.
java.lang.NullPointerException: null
	at nl.jk5.mqtt.MqttChannelHandler.handlePuback(MqttChannelHandler.java:215)
	at nl.jk5.mqtt.MqttChannelHandler.channelRead0(MqttChannelHandler.java:37)
	at nl.jk5.mqtt.MqttChannelHandler.channelRead0(MqttChannelHandler.java:11)
	at io.netty.channel.SimpleChannelInboundHandler.channelRead(SimpleChannelInboundHandler.java:105)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at nl.jk5.mqtt.MqttPingHandler.channelRead(MqttPingHandler.java:39)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.handler.timeout.IdleStateHandler.channelRead(IdleStateHandler.java:286)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
	at io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:310)
	at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:284)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:362)
	at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:348)
	at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:340)
[... cut ...]
	at org.apache.coyote.AbstractProtocol$ConnectionHandler.process(AbstractProtocol.java:802)
	at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.doRun(NioEndpoint.java:1410)
	at org.apache.tomcat.util.net.SocketProcessorBase.run(SocketProcessorBase.java:49)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61)
	at java.lang.Thread.run(Unknown Source)`

sebsoft, 2018-09-18:
Did some further digging , seems the error is related to the puback sent by the thingsboard server.  In this case the http extension keeps waiting, hits the timeout and telemetry is lost.

For the moment i got rid of the errors by avoiding the puback  in MqttMessageSender (fire and forget)

` private Future<Void> publishMqttMessage(MqttPersistentMessage message) {
       
	   // return tbClient.publish(message.getTopic(), Unpooled.wrappedBuffer(message.getPayload()), MqttQoS.AT_LEAST_ONCE).addListener(
       // changed SRA
	   return tbClient.publish(message.getTopic(), Unpooled.wrappedBuffer(message.getPayload()), MqttQoS.AT_MOST_ONCE).addListener(
            	   future -> incomingQueue.put(new MessageFuturePair(future, message))
        );
    }
`

zbeacon, 2019-12-31:
Hi, @sebsoft 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway.

We look forward to your feedback.

closed 2019-12-31 by zbeacon

DanielRichart, 2020-08-13:
Hello,

I'm facing a similar issue in my tb-gateway, where it suddenly stop receiving messages and only receives again after I remove the mqtt extension config file, restart the tb-gateway service and put the mqtt config file back in.

In the logs, i receive lot of lines with:
[pool-3-thread-1] INFO o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
[pool-3-thread-1] INFO o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [1] messages are still in progress

I wish to try the modification in the MqttMessageSender.java file described above, but **i can't find this file in my system**.

**Please, where can i find it?**

Thanks a lot,
Daniel Richart