thingsboard/thingsboard-gateway #127: there is no proper version of netty-mqtt for thingsboard-gateway compilation

hawkwang, 2018-10-23:
2.0.0TB is needed, but current is 2.0.1TB

mp-loki, 2018-10-23:
You can update the TB version in your thingsboard-gateway/pom.xml

stieuquy, 2018-10-25:
@mp-loki  there is compile error when I use netty-mqtt v2.0.1TB.
Because of **create** method the class **MqttClient**. In v2.0.0TB the method have only one argument, but with v2.0.1TB it has two arguments

stieuquy, 2018-11-05:
@felix-ma officially not, I just temporary implement (write a warning log) these methods base on API document.

mp-loki, 2018-11-06:
Build netty-mqtt from our repository first:
https://github.com/thingsboard/netty-mqtt

mp-loki, 2018-12-26:
In ThingsBoard v2.2 netty-mqtt is included in the main project, so now you don't have to build netty-mqtt separately. Just build ThingsBoard and then build thingsboard-gateway.

closed 2018-12-26 by mp-loki