thingsboard/thingsboard-gateway #130: remoteConfiguration configuration not effective

felix-ma, 2018-10-24:
thingsboard conf:
```
mqtt:
  # Enable/disable mqtt transport protocol.
  enabled: "${MQTT_ENABLED:true}"
  bind_address: "${MQTT_BIND_ADDRESS:0.0.0.0}"
  bind_port: "${MQTT_BIND_PORT:8883}"
  adaptor: "${MQTT_ADAPTOR_NAME:JsonMqttAdaptor}"
  timeout: "${MQTT_TIMEOUT:10000}"
  netty:
    leak_detector_level: "${NETTY_LEASK_DETECTOR_LVL:DISABLED}"
    boss_group_thread_count: "${NETTY_BOSS_GROUP_THREADS:1}"
    worker_group_thread_count: "${NETTY_WORKER_GROUP_THREADS:12}"
    max_payload_size: "${NETTY_MAX_PAYLOAD_SIZE:65536}"
  # MQTT SSL configuration
  ssl:
    # Enable/disable SSL support
    enabled: "${MQTT_SSL_ENABLED:true}"
    # SSL protocol: See http://docs.oracle.com/javase/8/docs/technotes/guides/security/StandardNames.html#SSLContext
    protocol: "${MQTT_SSL_PROTOCOL:TLSv1.2}"
    # Path to the key store that holds the SSL certificate
    key_store: "${MQTT_SSL_KEY_STORE:mqttserver.jks}"
    # Password used to access the key store
    key_store_password: "${MQTT_SSL_KEY_STORE_PASSWORD:server_ks_password}"
    # Password used to access the key
    key_password: "${MQTT_SSL_KEY_PASSWORD:server_key_password}"
    # Type of the key store
    key_store_type: "${MQTT_SSL_KEY_STORE_TYPE:JKS}"

```

tb-gateway conf:
```
server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"

gateways:
  tenants:
    -
      label: "Tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: "${GATEWAY_HOST:192.168.116.141}"
        port: 8883
        retryInterval: 3000
        maxInFlight: 1000
        security:
#          accessToken: "${GATEWAY_ACCESS_TOKEN:YOUR_TOKEN}"
          keystore: "mqttgateway.jks"
          keystorePassword: "password"
          keystoreKeyAlias: "gatewayalias"
          truststore: "mqttgateway.jks"
          truststorePassword: "password"
      remoteConfiguration: true
      extensions:
        -
          id: "http"
          type: "HTTP"
          extensionConfiguration: http-config.json

```
Nearly connected to ThingsBoard
“Latest Telemetry” tab  “devicesOnline”, “attributesUploaded” and “telemetryUploaded”. All values “0”.
but "Extensions” tab "Status" is "Not sync" ,"Last sync time" is "Not available"

Attached is the tb_gateway startup log.

felix-ma, 2018-10-24:
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/2510751/tb-gateway.txt)

felix-ma, 2018-10-24:
I saved an extension on the page and synchronized it.

closed 2018-10-24 by felix-ma

BIT-masterpy, 2018-11-15:
Did you solve it?What causes it?

felix-ma, 2018-11-15:
> Did you solve it?What causes it?

Because I did not configure extensions at that time and did not understand the principles of extension implementation. Just create an extension inside the IP and port are not configured. The default is localhost and I do not have mqtt service locally. So it has been unsynchronized.

felix-ma, 2018-11-15:
When I ran the source code of the gateway backstage, I found that the configuration was read because the connection failed to connect to the mqtt service and the configuration had not been effective.
Later, when I started the mqtt service and configured the corresponding IP and port, the configuration came into effect.