thingsboard/thingsboard-gateway #133: Port [ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]

yangchengjian, 2018-10-26:
```shell
2018-10-26 15:31:27,529 [pool-1-thread-12] INFO  o.t.g.s.gateway.MqttGatewayService - Configuration updates arrived! {"configuration":"[{\"id\":\"modbus\",\"type\":\"MODBUS\",\"configuration\":{\"servers\":[{\"transport\":{\"type\":\"rtu\",\"timeout\":3000,\"portName\":\"ttyUSB0\",\"encoding\":\"rtu\",\"parity\":\"none\",\"baudRate\":9600,\"dataBits\":8,\"stopBits\":1},\"devices\":[]}]}}]"}
2018-10-26 15:31:27,530 [pool-1-thread-12] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration
2018-10-26 15:31:27,543 [pool-1-thread-12] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [modbus][MODBUS]
2018-10-26 15:31:27,544 [pool-1-thread-12] INFO  o.t.g.e.modbus.DefaultModbusService - [Tenant] Initializing Modbus service
2018-10-26 15:31:27,545 [pool-1-thread-12] DEBUG o.t.g.e.modbus.DefaultModbusService - [Tenant] Modbus service configuration [ModbusConfiguration(servers=[ModbusServerConfiguration(transport=ModbusRtuTransportConfiguration(portName=ttyUSB0, timeout=3000, encoding=rtu, baudRate=9600, dataBits=8, stopBits=1.0, parity=none), devices=[])])]
2018-10-26 15:31:28,066 [pool-1-thread-12] ERROR o.t.g.extensions.modbus.ModbusClient - MBS[ttyUSB0] connection failed!
java.io.IOException: Port [ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:127)
	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:80)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:110)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.thingsboard.gateway.extensions.modbus.DefaultModbusService.init(DefaultModbusService.java:64)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.updateConfiguration(MqttGatewayService.java:520)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.onGatewayAttributesUpdate(MqttGatewayService.java:512)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$onMessage$17(MqttGatewayService.java:429)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
2018-10-26 15:31:28,068 [pool-1-thread-12] ERROR o.t.g.e.modbus.DefaultModbusService - [Tenant] Modbus service initialization failed
java.lang.RuntimeException: MBS connection failed!
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:119)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.thingsboard.gateway.extensions.modbus.DefaultModbusService.init(DefaultModbusService.java:64)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.updateConfiguration(MqttGatewayService.java:520)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.onGatewayAttributesUpdate(MqttGatewayService.java:512)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$onMessage$17(MqttGatewayService.java:429)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Port [ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:127)
	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:80)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:110)
	... 12 common frames omitted
2018-10-26 15:31:28,075 [pool-1-thread-12] INFO  o.t.g.service.TenantServiceRegistry - Failed to read configuration attribute
java.lang.RuntimeException: MBS connection failed!
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:119)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.thingsboard.gateway.extensions.modbus.DefaultModbusService.init(DefaultModbusService.java:64)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.updateConfiguration(MqttGatewayService.java:520)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.onGatewayAttributesUpdate(MqttGatewayService.java:512)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$onMessage$17(MqttGatewayService.java:429)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Port [ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:127)
	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:80)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:110)
	... 12 common frames omitted
2018-10-26 15:31:28,078 [pool-1-thread-12] WARN  o.t.g.s.gateway.MqttGatewayService - Failed to update extension configurations [[]]
java.lang.RuntimeException: Failed to update configuration
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:99)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.updateConfiguration(MqttGatewayService.java:520)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.onGatewayAttributesUpdate(MqttGatewayService.java:512)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$onMessage$17(MqttGatewayService.java:429)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.RuntimeException: MBS connection failed!
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:119)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.thingsboard.gateway.extensions.modbus.DefaultModbusService.init(DefaultModbusService.java:64)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	... 9 common frames omitted
Caused by: java.io.IOException: Port [ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:127)
	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:80)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:110)
	... 12 common frames omitted
2018-10-26 15:31:28,189 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='null', payload={"modbusExtensionError":"java.lang.RuntimeException: MBS connection failed!\n\tat org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusCl [... cut ...]
2018-10-26 15:31:28,191 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='null', payload={"modbusExtensionStatus":"Failure"}, timestamp=0, topic='v1/devices/me/telemetry'id=696496b2-0fcc-4ab7-a08a-182e7d8ea767, messageId=116}]
2018-10-26 15:31:28,193 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-10-26 15:31:28,250 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageReceiver - Successfully sent message: [null]
2018-10-26 15:31:28,256 [pool-3-thread-1] DEBUG o.t.g.service.MqttMessageReceiver - Successfully sent message: [null]
```

yangchengjian, 2018-10-26:
Why even "ttyUSB0" in Valid ports but still reporting an error?

steveohara, 2018-10-28:
Shouldn't you be using "/dev/ttyUSB0" as the port identifier?

yangchengjian, 2018-10-28:
@steveohara  I tried it too, but reported the same error
```sh
2018-10-28 11:03:37,478 [pool-1-thread-2] INFO  o.t.g.s.gateway.MqttGatewayService - Configuration updates arrived! {"configuration":"[{\"id\":\"mod\",\"type\":\"MODBUS\",\"configuration\":{\"servers\":[{\"transport\":{\"type\":\"rtu\",\"timeout\":3000,\"portName\":\"/dev/ttyUSB0\",\"encoding\":\"rtu\",\"parity\":\"none\",\"baudRate\":9600,\"dataBits\":8,\"stopBits\":1},\"devices\":[]}]}}]"}
2018-10-28 11:03:37,973 [pool-1-thread-2] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration
2018-10-28 11:03:37,981 [pool-1-thread-2] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [mod][MODBUS]
2018-10-28 11:03:37,982 [pool-1-thread-2] INFO  o.t.g.e.modbus.DefaultModbusService - [Tenant] Initializing Modbus service
2018-10-28 11:03:37,984 [pool-1-thread-2] DEBUG o.t.g.e.modbus.DefaultModbusService - [Tenant] Modbus service configuration [ModbusConfiguration(servers=[ModbusServerConfiguration(transport=ModbusRtuTransportConfiguration(portName=/dev/ttyUSB0, timeout=3000, encoding=rtu, baudRate=9600, dataBits=8, stopBits=1.0, parity=none), devices=[])])]
2018-10-28 11:03:38,004 [pool-1-thread-2] ERROR o.t.g.extensions.modbus.ModbusClient - MBS[/dev/ttyUSB0] connection failed!
java.io.IOException: Port [/dev/ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:127)
	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:96)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:110)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.thingsboard.gateway.extensions.modbus.DefaultModbusService.init(DefaultModbusService.java:64)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.updateConfiguration(MqttGatewayService.java:520)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.onGatewayAttributesUpdate(MqttGatewayService.java:512)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$onMessage$17(MqttGatewayService.java:429)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
2018-10-28 11:03:38,006 [pool-1-thread-2] ERROR o.t.g.e.modbus.DefaultModbusService - [Tenant] Modbus service initialization failed
java.lang.RuntimeException: MBS connection failed!
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:119)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.thingsboard.gateway.extensions.modbus.DefaultModbusService.init(DefaultModbusService.java:64)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.updateConfiguration(MqttGatewayService.java:520)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.onGatewayAttributesUpdate(MqttGatewayService.java:512)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$onMessage$17(MqttGatewayService.java:429)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: Port [/dev/ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:127)
	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:96)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:110)
	... 12 common frames omitted
2018-10-28 11:03:38,010 [pool-1-thread-2] INFO  o.t.g.service.TenantServiceRegistry - Failed to read configuration attribute
java.lang.RuntimeException: MBS connection failed!
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:119)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.thingsboard.gateway.extensions.modbus.DefaultModbusService.init(DefaultModbusService.java:64)
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.updateConfiguration(MqttGatewayService.java:520)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.onGatewayAttributesUpdate(MqttGatewayService.java:512)
[... cut ...]
	at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:83)
	... 9 common frames omitted
Caused by: java.io.IOException: Port [/dev/ttyUSB0] cannot be opened or does not exist - Valid ports are: [ttyAMA0,ttyUSB0,ttyUSB1]
	at com.ghgande.j2mod.modbus.net.SerialConnection.open(SerialConnection.java:127)
	at com.ghgande.j2mod.modbus.facade.ModbusSerialMaster.connect(ModbusSerialMaster.java:96)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.connect(ModbusClient.java:110)
	... 12 common frames omitted

```

steveohara, 2018-10-28:
If you login to the device shell, can you access the port from the command line?
Does it in fact exist? Does the user that you application is running as have privileges to open the port? Is the port being used by another process?
If you run modpoll, do you get the same results?

steveohara, 2018-10-29:
j2mod uses [jSerialComm ](https://github.com/Fazecast/jSerialComm) to handle access to the serial port and this error originates from there.
I would take a look at the source code of `SerialPort.openPort` to see what the problem might be.
You might find yourself having to look at the native code [here](https://github.com/Fazecast/jSerialComm/blob/master/src/main/c/Linux/SerialPort_Linux.c) as well because jSerialComm is a JNI library.

Does the problem go away if you run as root?

hasikp1, 2019-08-23:
Just pass port as ttyUSB0, and change ownership and group in all files of  /usr/share/tb-gateway to root

closed 2019-09-03 by yangchengjian