thingsboard/thingsboard-gateway #142: Thingsboard-gateway in docker

adamfadil, 2018-11-16:
how to run thingsboard-gateway in docker

Jamesxu182, 2018-11-17:
There are some things you need to do before you run thingsboard in docker-compose

* download __thingsboard-gateway.deb__ and put it in docker folder
``` sh
wget -O tb-gateway.deb https://github.com/thingsboard/thingsboard-gateway/releases/download/v1.4.0.1/tb-gateway-1.4.0.deb 
```

* then, build the thingsboard docker image and tag it like thingsboard/gateway:1.2.1
``` sh
docker build -t thingsboard/gateway:1.2.1 .
```

* finally, run __docker-compose.yml__ file
``` sh
docker-compose up -d
```

good luck

adamfadil, 2018-11-17:
> 
> 
> There are some things you need to do before you run thingsboard in docker-compose
> 
>     * download **thingsboard-gateway.deb** and put it in docker folder
> 
> 
> ```shell
> wget -O tb-gateway.deb https://github.com/thingsboard/thingsboard-gateway/releases/download/v1.4.0.1/tb-gateway-1.4.0.deb 
> ```
> 
>     * then, build the thingsboard docker image and tag it like thingsboard/gateway:1.2.1
> 
> 
> ```shell
> docker build -t thingsboard/gateway:1.2.1 .
> ```
> 
>     * finally, run **docker-compose.yml** file
> 
> 
> ```shell
> docker-compose up -d
> ```
> 
> good luck

Thank you very much, I was able to realize the first step, namely the docker image generation. but the step  to run docker-compose.yml file displays this error.
[image]

adamfadil, 2018-11-17:
> There are some things you need to do before you run thingsboard in docker-compose
>
>     * download **thingsboard-gateway.deb** and put it in docker folder
>
>
> ```shell
> wget -O tb-gateway.deb [...]

Thank you very much.

closed 2018-11-17 by adamfadil