thingsboard/thingsboard-gateway #146: Disable data buffering in gateway

JozsefGuyas, 2018-11-21:
Hi,
Can I disable the buffering in the tbgateway?

mp-loki, 2018-11-22:
You can decrease the buffer size, but you cannot disable the buffer, because that approach has proven to cause data loss.

JozsefGuyas, 2018-11-22:
what will happen if i wiil decrase the buffer size?

mp-loki, 2018-12-26:
it will just decrease the number of messages stored in a single file, thus increasing the number of files. You may want to do this for performance reasons

zbeacon, 2019-12-31:
Hi, @JozsefGuyas 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway.
The new gateway supports "memory" storage and you can set it size.
You can read more [here](https://thingsboard.io/docs/iot-gateway/configuration/#storage-configuration).

closed 2019-12-31 by zbeacon