thingsboard/thingsboard-gateway #150: How sends the client and shared attributes to device through the Thingsboard Gateway

amarras63, 2018-12-06:
I'm trying to use the thingsboard gateway. I have no problems sending data from a device via the gateway to thingsboard using mqtt: the device is created and the telemetry data sent.
I can not read and send device attributes (client and shared attributes).
It's possible to do it?
I need a sample code where the device (eg Arduino or an ESP) is connected to the Thingsboard server via the Thingsboard Gateway (for example installed in a Raspberry) and receives and sends the client and shared attributes through the gateway (in addition to the telemetry that I can already send).

mp-loki, 2018-12-08:
What gateway extension are you using?

amarras63, 2018-12-08:
Hi Valerij,
I used the default configuration with the file
https://thingsboard.io/docs/iot-gateway/resources/mqtt-gui-extension-configuration.json

zbeacon, 2019-12-31:
Hi, @amarras63 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
 - Less footprint and memory impact
 - Easier to extend and customize
 - Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway.

We look forward to your feedback.

closed 2019-12-31 by zbeacon