thingsboard/thingsboard-gateway #153: I can't subscribe to v1/gateway/attributes/response

Tasosgi, 2018-12-11:
I am trying to get shared device attributes from ThingsBoard server node. The first step is to subscribe to "v1/gateway/attributes/response" but these is not possible. I can subscribe to other topics like "v1/gateway/attributes" but to to "v1/gateway/attributes/response".

I am using Eclipse Paho MQTT C client but also tested with MQTT.js
[image]

Tasosgi, 2018-12-12:
After investigation I found out the following

When I subscribe to "v1/gateway/attributes" I receive the SUBACK packet 
**0x92 0x03 0x00 0x02 0x00**
These is fine. The last byte means "success QoS0 granted"

When I subscribe to "v1/gateway/attributes/response" I receive the SUBACK packet
**0x92 0x03 0x00 0x02 0x80**
The last byte means "Failure, not QoS granted". Although after that I am subscribed to the topic, various issues raised in Paho MQTT C client for embedded platform which I use.

kpranga, 2019-02-07:
Hi, 
I have the same problem, can't subscribe to any topic like rpc and attributes. 
I have updated TB to v2.3 and TB-gateway to v2.2.1. and Still mosquitto on windows and Node-RED on linux disconnect me after connect.
Anyone has the solution for this problem?

biagio-ra, 2019-07-12:
I am not able to subscribe to any of the topics either. Has anyone found out what is happening?

alzughoul, 2019-07-29:
I cannot MQTT subscribe either what wrong?. Please if you found the solution help us.

biagio-ra, 2019-07-29:
I don't have a solution, but I noticed that this seems to behave like a request/response instead of the MQTT pub/sub.

So, the message that comes back when making an attribute request is a message that has the topic "v1/gateway/attributes/response" to the same client making the request.

I also noticed that ALL of the listening devices get a "ping" of sorts, whether or not they are subscribed to the appropriate topic. However, no data is provided to them.

alzughoul, 2019-08-01:
Thanks Biagio-ra for the update. I use the live demo i.e., demo.thingsboards.io for this project and I can MQTT connect and post just fine. I tried to subscribe to rpc commands from the control widget but when I check the switch type filter in rules chain (in debug mode) it only records the connect and post events. Subscribtion never reached TB apparently. 

However, I always subscribe to "v1/devices/me/rpc/request/+" and wondering if I should actually subscribe to v1/gateway/rpc for example? would that be my problem??

Many Thanks

closed 2020-02-14 by zbeacon