thingsboard/thingsboard-gateway #154: Can't set deviceTypeJsonExpression  from mqtt-config.json

projectmentor, 2018-12-19:
using tb-gateway 1.4.0 with mqtt-config.json against demo.thingsboard.io.  When I connect first time with new device, it DOES create a new device in Thingsboard but DOES NOT change the deviceType  attribute from "default".  

[image]

mqtt-config.json:
```
...
          "topicFilter": "sensor/+/mtdotbox",
          "converter": {
            "type": "json",
            "filterExpression": "",
            "deviceNameTopicExpression": "(?<=sensor\/)(.*?)(?=\/mtdotbox)",
            "deviceTypeJsonExpression": "${$.model}",  // <--- this should work?!?
            "timeout": 60000,
            "attributes": [
...
              {
                "type": "string",
                "key": "model",
                "value": "${$.model}"
...

Here is the data I am sending to the tb-gateway from my local gateway
```
{ 
    "serialNumber": "00-08-00-4A-38-A3",
     "model": "MT-DOTBOX-G-915",
     "light": 972.75, 
     "temperature": 25.1875, 
      "humidity": 101.6765,
      "x_acc": 0,
      "y_acc": 0, 
      "z_acc": 1
 }
```

```   
Here is the tb-gateway log showing connect. attributes, telemetry. Seems ok.
```
2018-12-19 13:50:33,224 [MQTT Call: cfa9d2cb-09fe-4512-a869-5ba9b2694550] INFO  o.t.g.s.gateway.MqttGatewayService - [00-08-00-4A-38-A3] Device Connected!
2018-12-19 13:50:33,691 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Sending message [MqttPersistentMessage(id=25c321df-6f6b-4f8e-b5e2-700921d43a18, timestamp=0, deviceId=00-08-00-4A-38-A3, messageId=3, topic=v1/gateway/connect, payload=[123, 34, 100, 101, 
118, 105, 99, 101, 34, 58, 34, 48, 48, 45, 48, 56, 45, 48, 48, 45, 52, 65, 45, 51, 56, 45, 65, 51, 34, 125])]
2018-12-19 13:50:33,693 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Sending message [MqttPersistentMessage(id=ff218082-100a-4890-aed6-7eeacc183180, timestamp=0, deviceId=00-08-00-4A-38-A3, messageId=4, topic=v1/gateway/attributes, payload=[123, 34, 48, 48, 45, 48, 56, 45, 48, 48, 45, 52, 65, 45, 51, 56, 45, 65, 51, 34, 58, 123, 34, 115, 101, 114, 105, 97, 108, 78, 117, 109, 98, 101, 114, 34, 58, 34, 48, 48, 45, 48, 56, 45, 48, 48, 45, 52, 65, 45, 51, 56, 45, 65, 51, 34, 44, 34, 109, 111, 100, 101, 108, 34, 58, 34, 77, 84, 45, 68, 79, 84, 66, 79, 88, 45, 71, 45, 57, 49, 53, 34, 125, 125])]
2018-12-19 13:50:33,693 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Sending message [MqttPersistentMessage(id=6a1ffaf9-a855-489c-8c94-533ad869c4aa, timestamp=0, deviceId=00-08-00-4A-38-A3, messageId=5, topic=v1/gateway/telemetry, payload=[123, 34, 48, 48, 45, 48, 56, 45, 48, 48, 45, 52, 65, 45, 51, 56, 45, 65, 51, 34, 58, 91, 123, 34, 116, 115, 34, 58, 49, 53, 52, 53, 50, 50, 55, 52, 51, 51, 50, 48, 49, 44, 34, 118, 97, 108, 117, 101, 115, 34, 58, 123, 34, 108, 105, 103, 104, 116, 34, 58, 57, 55, 50, 46, 55, 53, 44, 34, 116, 101, 109, 112, 101, 114, 97, 116, 117, 114, 101, 34, 58, 50, 53, 46, 49, 56, 55, 53, 44, 34, 104, 117, 109, 105, 100, 105, 116, 121, 34, 58, 49, 48, 49, 46, 54, 55, 54, 53, 44, 34, 120, 95, 97, 99, 99, 34, 58, 48, 46, 48, 44, 34, 121, 95, 97, 99, 99, 34, 58, 48, 46, 48, 44, 34, 122, 95, 97, 99, 99, 34, 58, 49, 46, 48, 125, 125, 93, 125])]
2018-12-19 13:50:33,694 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [3] messages are still in progress
2018-12-19 13:50:33,694 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2018-12-19 13:50:33,892 [pool-4-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [00-08-00-4A-38-A3][3] Device connect event is reported to Thingsboard!

```

mp-loki, 2018-12-26:
Which version of the gateway are you using?

mp-loki, 2018-12-26:
Try upgrading to the latest version. At least it has meaningful logging for  payload

projectmentor, 2018-12-27:
Where can i find upgrade instructions/ procedure?

projectmentor, 2019-01-17:
Thank you for the reply, I am in the process of setting up tb-gateway 2.2.1 using the docker image. And have  different problem which I am troubleshooting now.  As soon as it is fixed I will try your solution and edit this comment. (its just to let you know I got your reply) 

FYI This is the current issue, (which i will move to another thread)
 have a separate docker container for mosquitto broker running on the same host on port 1883. The broker does receive data from the device.  On intiialization the tb-gateway does connect to the thingsboard server.  HOWEVER the tb-gateway fails to subscribe to the mosquitto broker.  The log message I am seeing in tb-gateway is: 

`
[main] DEBUG o.t.g.e.m.client.MqttBrokerMonitor - [localhost:1883] MQTT broker connection attempt!
06:05:03.144 [MQTT Con: 894f6d28-d09d-4f9d-8577-e7d7cacc98db] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:1883] MQTT broker connect
ion faied! Unable to connect to server
org.eclipse.paho.client.mqttv3.MqttException: Unable to connect to server
        at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:94)
        at org.eclipse.paho.client.mqttv3.internal.ClientComms$ConnectBG.run(ClientComms.java:701)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
        at java.net.PlainSocketImpl.socketConnect(Native Method)
        at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
        at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
        at java.net.Socket.connect(Socket.java:589)
        at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:84)
        ... 8 common frames omitted
`

mp-loki, 2019-01-18:
Are mosquitto and ThingsBoard running both on the same host:port (localhost:1883)?

projectmentor, 2019-01-19:
In the above configuration mosquitto was in its own docker container running on 1882 !  -  tb-gateway was running in another container with port 1883 exposed.  Both tb-gateway and mossquitto were running on same docker host ( ubuntu 18.04, Docker 18.06)   and port 1882, 1883 and 9090 were exposed.

the tb-gateway did connect to the thingsboard instance running on a completely separate host, (non-dockerized) ... but what was strange  is this line from tb-gateway log above:
```
 06:05:03.144 [MQTT Con: 894f6d28-d09d-4f9d-8577-e7d7cacc98db] WARN o.t.g.e.m.client.MqttBrokerMonitor - [localhost:1883] MQTT broker connect ion failed!
```
which shows the gateway trying to subscribe to mosquitto on *localhost* which should not work because the tb-gateway docker container did not have a mosquitto broker installed.

Im still in the process of building ver 2.2.1 of both tb-gateway and thingsboard from sources. and just finished the build.  Now I am going to try with newer versions.

But my question is: Where is the configuration setting that tells tb-gateway which port to use for mosquitto?

projectmentor, 2019-01-22:
Fixed:  the environment variable LOADER_PATH must specify the config directory and be available from within the docker container. 

in the docker host environment
```
LOADER_PATH=/usr/share/tb-gateway/conf
```

in the docker-compose.yml
```
### THINGSBOARD GATEWAY ######################################                                  
    thingsboard-gateway:                                                                        
      build:
        context: ./thingsboard-gateway                                                          
      volumes:
        - "${DATA_PATH_HOST}/thingsboard-gateway/conf:${GATEWAY_LOADER_PATH}"
        - "${DATA_PATH_HOST}/thingsboard-gateway/log:${GATEWAY_LOG_PATH}"                                                               
      environment:
        - GATEWAY_ACCESS_TOKEN=${GATEWAY_ACCESS_TOKEN}                                          
        - GATEWAY_HOST=${GATEWAY_HOST}
        - JAVA_OPTS=${GATEWAY_JAVA_OPTS}                                                        
        - LOG_FILENAME=${GATEWAY_LOG_FILENAME}
        - LOG_PATH=${GATEWAY_LOG_PATH}                                                          
        - LOADER_PATH=${GATEWAY_LOADER_PATH}                                                    
        - GATEWAY_VERSION=${GATEWAY_VERSION}
        - CONFIG_ENV=${GATEWAY_CONFIG_ENV}                                                      
        - CONFIG_YML=${GATEWAY_CONFIG_YML}                                                      
        - CONFIG_MQTT=${GATEWAY_CONFIG_MQTT}
      depends_on:
        - mosquitto
      ports:
        - "${GATEWAY_PORT}:9090"                                                                
        - "1883:1883"
        - "1882"
      networks:
        - frontend
        - backend 

```

closed 2019-01-22 by projectmentor

MisterBearBear, 2019-04-20:
Hi,

I had the same problem - deviceTypeJsonExpression had no effect in http-config.json and always returned "default" as the device type. Using 1.4.0 now. How to get this resolved? Will the latest version of Gateway help?