thingsboard/thingsboard-gateway #158: Can't get persistence to work

skangmy, 2019-01-07:
I'm trying to setup thingsboard gateway (1.4.0) with offline storage in case of internet connection lost. However, I noticed that after recovering the internet connection, gateway is not sending any offline data over to the server.

Is my configuration (see below) correct? All of the settings are default except maxInFlight where I changed it to 10000.

`
server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"

# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"

gateways:
  tenants:
    -
      label: "Tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: "myhost.com"
        port: 1883
        retryInterval: 3000
        maxInFlight: 10000
        security:
          accessToken: "<secret>"
      remoteConfiguration: false
      extensions:
        - 
          id: "mqtt"
          type: "MQTT"
          extensionConfiguration: mqtt-config.json
`

Appreciate if someone can point me to the right place on how to set this up properly. Thanks in advance!

mp-loki, 2019-01-15:
Storage is no different between versions 1.4 and 2.1.1. With your configuration, the offline data is stored under the 'storage' dir, 1000 messages per file. Do you see that files are created in the 'storage' and not removed after the connection is recovered? Anything specific in the log file? I suggest you lower the log level to DEBUG and then re-run your test.

mp-loki, 2019-01-17:
Please try new [2.2.1.rc release](https://github.com/thingsboard/thingsboard-gateway/releases/tag/2.2.1rc). instead of 2.1.1

skangmy, 2019-04-15:
Hi, sorry for replying so late as we did not proceed further on thingsboard gateway until recently. We have just tried version 2.2.1rc and is working now.

To me, this issue is solved. Thanks for your help!

closed 2019-04-15 by skangmy