thingsboard/thingsboard-gateway #159: Help me not working gateway

oroi1991, 2019-01-09:
I used TB 2.2.1 and tb-gateway 2.1.1.

The gateway only send reportStats data as bellow.
`01:49:34.848 [pool-4-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1547052574838,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!`

But doesn't send mqtt data.
I don't understand why tb-gateway doesn't send mqtt data.

Please, See attached config files.

oroi1991, 2019-01-16:
TB has got a telemetry data after when i was changing remoteConfiguration value to false in tb-gateway.yml.
But mqtt is repeating connecting and disconnecting the following messages.
`08:20:37.439 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Attempting to reconnect to ThingsBoard.
08:20:37.448 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Successfully reconnected to ThingsBoard.
08:20:37.449 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Outgoing queue is not empty. [2] messages are still in progress
08:20:37.456 [pool-4-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [0_GW_Timon][gw][4] Device connect event is reported to Thingsboard!
08:20:37.456 [nioEventLoopGroup-2-28] WARN  o.t.g.s.gateway.MqttGatewayService - Lost connection to ThingsBoard.
08:20:37.568 [MQTT Call: e4f30b02-9a39-403e-99ec-d6baf86eaaa3] INFO  o.t.g.s.gateway.MqttGatewayService - [0_GW_Timon] Device Connected!
08:20:38.450 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
08:20:41.450 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Attempting to reconnect to ThingsBoard.
08:20:41.475 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Successfully reconnected to ThingsBoard.`

And then i have got another error message.
`08:16:46.618 [pool-4-thread-1] WARN  o.t.g.service.MqttMessageReceiver - Failed to send message [{deviceId='0_GW_Timon', payload={"0_GW_Timon":[{"ts":1547680589727,"values":{"M_D1":9520.0}}]}, timestamp=0, topic='v1/gateway/telemetry', id=54d24eb4-11d9-410d-909a-7dffadcab286, messageId=42}] due to [{}]
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)`

I saw another [issue](https://github.com/thingsboard/thingsboard-gateway/issues/92) that the error was solved latest tb-gateway version.

I have tested two versions of master(2.1.1) and 2.0.4-draft1.

mp-loki, 2019-01-17:
Please try new [2.2.1.rc release](https://github.com/thingsboard/thingsboard-gateway/releases/tag/2.2.1rc).

oroi1991, 2019-01-17:
Thank you for your supporting.
I am going to test new version.

oroi1991, 2019-01-21:
> Please try new [2.2.1.rc release](https://github.com/thingsboard/thingsboard-gateway/releases/tag/2.2.1rc).

I just have tested the new version.
Unfortunately, There are appeared same errors.

`15:34:03.640 [main] INFO  o.a.catalina.core.StandardService - Starting service Tomcat
15:34:03.641 [main] INFO  o.a.catalina.core.StandardEngine - Starting Servlet Engine: Apache Tomcat/8.5.6
15:34:03.738 [localhost-startStop-1] INFO  o.a.c.c.C.[Tomcat].[localhost].[/] - Initializing Spring embedded WebApplicationContext
15:34:03.738 [localhost-startStop-1] INFO  o.s.web.context.ContextLoader - Root WebApplicationContext: initialization completed in 2101 ms
15:34:03.928 [localhost-startStop-1] INFO  o.s.b.w.s.ServletRegistrationBean - Mapping servlet: 'dispatcherServlet' to [/]
15:34:03.932 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'characterEncodingFilter' to: [/*]
15:34:03.932 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'hiddenHttpMethodFilter' to: [/*]
15:34:03.932 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'httpPutFormContentFilter' to: [/*]
15:34:03.932 [localhost-startStop-1] INFO  o.s.b.w.s.FilterRegistrationBean - Mapping filter: 'requestContextFilter' to: [/*]
15:34:04.051 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Tenant] Initializing gateway
15:34:05.174 [main] INFO  o.t.g.s.DefaultTenantManagerService - [Tenant] Initializing extension: [MQTT]
15:34:05.177 [main] INFO  o.t.g.e.m.c.DefaultMqttClientService - [Tenant] Initializing MQTT client service!
15:34:05.273 [nioEventLoopGroup-2-1] WARN  o.t.g.s.gateway.MqttGatewayService - Lost connection to ThingsBoard.
15:34:05.276 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - ThingsBoard MQTT connection failed. Reconnecting in [3000] milliseconds
15:34:05.292 [pool-4-thread-1] WARN  o.t.g.service.MqttMessageReceiver - Failed to send message: [{}]
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)
15:34:05.293 [pool-4-thread-1] WARN  o.t.g.service.MqttMessageReceiver - Failed to send message [{deviceId='0_GW_Timon', payload={"0_GW_Timon":[{"ts":1548051035441,"values":{"M_D1":3936.0}}]}, timestamp=0, topic='v1/gateway/telemetry', id=06f28b04-15d1-4804-b2fb-c9ad1c1a66cd, messageId=77}] due to [{}]
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)
15:34:05.295 [pool-4-thread-1] WARN  o.t.g.service.MqttMessageReceiver - Failed to send message: [{}]
java.util.concurrent.CancellationException: null
	at io.netty.util.concurrent.DefaultPromise.cancel(...)(Unknown Source)`

mp-loki, 2019-01-23:
please provide tb-gateway.yaml, mqtt-config.json (or export gui config) and tb-gateway.log files

closed 2019-12-31 by zbeacon