thingsboard/thingsboard-gateway #160: Telemetry timeseries needs to contain all the key-value pairs #40

aderusha, 2019-01-13:
## Description

This issue is related to [issue 40](https://github.com/thingsboard/thingsboard-gateway/issues/40) that was resolved in 1.2.2 & 1.4.0.  Testing tb-gateway 2.2.0 and I see the issue has returned.

## Environment

**Host OS:** Ubuntu 18.04 LTS
**ThingsBoard:** 2.2.0 Release
**Thingsboard IoT Gateway:** 2.1.1 Release

## Configuration
Clean OS deployment, TB and TB-gateway deployed via the following:

```bash
sudo timedatectl set-timezone America/Detroit
HOSTNAME=thingsboard
sudo hostnamectl set-hostname $HOSTNAME
sudo sed -i 's/preserve_hostname: false/preserve_hostname: true/' /etc/cloud/cloud.cfg
echo "127.0.1.1       $HOSTNAME" | sudo tee --append /etc/hosts
sudo apt-add-repository universe
sudo add-apt-repository ppa:webupd8team/java -y
sudo apt-get update && sudo apt-get upgrade -y
sudo reboot now
echo debconf shared/accepted-oracle-license-v1-1 select true | sudo debconf-set-selections
echo debconf shared/accepted-oracle-license-v1-1 seen true | sudo debconf-set-selections
sudo apt-get install -y oracle-java8-installer 
wget https://github.com/thingsboard/thingsboard/releases/download/v2.2/thingsboard-2.2.deb
sudo dpkg -i thingsboard-2.2.deb
sudo /usr/share/thingsboard/bin/install/install.sh --loadDemo
sudo service thingsboard start
# connect to http://thingsboard:8080 
# user: tenant@thingsboard.org
# pass: tenant
# create gateway https://thingsboard.io/docs/iot-gateway/getting-started/#step-3-gateway-provisioning
wget https://github.com/thingsboard/thingsboard-gateway/releases/download/2.1.1/tb-gateway.deb
sudo dpkg -i tb-gateway.deb
ACCESS_TOKEN=aBcDeFgHiJlKmNoPqRsT
sudo sed -i "s/\${GATEWAY_HOST:.*}/$HOSTNAME/" /etc/tb-gateway/conf/tb-gateway.yml
sudo sed -i "s/\${GATEWAY_ACCESS_TOKEN:.*}/$ACCESS_TOKEN/" /etc/tb-gateway/conf/tb-gateway.yml
sudo apt-get install -y mosquitto mosquitto-clients
mosquitto -p 1884 -d
sudo service tb-gateway start
wget https://thingsboard.io/docs/iot-gateway/resources/mqtt-gui-extension-configuration.json
sudo sed -i "s/1883/1884/" mqtt-gui-extension-configuration.json
# Import extension configuration per https://thingsboard.io/docs/iot-gateway/getting-started/#step-82-mqtt-broker-gui-configuration
```

## Working test case

This example from the docs creates the device and updates telemetry as expected (only mqtt broker port changed from example): `mosquitto_pub -h localhost -p 1884 -t "sensors" -m '{"serialNumber":"SN-001", "model":"T1000", "temperature":36.6}'`

## Non-working test case

Same as above, this time without `temperature` in JSON: `mosquitto_pub -h localhost -p 1884 -t "sensors" -m '{"serialNumber":"SN-002","model":"T1000"}'`

In this test case the device `SN-002` is not created.  `tb-gateway.log` records the following:

```
2019-01-13 22:33:11,952 [MQTT Call: 6cdf0270-66bd-42fd-b22c-f5a1da40aca5] INFO  o.t.g.e.m.c.l.MqttTelemetryMessageListener - [sensors] Failed to decode message: [123, 34, 115, 101, 114, 105, 97, 108, 78, 117, 109, 98, 101, 114, 34, 58, 34, 83, 78, 45, 48, 48, 50, 34, 44, 34, 109, 111, 100, 101, 108, 34, 58, 34, 84, 49, 48, 48, 48, 34, 125]
java.lang.NumberFormatException: For input string: "${$.temperature}"
```

Extending this out to multiple fields yields similar results.  In the previous issue it was suggested to use filter strings, and that also doesn't work due to problems that were outlined in the earlier issue (only one field will ever be recorded when using filters).

moracarlos, 2019-02-05:
Same issue here, any idea?

aderusha, 2019-02-05:
Currently, the only workaround I know of is to revert all the way back to 1.2.2.  All later versions have this problem.  If you need multi-tenancy, I don't know that there is any resolution at all.

moracarlos, 2019-02-05:
Oh I see. Thanks!!

closed 2019-12-31 by zbeacon