thingsboard/thingsboard-gateway #163: Failed to decode message in mqtt

arrowcircle, 2019-01-17:
Hi! I am using TB-gateway version 2.1.1 with thingsboard 2.2.0. I can't make gateway parse messages correctly. Here is errors and messages.

tb-gateway.log
```
2019-01-17 17:00:31,071 [MQTT Call: 8c10e4bf-bc7a-4649-8750-b48ad5f66819] INFO  o.t.g.e.m.c.l.MqttTelemetryMessageListener - [application/2/device/9473730323239372/rx] Failed to decode message: [123, 34, 97, 112, 112, 108, 105, 99, 97, 116, 105, 111, 110, 73, 68, 34, 58, 34, 50, 34, 44, 34, 97, 112, 112, 108, 105, 99, 97, 116, 105, 111, 110, 78, 97, 109, 101, 34, 58, 34, 68, 97, 118, 105, 100, 34, 44, 34, 100, 101, 118, 105, 99, 101, 78, 97, 109, 101, 34, 58, 34, 99, 114, 105, 99, 107, 101, 116, 111, 114, 105, 103, 105, 110, 97, 108, 34, 44, 34, 100, 101, 118, 69, 85, 73, 34, 58, 34, 57, 52, 55, 51, 55, 51, 48, 51, 50, 51, 50, 51, 57, 51, 55, 50, 34, 44, 34, 114, 120, 73, 110, 102, 111, 34, 58, 91, 123, 34, 103, 97, 116, 101, 119, 97, 121, 73, 68, 34, 58, 34, 99, 48, 101, 101, 52, 48, 102, 102, 102, 102, 50, 57, 51, 102, 51, 52, 34, 44, 34, 110, 97, 109, 101, 34, 58, 34, 108, 97, 105, 114, 100, 34, 44, 34, 116, 105, 109, 101, 34, 58, 34, 48, 48, 48, 49, 45, 48, 49, 45, 48, 49, 84, 48, 48, 58, 48, 48, 58, 48, 48, 90, 34, 44, 34, 114, 115, 115, 105, 34, 58, 45, 51, 55, 44, 34, 108, 111, 82, 97, 83, 78, 82, 34, 58, 49, 51, 46, 50, 44, 34, 108, 111, 99, 97, 116, 105, 111, 110, 34, 58, 123, 34, 108, 97, 116, 105, 116, 117, 100, 101, 34, 58, 48, 44, 34, 108, 111, 110, 103, 105, 116, 117, 100, 101, 34, 58, 48, 44, 34, 97, 108, 116, 105, 116, 117, 100, 101, 34, 58, 48, 125, 125, 93, 44, 34, 116, 120, 73, 110, 102, 111, 34, 58, 123, 34, 102, 114, 101, 113, 117, 101, 110, 99, 121, 34, 58, 57, 48, 52, 57, 48, 48, 48, 48, 48, 44, 34, 100, 114, 34, 58, 49, 125, 44, 34, 97, 100, 114, 34, 58, 102, 97, 108, 115, 101, 44, 34, 102, 67, 110, 116, 34, 58, 49, 50, 49, 44, 34, 102, 80, 111, 114, 116, 34, 58, 49, 44, 34, 100, 97, 116, 97, 34, 58, 34, 65, 87, 99, 65, 49, 65, 74, 122, 75, 101, 119, 68, 105, 65, 104, 54, 73, 119, 104, 54, 73, 119, 65, 65, 121, 65, 86, 111, 81, 65, 61, 61, 34, 44, 34, 111, 98, 106, 101, 99, 116, 34, 58, 123, 34, 116, 101, 109, 112, 101, 114, 97, 116, 117, 114, 101, 83, 101, 110, 115, 111, 114, 34, 58, 123, 34, 49, 34, 58, 50, 49, 46, 50, 125, 44, 34, 104, 117, 109, 105, 100, 105, 116, 121, 83, 101, 110, 115, 111, 114, 34, 58, 123, 34, 53, 34, 58, 51, 50, 125, 44, 34, 98, 97, 114, 111, 109, 101, 116, 101, 114, 34, 58, 123, 34, 50, 34, 58, 49, 48, 55, 51, 46, 50, 125, 44, 34, 103, 112, 115, 76, 111, 99, 97, 116, 105, 111, 110, 34, 58, 123, 34, 51, 34, 58, 123, 34, 108, 97, 116, 105, 116, 117, 100, 101, 34, 58, 53, 53, 46, 53, 53, 53, 53, 44, 34, 108, 111, 110, 103, 105, 116, 117, 100, 101, 34, 58, 53, 53, 46, 53, 53, 53, 53, 44, 34, 97, 108, 116, 105, 116, 117, 100, 101, 34, 58, 50, 125, 125, 125, 125]
java.lang.NumberFormatException: For input string: "${$.object.temperatureSensor.1}"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
	at java.lang.Long.parseLong(Long.java:578)
	at java.lang.Long.valueOf(Long.java:803)
	at org.thingsboard.gateway.util.converter.BasicJsonConverter.getKvEntry(BasicJsonConverter.java:154)
	at org.thingsboard.gateway.util.converter.BasicJsonConverter.getTsKvEntries(BasicJsonConverter.java:94)
	at org.thingsboard.gateway.extensions.mqtt.client.conf.mapping.MqttJsonConverter.parse(MqttJsonConverter.java:110)
	at org.thingsboard.gateway.extensions.mqtt.client.conf.mapping.MqttJsonConverter.convert(MqttJsonConverter.java:84)
	at org.thingsboard.gateway.extensions.mqtt.client.listener.MqttTelemetryMessageListener.messageArrived(MqttTelemetryMessageListener.java:42)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.deliverMessage(CommsCallback.java:491)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.handleMessage(CommsCallback.java:402)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.run(CommsCallback.java:206)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
```

mqtt message:

```
{
	"applicationID": "2",
	"applicationName": "Test",
	"deviceName": "cricketoriginal",
	"devEUI": "9473730323239372",
	"rxInfo": [{
		"gatewayID": "c0ffffffff293f34",
		"name": "laird",
		"time": "0001-01-01T00:00:00Z",
		"rssi": -36,
		"loRaSNR": 11.2,
		"location": {
			"latitude": 0,
			"longitude": 0,
			"altitude": 0
		}
	}],
	"txInfo": {
		"frequency": 904500000,
		"dr": 1
	},
	"adr": false,
	"fCnt": 106,
	"fPort": 1,
	"data": "AWcA7QJzKewDiAh6Iwh6IwAAyAVoPg==",
	"object": {
		"temperatureSensor": {
			"1": 23.7
		},
		"humiditySensor": {
			"5": 31
		},
		"barometer": {
			"2": 1073.2
		},
		"gpsLocation": {
			"3": {
				"latitude": 55.5555,
				"longitude": 55.5555,
				"altitude": 2
			}
		}
	}
}
```

mqtt-extension.conf

```
[{
	"id": "mqtt",
	"type": "MQTT",
	"configuration": {
		"brokers": [{
			"host": "localhost",
			"port": 1884,
			"ssl": false,
			"retryInterval": 3000,
			"credentials": {
				"type": "basic",
				"username": "username",
				"password": "password"
			},
			"mapping": [{
				"topicFilter": "application/+/device/+/rx",
				"converter": {
					"type": "json",
					"filterExpression": "$.*",
					"deviceNameJsonExpression": "${$.devEUI}",
					"deviceTypeJsonExpression": "${$.applicationID}",
					"timeout": 60000,
					"timeseries": [{
						"type": "double",
						"key": "temperature",
						"value": "${$.object.temperatureSensor.1}"
					}, {
						"type": "double",
						"key": "humidity",
						"value": "${$.object.humiditySensor.5}"
					}, {
						"type": "double",
						"key": "latitude",
						"value": "${$.object.gpsLocation.3.latitude}"
					}, {
						"type": "double",
						"key": "longitude",
						"value": "${$.object.gpsLocation.3.longitude}"
					}]
				}
			}],
			"connectRequests": [],
			"disconnectRequests": [],
			"attributeRequests": [],
			"attributeUpdates": [],
			"serverSideRpc": []
		}]
	}
}]
```

How can I solve these problems?

mp-loki, 2019-01-17:
I think $.object.temperatureSensor.1 is not a valid JSONPath expression. try changing it to $.object.temperatureSensor["1"]. same for humiditySensor, gpsLocation etc

arrowcircle, 2019-01-17:
@mp-loki Same error, but for new key:

2019-01-17 18:17:06,408 [MQTT Call: a32e85a2-3081-4369-8f12-30ba123d210b] INFO  o.t.g.e.m.c.l.MqttTelemetryMessageListener - [application/2/device/9473730323239372/rx] Failed to decode message: [123, 34, 97, 112, 112, 108, 105, 99, 97, 116, 105, 111, 110, 73, 68, 34, 58, 34, 50, 34, 44, 34, 97, 112, 112, 108, 105, 99, 97, 116, 105, 111, 110, 78, 97, 109, 101, 34, 58, 34, 68, 97, 118, 105, 100, 34, 44, 34, 100, 101, 118, 105, 99, 101, 78, 97, 109, 101, 34, 58, 34, 99, 114, 105, 99, 107, 101, 116, 111, 114, 105, 103, 105, 110, 97, 108, 34, 44, 34, 100, 101, 118, 69, 85, 73, 34, 58, 34, 57, 52, 55, 51, 55, 51, 48, 51, 50, 51, 50, 51, 57, 51, 55, 50, 34, 44, 34, 114, 120, 73, 110, 102, 111, 34, 58, 91, 123, 34, 103, 97, 116, 101, 119, 97, 121, 73, 68, 34, 58, 34, 99, 48, 101, 101, 52, 48, 102, 102, 102, 102, 50, 57, 51, 102, 51, 52, 34, 44, 34, 110, 97, 109, 101, 34, 58, 34, 108, 97, 105, 114, 100, 34, 44, 34, 116, 105, 109, 101, 34, 58, 34, 48, 48, 48, 49, 45, 48, 49, 45, 48, 49, 84, 48, 48, 58, 48, 48, 58, 48, 48, 90, 34, 44, 34, 114, 115, 115, 105, 34, 58, 45, 52, 49, 44, 34, 108, 111, 82, 97, 83, 78, 82, 34, 58, 49, 51, 46, 53, 44, 34, 108, 111, 99, 97, 116, 105, 111, 110, 34, 58, 123, 34, 108, 97, 116, 105, 116, 117, 100, 101, 34, 58, 48, 44, 34, 108, 111, 110, 103, 105, 116, 117, 100, 101, 34, 58, 48, 44, 34, 97, 108, 116, 105, 116, 117, 100, 101, 34, 58, 48, 125, 125, 93, 44, 34, 116, 120, 73, 110, 102, 111, 34, 58, 123, 34, 102, 114, 101, 113, 117, 101, 110, 99, 121, 34, 58, 57, 48, 51, 57, 48, 48, 48, 48, 48, 44, 34, 100, 114, 34, 58, 49, 125, 44, 34, 97, 100, 114, 34, 58, 102, 97, 108, 115, 101, 44, 34, 102, 67, 110, 116, 34, 58, 51, 50, 44, 34, 102, 80, 111, 114, 116, 34, 58, 49, 44, 34, 100, 97, 116, 97, 34, 58, 34, 65, 87, 99, 65, 49, 81, 74, 122, 75, 101, 119, 68, 105, 65, 104, 54, 73, 119, 104, 54, 73, 119, 65, 65, 121, 65, 86, 111, 81, 81, 61, 61, 34, 44, 34, 111, 98, 106, 101, 99, 116, 34, 58, 123, 34, 116, 101, 109, 112, 101, 114, 97, 116, 117, 114, 101, 83, 101, 110, 115, 111, 114, 34, 58, 123, 34, 49, 34, 58, 50, 49, 46, 51, 125, 44, 34, 104, 117, 109, 105, 100, 105, 116, 121, 83, 101, 110, 115, 111, 114, 34, 58, 123, 34, 53, 34, 58, 51, 50, 46, 53, 125, 44, 34, 98, 97, 114, 111, 109, 101, 116, 101, 114, 34, 58, 123, 34, 50, 34, 58, 49, 48, 55, 51, 46, 50, 125, 44, 34, 103, 112, 115, 76, 111, 99, 97, 116, 105, 111, 110, 34, 58, 123, 34, 51, 34, 58, 123, 34, 108, 97, 116, 105, 116, 117, 100, 101, 34, 58, 53, 53, 46, 53, 53, 53, 53, 44, 34, 108, 111, 110, 103, 105, 116, 117, 100, 101, 34, 58, 53, 53, 46, 53, 53, 53, 53, 44, 34, 97, 108, 116, 105, 116, 117, 100, 101, 34, 58, 50, 125, 125, 125, 125]
java.lang.NumberFormatException: For input string: "${$.object.temperatureSensor["1"]}"
	at sun.misc.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2043)
	at sun.misc.FloatingDecimal.parseDouble(FloatingDecimal.java:110)
	at java.lang.Double.parseDouble(Double.java:538)
	at java.lang.Double.valueOf(Double.java:502)
	at org.thingsboard.gateway.util.converter.BasicJsonConverter.getKvEntry(BasicJsonConverter.java:152)
	at org.thingsboard.gateway.util.converter.BasicJsonConverter.getTsKvEntries(BasicJsonConverter.java:94)
	at org.thingsboard.gateway.extensions.mqtt.client.conf.mapping.MqttJsonConverter.parse(MqttJsonConverter.java:110)
	at org.thingsboard.gateway.extensions.mqtt.client.conf.mapping.MqttJsonConverter.convert(MqttJsonConverter.java:84)
	at org.thingsboard.gateway.extensions.mqtt.client.listener.MqttTelemetryMessageListener.messageArrived(MqttTelemetryMessageListener.java:42)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.deliverMessage(CommsCallback.java:491)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.handleMessage(CommsCallback.java:402)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.run(CommsCallback.java:206)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)

mp-loki, 2019-01-17:
@arrowcircle please disregard my previous comment. Actually, your initial configuration is almost correct, I have just changed filterExpression to '$' instead of '$.*' and it worked with your input.

arrowcircle, 2019-01-17:
@mp-loki Thanks! That worked fine!

closed 2019-01-17 by arrowcircle